MichaIng/DietPi #6809: Very first run from a fresh install gives error "HTTP/2 stream 1 was not closed cleanly before end of the underlying stream" while doing the DietPi-Update".

Sushiman1787, 2023-12-18:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information

**FRESH INSTALL**

- DietPi version | version 8.24 (Very last one, downloaded today).
- SBC model | RPi 1 model B
- Power supply used | 5V 1A
- SD card used | SanDisk

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes.

- Bug report ID | 003ec5f0-58ae-4ad1-97ab-672e708212cd

#### Steps to reproduce
Just doing a normal first run. 
Doing the "DietPi-Update" default and normal procedure.

#### Expected behaviour
This error should not be happening.

#### Actual behaviour
-  While doing the "DietPi-Update" this error happens:

"HTTP/2 stream 1 was not closed cleanly before end of the underlying stream".

#### Extra details
- Nothing was changed on any file. Everything is completely default even the details of the dietpi.txt are the default ones.

Sushiman1787, 2023-12-18:
Sure. The full Error log is exactly:

curl: (18) HTTP/2 stream 1 was not closed cleanly before end of the underlying stream

[image]

MichaIng, 2023-12-18:
Looks like a network issue between you and the GitHub server. Probably temporary. Does it still fail when you try the command manually?
```sh
cd /tmp
curl -fLO https://github.com/MichaIng/DietPi/archive/master.tar.gz
```

Sushiman1787, 2023-12-18:
Thank you for the possible solution.
You mean by By using the "Open subshell" option?

MichaIng, 2023-12-18:
You get a console, so you can run commands to check what's going on and in case fix things, without aborting the script. When exiting the subshell, you can "Retry" and continue the script. However this is more relevant for debugging issues or when something failed on a long software install process. In your particular case you can just exit and retry it later.

The command above it not for solving the issue, it is just the exact same command that failed during the update. So you can test outside the script whether it works or fails the same way.

Sushiman1787, 2023-12-19:
Hello there.
I tried your command manually and seeing I didn't know if in the beginning (-fLO) there was a 0 or a O I tried both.

When entering one, the bash just accepted it without any output or error.

When entering the other one, the bash output some timers and downloads speeds and ended without saying anything.

After that, I returned to the previous stuck step and did "retry" and the installation now carried on without a itch.

I don't know if it was related to your command or the fact that a dietpi version update appeared in the meantime, but it worked. 

Thank you very much for your help !!!

Now, I'm on to carry on with my search for a simple webcam viewer (which appears to be difficult as I apparently need an internet browser but Firefox and Chromium were disabled on armv6).

Do you by any chance know of a simple software or way, to simply see the output of my USB webcam (without internet or doing some stream) on a window and not a browser ?

MichaIng, 2023-12-19:
Great! Then it was just a temporary network issue, or one with the GitHub servers/mirrors accessed from your location.

closed 2023-12-19 by MichaIng