MichaIng/DietPi #6815: cant make your own distribution via ssh, please add warning

hanscees, 2023-12-22:
Hi,
I have been trying to get dietpi on an bananapi m1. 
But I have done this via ssh and I think the script uninstalls ssh server so the script stops and no working distro is produced. 

Perhaps you can add a wrning migh a foolish person like myself try installing via ssh.

#### Steps to reproduce
Follow this page
https://dietpi.com/docs/hardware/#make-your-own-distribution

I have installed raspbian on a bananapi m1
logged in via ssh and followed the page above. 

After running the dietpi script all goes well until

Purging configuration files for openssh-server (1:9.2p1-2) ...
client_loop: send disconnect: Broken pipe

#### Expected behaviour
dietpi script should not stop
- ...

#### Actual behaviour
connection to ssh is lost and presumably dietpi install script stops

Will go and look for a screen and keyboard now.

MichaIng, 2023-12-22:
Which Debian version is this? On all versions we support, the natively provided OpenSSH can be purged without aborting the current SSH session. See here:
```console
root@DietPi:~# ss -tulpn
Netid                         State                          Recv-Q                          Send-Q                                                   Local Address:Port                                                   Peer Address:Port                         Process
tcp                           LISTEN                         0                               128                                                            0.0.0.0:22                                                          0.0.0.0:*                             users:(("sshd",pid=3675,fd=3))
tcp                           LISTEN                         0                               128                                                               [::]:22                                                             [::]:*                             users:(("sshd",pid=3675,fd=4))
root@DietPi:~# tty
/dev/pts/0
root@DietPi:~# echo $SSH_CLIENT
192.168.1.11 54059 22
root@DietPi:~# apt autopurge openssh-server
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  libnsl2* libwrap0* openssh-server* openssh-sftp-server* runit-helper*
0 upgraded, 0 newly installed, 5 to remove and 0 not upgraded.
After this operation, 2457 kB disk space will be freed.
Do you want to continue? [Y/n]
(Reading database ... 16024 files and directories currently installed.)
Removing openssh-server (1:9.2p1-2+deb12u2) ...
Removing libwrap0:amd64 (7.6.q-32) ...
Removing libnsl2:amd64 (1.3.0-2) ...
Removing openssh-sftp-server (1:9.2p1-2+deb12u2) ...
Removing runit-helper (2.15.2) ...
Processing triggers for libc-bin (2.36-9+deb12u3) ...
(Reading database ... 15974 files and directories currently installed.)
Purging configuration files for openssh-server (1:9.2p1-2+deb12u2) ...
dpkg: warning: while removing openssh-server, directory '/etc/ssh/sshd_config.d' not empty so not removed
Purging configuration files for libwrap0:amd64 (7.6.q-32) ...
root@DietPi:~# tty
/dev/pts/0
root@DietPi:~# echo $SSH_CLIENT
192.168.1.11 54059 22
root@DietPi:~# ss -tulpn
Netid                            State                            Recv-Q                            Send-Q                                                        Local Address:Port                                                         Peer Address:Port                            Process
root@DietPi:~#
```
You see the "Purging configuration files for openssh-server", but no connection loss 🤔. We allow to switch back and forth between OpenSSH and Dropbear from SSH sessions, and I am not aware of anyone reporting issues with this.

At some point I plan to restructure the `dietpi-installer` to install new packages first and remove obsolete one afterwards. In case of the SSH server, this would mean that we need to stop the OpenSSH service at least, otherwise Dropbear could not be installed, or we would need to temporarily mask the Dropbear service to prevent it from trying to start. If you find time to test: With the fresh BananaPi image you use, does stopping the OpenSSH server process as well abort the connection?
```sh
systemctl stop ssh
```
This as well works fine on Debian/DietPi, as a mandatory step to allow switching between SSH servers.

hanscees, 2023-12-23:
purging here is succesfull too.
Yet the script is not 

```
root@bananapi:~# apt autopurge openssh-server
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  libproc2-0* openssh-server* openssh-sftp-server* procps* runit-helper*
0 upgraded, 0 newly installed, 5 to remove and 1 not upgraded.
After this operation, 4,599 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 40394 files and directories currently installed.)
Removing openssh-server (1:9.2p1-2) ...
Removing procps (2:4.0.2-3) ...
Removing libproc2-0:armhf (2:4.0.2-3) ...
Removing openssh-sftp-server (1:9.2p1-2) ...
Removing runit-helper (2.15.2) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for libc-bin (2.36-9) ...
(Reading database ... 40196 files and directories currently installed.)
Purging configuration files for openssh-server (1:9.2p1-2) ...
Purging configuration files for procps (2:4.0.2-3) ...
root@bananapi:~# 

```

MichaIng, 2023-12-23:
Then it was probably a coincidental network issue, since the script does just exactly this.

> Description:	Armbian 23.08.0-trunk bookworm

So you did use Armbian in the first place? Because you said "Raspbian".

hanscees, 2023-12-23:
Oh I am sorry, armbian. I am currently flashing the latest version.
But it happened twice, so I dont think it was a coincidence, but we will see with the latest image. 

Thanks for reacting so fast.

hanscees, 2023-12-23:
it died again but somewhere else. 

I am going to assume a bad card now. I will let you know here if a new card fixes this.

```
Purging configuration files for libsensors-config (1:3.6.0-7.1) ...
Purging configuration files for libmagic1:armhf (1:5.44-3) ...
Purging configuration files for bc (1.07.1-3) ...
Purging configuration files for screen (4.9.0-4) ...
removed '/etc/tmpfiles.d/screen-cleanup.conf'
Purging configuration files for rpcbind (1.2.6-6+b1) ...
```

hanscees, 2023-12-23:
Hmm. Tried it now running without ssh directly on the bananapi with screen and keyboard.

It died just before doing apt lots of packages including dropbear. 
I started a shell and the interface was up, but no ip address and route. So looks like dhcp does not work.

Added ip address and default route and script is now running installing things.

Probably the dns bug you mentioned. It causes interface problems which kills any ssh session of course.

MichaIng, 2023-12-25:
I guess so. I'll try to replicate in a few days and apply a fix. I was able to replicate (and understand from NetetwokManager `postrm` script) why DNS is lost, but not that/why the IP address or default route would go. While we can add a generic public DNS server, we would need to find a way to keep the previously set IP address and route. In case just hack the `postrm`, if we find the code which does this dumb thing.

hanscees, 2023-12-25:
I am just guessing here but my ethernet device was called end0 before dietpi took over armbian.
.
After installing its called eth0. /etc/networks/interfaces shows Ethernet and WIFI. 
It looks like scripts tried to put the dhcp information to the wifi network (I had disabled) 
 
I have disabled wifi
On my network neither ntp not dns traffic is allowed to internet. 

greetings
Hans-Cees

MichaIng, 2023-12-27:
The renamed network interfaces are expected, but this should not affect the `dietpi-installer` since it takes effect only after reboot. It is expected that `/etc/network/interfaces` contains those two interface definitions, but the WiFi interface is only enabled and in case edited at first boot if setup via `dietpi.txt`.