MichaIng/DietPi #6816: Migration script causes failure to load kernel modules on RPi 4B

Kreeblah, 2023-12-23:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 8.25.1
- Distro version | Bookworm
- Kernel version | `Linux ntp 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux`
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official RPi 4 PSU
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | Linux kernel
- Was the software title installed freshly or updated/migrated? Migrated using partition migration script for updated partition layout in prep for RPi 5
- Can this issue be replicated on a fresh installation of DietPi? I haven't tried, but this installation is very close to a fresh install, so I'd presume so.  I can try if you'd like, though.

- Bug report ID | `5ac5ec45-2d97-4b86-9c04-7f9f21d12ffe`

#### Steps to reproduce

1. I have a GPS-based time server set up, per the instructions I wrote at https://github.com/Kreeblah/DietPiTimeServer
2. After using `dietpi-update` to update to 8.25.1, I used the migration script per the instructions at https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1856412915

#### Expected behaviour

- I'd expect `chronyc sources` to return a list of time sources which includes a PPS source, and for the RTC to be available.

#### Actual behaviour

- `chrony` fails to start because `/dev/pps0` doesn't exist.  The PPS driver doesn't load, nor does the RTC driver.

#### Extra details

- I did some investigation in https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1868056698 and https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1868365932.  The settings from `/boot/config.txt` do seem to have been migrated over to `/boot/firmware/config.txt`, but for some reason, these overlays aren't being applied.

I do, however, see the PPS driver being loaded if I run `dmesg | grep pps`:

```
[    0.059924] pps_core: LinuxPPS API ver. 1 registered
[    0.059933] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
```

So, it seems to be confined to the overlay part.

MichaIng, 2023-12-25:
If the device tree overlay is loaded, `/etc/modules(-load.d)` entries shouldn't be required, since the overlays trigger the driver load.

So indeed it seems this overlay is broken with the new kernel packages. We'll need to have a look at and in case report it here: https://github.com/raspberrypi/linux/issues

Kreeblah, 2023-12-25:
Yep, the overlays for these not working is the only thing that makes sense to me for what's causing the issue here.

I grabbed some additional info on this (in case it helps folks track it down) since I reflashed the microSD card with a fresh DietPi image.  When I don't run the migration script, this is what I get for `dmesg | grep pps`:

```
[    0.060548] pps_core: LinuxPPS API ver. 1 registered
[    0.060558] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    2.049581] pps pps0: new PPS source pps@12.-1
[    2.049660] pps pps0: Registered IRQ 37 as PPS source
```

And those PPS source lines are missing afterwards.  Similarly, running `dmesg | grep rtc` before I run the script gets me:

```
[    2.938576] rtc-rv3028 1-0052: registered as rtc0
[    2.939998] rtc-rv3028 1-0052: setting system clock to 2023-12-25T06:18:05 UTC (1703485085)
```

But it doesn't find any lines afterwards.  Similarly, running `i2cdetect -y 1` afterwards gets me this:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- 52 -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

That shows something (the RTC) is detected at 0x52, but the driver isn't loaded because the overlay hasn't been applied.  When it's loaded (before running the migration script), that shows `UU` instead of `52` there.

Joulinar, 2023-12-25:
How does it behaves on latest RPi OS 64 bit? Important to ensure all apt packages updated to latest version.

Kreeblah, 2023-12-26:
> How does it behaves on latest RPi OS 64 bit? Important to ensure all apt packages updated to latest version.

That's a good question.  Interestingly enough, it works there.  I just flashed a microSD card with the current minimal 64-bit RPi OS available in the imaging program, upgraded all the packages on it, and ran through installing/configuring it for my GPS hat, and it does detect the PPS device and RTC as I'd expect when getting things configured (using `/boot/firmware/config.txt` instead of `/boot/config.txt`).

```
root@ntp:~# uname -a
Linux ntp 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
root@ntp:~# ls /dev/pps0
/dev/pps0
root@ntp:~# ls /dev/rtc0
/dev/rtc0
```

So, just in case it was an issue with my previous installation, I just tried reflashing the current DietPi Bookworm image, unmarked the rpi-eeprom package so I could upgrade it, upgraded that package, ran the migration script, and then went through setting things up the same way I did on RPi OS, and I'm getting the same results as before.  The PPS and RTC overlays aren't being applied and `/dev/pps0` and `/dev/rtc0` aren't being created.

Kreeblah, 2023-12-29:
There's followup at https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1870313108 and https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1871731055 if anybody finds this later and is looking for context.  Seems to be a case of the old kernel package removing some of the overlay files when it gets purged.

closed 2025-05-31 by MichaIng