MichaIng/DietPi #6817: Raspberry Pi 5 - Bsckup failed after migration from Pi 4

adrianog91, 2023-12-24:
#### Required Information
- DietPi version | 8.25.1
- Distro version | bookworm 0
- Kernel version | Linux DietPi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
- SBC model | RPi 5 Model B (aarch64)
- Power supply used | Original Pi 5 Power supply
- SD card used | SSD

#### Additional Information (if applicable)
- Software title | dietpi-backup

#### Steps to reproduce
1. Execute manual backup from dietpi-backup utlity

#### Expected behaviour
- Backup completed successfully

#### Actual behaviour
- Error during backup procedure

#### Extra details
- In the log of the backup procedure i found this error:
`rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]`
- I started the backup before migrating from my Pi4 and it worked fine. After migrating to the Pi 5 it stopped working.

Joulinar, 2023-12-24:
We would need the whole log from the backup. Should be possible to copy from SSH terminal.

@MichaIng it is the backup not the restore.

adrianog91, 2023-12-24:
I splitted the entire log in 2 files:

- 1_dietpi-backup.log is the first part
- 2_dietpi-backup.log is the second

[1_dietpi-backup.log](https://github.com/MichaIng/DietPi/files/13761801/1_dietpi-backup.log)
[2_dietpi-backup.log](https://github.com/MichaIng/DietPi/files/13761802/2_dietpi-backup.log)

MichaIng, 2023-12-24:
Here are the errors:
```
2023/12/24 12:44:26 [225081] >f+++++++++ home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/test/test_json/__pycache__/test_unicode.cpython-311.pyc
2023/12/24 12:44:36 [225081] rsync: [sender] readlink_stat("/mnt/myNASMedia") failed: Host is down (112)
2023/12/24 12:44:36 [225081] cd+++++++++ home/homeassistant/.pyenv/versions/3.11.4/lib/python3.11/test/test_peg_generator/
...
2023/12/24 12:44:39 [225081] >f+++++++++ mnt/dietpi_userdata/homeassistant/custom_components/hacs/repositories/theme.py
2023/12/24 12:44:39 [225081] rsync: [sender] opendir "/mnt/mynas" failed: No such device (19)
2023/12/24 12:44:39 [225081] cd+++++++++ mnt/dietpi_userdata/homeassistant/custom_components/hacs/repositories/__pycache__/
```
Is one of `/mnt/myNASMedia` or `/mnt/mynas` the (NFS) mount point you are backing this up to? "Host is down" seems self-explaining. If it really was down for 1 file write, at least this `__pycache__` directory is unimportant, as it gets recreated automatically if missing. And the directory afterwards got created, as its content was successfully copied (and it is for manual code tests only).

The second one sounds more like an abandoned mount point? Please check `df`, your `/etc/fstab` or `dietpi-drive_manager` for obsolete mounts, which are not attached, respectively where the remote NFS server is down.

adrianog91, 2023-12-24:
Yes I found the problem. It was the myNas disk that I removed a few months ago. On dietpi-driver-manager it wasn't displayed but I removed it by editing the file from: `/etc/fstab`.

The strange thing is that on the Pi 4 the backup still worked. I do not know the reason. Now I've solved it! Thanks and merry Christmas! I'm available if you need to do some tests on the Pi 5

closed 2023-12-24 by adrianog91