MichaIng/DietPi #6820: RPi 4 fails to upgrade packages

SMillerDev, 2023-12-25:
### Details:
- Date           | Mon Dec 25 16:44:54 GMT 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install vsftpd`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

Try upgrading any package
#### Expected behaviour:

- pkg upgrade
#### Actual behaviour:

- Upgrade is failing with unknown errors
#### Extra details:

- ...
#### Additional logs:
```console
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libwrap0
Recommended packages:
  logrotate ssl-cert
The following NEW packages will be installed:
  libwrap0 vsftpd
0 upgraded, 2 newly installed, 0 to remove and 17 not upgraded.
1 not fully installed or removed.
Need to get 205 kB of archives.
After this operation, 465 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 libwrap0 arm64 7.6.q-31 [58.7 kB]
Get:2 https://deb.debian.org/debian bullseye/main arm64 vsftpd arm64 3.0.3-12 [147 kB]
Bus error
Fetched 205 kB in 0s (957 kB/s)
Selecting previously unselected package libwrap0:arm64.
(Reading database ... 26793 files and directories currently installed.)
Preparing to unpack .../libwrap0_7.6.q-31_arm64.deb ...
Unpacking libwrap0:arm64 (7.6.q-31) ...
Selecting previously unselected package vsftpd.
Preparing to unpack .../vsftpd_3.0.3-12_arm64.deb ...
Unpacking vsftpd (3.0.3-12) ...
Setting up locales (2.31-13+rpt2+rpi1+deb11u7) ...
dpkg: error processing package locales (--configure):
 installed locales package post-installation script subprocess was killed by signal (Bus error)
Setting up libwrap0:arm64 (7.6.q-31) ...
Setting up vsftpd (3.0.3-12) ...

Configuration file '/etc/vsftpd.conf'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
 ==> Keeping old config file as default.
dpkg: error processing package vsftpd (--configure):
 installed vsftpd package post-installation script subprocess was killed by signal (Bus error)
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u7) ...
Errors were encountered while processing:
 locales
 vsftpd
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2023-12-25:
```
installed locales package post-installation script subprocess was killed by signal (Bus error)
```
The `postinst` script of the `locale` package fails. Not sure what this "Bus error" signal means. It fails as well when you run the configuration step manually?
```sh
dpkg --configure locales
```
If so, can you run this:
```sh
bash -x /var/lib/dpkg/info/locales.postinst configure
```

SMillerDev, 2023-12-25:
No dice on the first one:
```console
root@DietPi:~# dpkg --configure locales
Setting up locales (2.31-13+rpt2+rpi1+deb11u7) ...
dpkg: error processing package locales (--configure):
 installed locales package post-installation script subprocess was killed by signal (Bus error)
Errors were encountered while processing:
 locales
```

SMillerDev, 2023-12-25:
And the second is not showing any more info either:
```console
root@DietPi:~# bash -x /var/lib/dpkg/info/locales.postinst configure
+ set -e
+ LG=/etc/locale.gen
+ EE=/etc/default/locale
+ LC_ALL=C
+ LANG=C
+ '[' configure = configure ']'
+ . /usr/share/debconf/confmodule
++ '[' '!' '' ']'
++ PERL_DL_NONLAZY=1
++ export PERL_DL_NONLAZY
++ '[' '' ']'
++ exec /usr/share/debconf/frontend /var/lib/dpkg/info/locales.postinst configure
Bus error

```

MichaIng, 2023-12-25:
It is the `debconf` frontend call which fails. Since we force `noninteractive`, it is the perl script itself or `perl-base`. Try to reinstall those two packages:
```sh
apt install --reinstall debconf perl-base
```

SMillerDev, 2023-12-25:
That seems to have worked, thanks a ton @MichaIng !

closed 2023-12-25 by SMillerDev