MichaIng/DietPi #6822: FAILED first run, Rockport QuartzA board, flashed latest dietpi image to 128G eMMC card, powered on

Wilford888, 2023-12-27:
Date | Tue Dec. 27 2023
    DietPi version | v8.24.1 (MichaIng/master)
    Image creator |
    Pre-image |
    Hardware | RockPort QuartzA
    Kernel version | download from dietpi website
    Command | first run/bootup
    Exit code | Unknown install state/First run setup failed
    Software title | DietPi-Update

Steps to reproduce:

    Flash current Quartz dietpi image to 128G eMMC
    Install, power on

Expected behaviour:

    Expect to be able to log in and start making changes

Actual behaviour:

    Error Message:  [FAILED] Unknown install state/First run setup failed

Extra details:  interactive runs don't work, same error
* plenty of free space on the eMMC, log error below not making sense to me.

Additional logs:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v8.24.1 : 03:45 - Wed 12/27/23
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.xxx.xxx (eth0)

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] DietPi-TimeSync | systemctl stop systemd-timesyncd
[  OK  ] DietPi-TimeSync | mkdir -p /run/systemd/timesync
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.25.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.24.1
[ INFO ] DietPi-Update | Latest version  : v8.25.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[FAILED] DietPi-Update | Free space check: path=/ | available=84 MiB | required=100 MiB
 ─────────────────────────────────────────────────────
 DietPi v8.24.1 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.xxx.xxx (eth0)
[  OK  ] DietPi-Login | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_AUTOMATED=0
 ─────────────────────────────────────────────────────
 DietPi v8.24.1 : Update available
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.xxx.xxx (eth0)

Joulinar, 2023-12-27:
> [FAILED] DietPi-Update | Free space check: path=/ | available=84 MiB | required=100 MiB

looks like the disk has not been extended correctly. Can you share following

```
df -h
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the issue persists.

closed 2025-05-31 by MichaIng