MichaIng/DietPi #6823: Installing Docker and Portainer fails

binkleyz2, 2023-12-27:
### Creating a bug report/issue
- [ Y] I have searched the existing open and closed issues

#### Details:
- Date           | Wed Dec 27 09:59:59 EST 2023
- Bug report     | d974e242-9339-477c-9c94-fb614a3fa65d
- DietPi version | v8.25.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux 617NUC 6.1.0-16-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.67-1 (2023-12-12) x86_64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `pip3 uninstall -y docker-compose`
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... Adding Docker and Portainer via Dietpi-software
2. ...
#### Expected behaviour:

- ... Docker should install
#### Actual behaviour:

- ... Install fails
#### Extra details:

- ...
#### Additional logs:
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

MichaIng, 2023-12-27:
Hmm, did you upgrade this system from Bullseye to Bookworm? I am wondering how `pip` and the `docker-compose` module were installed in the first place without the related override for this `pip` block in place.

However, do the following and then retry:
```sh
grep -q '\[global\]' /etc/pip.conf 2> /dev/null || printf '%b' '[global]\n' | sudo tee -a /etc/pip.conf > /dev/null
sudo sed -i '/^\[global\]/a\break-system-packages=true' /etc/pip.conf
```

binkleyz2, 2023-12-27:
Actually, yes, I did just recently upgrade to bookworm using the script here:  https://dietpi.com/blog/?p=3128

I ran the two commands provided and it seems to be fine now.

Created a post-install debug report in case you want to see it - d974e242-9339-477c-9c94-fb614a3fa65d

MichaIng, 2023-12-27:
Did you install Docker Compose via `dietpi-software` before or manually? Python 3 should have been reinstalled, adding this setting to `/etc/pip.conf`, but not if Python was installed manually. And now Docker Compose does not require Python anymore.

binkleyz2, 2023-12-27:
I don't recall installing Docker Compose via the dietpi-software route before, and I also do not recall installing it manually, though that is a possibility..I've had this box up and running for a few years now and early on I was kind of just hacking at it to get functionality.  I use the box primarily for a VPN server and PiHole/Unbound setup, but have recently wanted to expand what the box is doing so figured Docker would be a good path forward.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the problem persists.

closed 2025-05-31 by MichaIng