MichaIng/DietPi #6825: DietPi-Software | Uninstall / Install / Update

twikedk, 2023-12-27:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
No problem just a feature request

#### Describe the solution you'd like:
I was about to update my baikal and gitea installation and it came to mind, that an update menu like the already existing uninstall and install menu would be nice.

And yes, I already know of the dietpi-software reinstall id command. :-)

MichaIng, 2023-12-27:
Agreed, at least the reinstall command could be easily added to the menu.

MichaIng, 2023-12-27:
Right, in some cases it skips updating the actual app, just assures dependencies are installed an service + probably configs. It also prints an info then, referring to the internal updater.

At some point it would be nice to have separate reinstall/update and reinstall/repair options, where the latter more aggressively installs new app files, regardless whether there is an internal/web updater available. But 1st things 1st, and we can add this info to the menu.