MichaIng/DietPi #6827: Image | Orange Pi Zero 2W

jamsinclair, 2023-12-30:
### Creating an image request

#### Formal device information
- Device name | Orange Pi Zero 2W
- Official product URL | http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-Zero-2W.html
- GitHub resource URL (if available) | https://github.com/orangepi-xunlong
- Image download URL | [Google Drive Link](https://drive.google.com/drive/folders/1Lnosr3bmdbG6F9RsDjY_RCLJl2zT4l61) – There's different images depending on the RAM installed, 1-2GB RAM, 1.5GB RAM and 4GB RAM.

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Kind of, made by the manufacturer (see image download URL)

#### If not, are there install instructions for Debian available?

----

I see previous issues that requested an Orange Pi image had been closed. If this is still the case, feel free to close this issue 🙇

jamsinclair, 2023-12-30:
I tried making my own distribution, following [the guide](https://dietpi.com/docs/hardware/#make-your-own-distribution), however,  I couldn't get the created image working on the device. I can only connect headlessly over wifi to the device, so this may have been the overall issue 😅 

Looking at the specs between the OPi Zero 3 and the  OPI Zero 2W there's a lot of similar components used, I've just flashed the Zero 3 image to my device and it _seems_ to be working ok so far... 🤞  and the wifi definitely works – due to my reliance on headless setup.

<table>
<tr>
 <td>
 <td> OPi Zero 3
 <td> OPi Zero 2W
 <td>
<tr>
 <td> CPU
 <td> Allwinner H618 Quad-core Cortex-A53 processor 1.5GHz 
 <td> Allwinner H618 Quad-core Cortex-A53 processor 1.5GHz
 <td>  ✅ 
<tr>
 <td> GPU
 <td> Mali G31 MP2 
 <td> Mali G31 MP2 
 <td> ✅ 
<tr>
 <td> Memory
 <td> 1GB/1.5GB/2GB/4GB LPDDR4
 <td> 1GB/1.5GB/2GB/4GB LPDDR4
 <td> ✅ 
<tr>
 <td> On-board storage
 <td> 16MB SPI Flash
 <td> 16MB SPI Flash
 <td> ✅ 
</table>

Are there any other board features that we should care about that might affect the usage of the Zero 3 image on the Zero 2W? On paper they look pretty similar, the only difference really being their form factor and the Zero 3 having some additional physical I/O, like ethernet.

deftdawg, 2024-01-23:
I'm also using the `OPi Zero 3` dietpi image on an `OPi Zero 2W`, only issue I've had is some instability because of thermals (system hot to the touch when it crashed).   

I changed the Governor to max out at 1008Mhz (down from 1502Mhz) and that seems to have stabilised the system.

MichaIng, 2024-01-23:
It actually makes sense, since the kernel/dtb package contains device trees for the Zero 2 and Zero 2W as well:
[image]

The bootloader is Zero 3 specific, but since it is the same SoC/RAM/SPI flash, it makes sense that this works as well on Zero 2. And after the kernel booted, the bootloader is not relevant anymore.

Only one thing would be good to check: Does the bootloader enforce the Zero 3 device tree or does it choose the correct one based on the actual model:
```sh
cat /proc/device-tree/model
```

Does someone have a Zero 2 non-W and can check whether it is the same issue with the Ethernet adapter disappearing on reboot?

jamsinclair, 2024-01-24:
>Only one thing would be good to check: Does the bootloader enforce the Zero 3 device tree or does it choose the correct one based on the actual model:
> ```
> cat /proc/device-tree/model
> ```
 
@MichaIng it doesn't seem to. Running this on the `OPi Zero 2W` returns `OrangePi Zero3`.

Are there any quick workaround to selecting the correct device tree during the bootload (or after the fact)?

MichaIng, 2024-01-24:
Try the following:
```sh
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=allwinner/sun50i-h616-orangepi-zero2w.dtb' /boot/dietpiEnv.txt
reboot
```

jamsinclair, 2024-01-25:
Thanks for the quick response! That makes sense, I'll be away from my OPi for a couple of weeks. If someone else can confirm in the interim, go ahead!

deftdawg, 2024-01-25:
G_CONFIG_INJECT said I didn't have permissions, when I prefixed with sudo command wasn't found. So applied the change with sudo nano and rebooted.

~~~
dietpi@DietPi:~$ cat /proc/device-tree/model
OrangePi Zero2 W
~~~

One thing I noticed is that with this Zero2 W booted with the proper config my dual nic carrier board doesn't work properly (nics light up, but don't work, stay lit after disconnecting Ethernet).  Reverting dietpiEnv.txt and rebooting restores function of the nics.

The carrier board connects via 4 pogo pins to the middle USB port (visible around green led)

[image]

As an aside, the following message is present regardless which dtb is used
~~~
-
[
57.0884801 sproul:wlan_delba_event: NOT FOUND sta_lut_index: 6, tid: 4
57.107166]
sprowl:wlan_delba_event: NOT FOUND
sta_lut_index: 6, tid: 5
57. 109182)
sprdul: wlan_delba_event: NOT FOUND sta_lut_index: 6,
_index: 6, tid: 6
57.1159271 sprdwl: wlan_delba_event: NOT FOUND sta_lut_index: 6, tid: 7
~~~

deftdawg, 2024-02-13:
I still don't get why the carrier board (which is a usb hub connected over pogo pins) would work fine out of the box with the ~OrangePi Zero 3~ device tree but not work at all with the ~OrangePi Zero 2 W~ tree.

MichaIng, 2024-02-15:
> I still don't get why the carrier board (which is a usb hub connected over pogo pins) would work fine out of the box with the OrangePi Zero 3 device tree but not work at all with the OrangePi Zero 2 W tree.

Hmm, that is nasty indeed 🤔.

samjw-nz, 2024-03-03:
Can confirm the Orange Pi Zero 3 image DOES technically work on the Zero 2W, however I can't get GPIO serial to work. I'm trying to use the Zero 2W as an Octoprint board in my Prusa, and can't get it working on any of the 5 GPIO serial connections. No issue with DietPi-Software or any of the typical usage stuff, but for the life of me I can't get communication working.
Having an attempt with converting the OrangePi image to DietPi but haven't gotten far yet.

MichaIng, 2024-03-06:
Some someone else test/verify this: https://dietpi.com/forum/t/manually-install-wifi-package-or-firmware-file/13182/21?u=michaing
- Bluetooth only works when WiFi is enabled as well. Or more precisely, the `modprobe sprdbt_tty` fails as long as `modprobe sprdwl_ng` has not been loaded.
- Loading `sprdwl_ng` via `/etc/modules-load.d/` at boot time fails, but it can be loaded later from console via `modprobe`.

samjw-nz, 2024-03-24:
Reckon I can give this a test this week. Do you have any example tests / commands I can use to verify? I'm not experienced with testing these things and don't currently have a bluetooth use-case, I'm just a user who just happens to own the hardware and have some time to chip in.

srimanachanta, 2024-03-30:
I installed the Orange Pi Zero 3 image on my Zero 2W and when running i2c-detect over the ports (should be i2c-1 on zero 2w so im checking i2c-3 on the zero3) my device isn't connected. I have went ahead and checked all other ports (0 and 1 are present when checking `/dev/i2c*` but also 2, 3, and 4.

Also, when modifying `/boot/dietpiEnv.txt` to set the dtb file to zero2w, it doesn't work, and after checking the dtb directory, only the file for zero3 is present.

el-bakkali, 2024-03-30:
I have the same issue,

I ran Michael command as recommended, and after reboot i can no longer ssh into the device.

srimanachanta, 2024-03-30:
Orange Pi supplies a Debian image [here](https://drive.google.com/drive/folders/1wjhR3YDvZzoBq7UiTYBgAUEWATIPNAjJ). Im going to install that and then install DietPi using [these docs](https://dietpi.com/docs/hardware/#make-your-own-distribution). Hopefully, that works better.

srimanachanta, 2024-03-30:
> I have the same issue,
> 
> I ran Michael command as recommended, and after reboot i can no longer ssh into the device.
> 
> As mentioned the zero2w dtb files does not exist in the dtb directory
> 
> Since this device uses the same chip as zero3 , it would be nice to have a universal image for it same way we do for raspberry pi etc. as the saying goes it will kill two bird with one stone.

I think the issue with a universal issue is that they have different IOs and bus-to-pin layouts. I'm not super knowledgeable on the DietPi system though, so I may be wrong here.

MichaIng, 2024-03-30:
Without a matching device tree, most hardware features won't work, simple as that. And indeed, the latest kernel build does not contain a Zero 2W device tree anymore:
[image]

Not sure why it was present in the first place. Mainline Linux 6.6 does not contain it either: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner?h=linux-6.6.y
But Linux 6.8 added one: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner?h=linux-6.8.y
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?id=c505ee1

There is already a draft at Armbian to add support based on this DTB: https://github.com/armbian/build/pull/6451

Allwinner edge kernel builds with Armbian are still Linux 6.7, otherwise we would not need to wait for the PR: https://github.com/armbian/build/blob/main/config/sources/families/include/sunxi64_common.inc

I just started builds based on this PR:
- https://github.com/MichaIng/DietPi/actions/runs/8491375886/job/23263287258
- https://github.com/MichaIng/DietPi/actions/runs/8491377766/job/23263290664

srimanachanta, 2024-03-30:
Is there is an Image I can test with? I don't see any artifacts under those builds unless I'm just blind lol.

Joulinar, 2024-03-30:
There is no image yet.

MichaIng, 2024-03-30:
These were kernel and bootloader builds. I triggered an image build here: https://github.com/MichaIng/DietPi/actions/runs/8491738902/job/23264002541

It can be found here once done: https://dietpi.com/downloads/images/testing/

srimanachanta, 2024-03-30:
> These were kernel and bootloader builds. I triggered an image build here: https://github.com/MichaIng/DietPi/actions/runs/8491738902/job/23264002541
> 
> It can be found here once done: https://dietpi.com/downloads/images/testing/

Thank you, I will test that out after its done and report back

MichaIng, 2024-03-30:
Btw, the SBC has no GPIO header exposed by default, but it can be soldered onto the PCB to be usable? I.e. does it make sense to enable a UART console by default, and e.g. expose the SPI toggle we added for Orange Pi Zero 3?

Or does it support USB UART on one of the USB ports in OTG/gadget mode?

srimanachanta, 2024-03-30:
> Btw, the SBC has no GPIO header exposed by default, but it can be soldered onto the PCB to be usable? I.e. does it make sense to enable a UART console by default, and e.g. expose the SPI toggle we added for Orange Pi Zero 3?
> 
> Or does it support USB UART on one of the USB ports in OTG/gadget mode?

Not sure about the latter part but yes, it includes a 2x20 header set that I had to solder on the SBC. I think UART console should be left the same as the OPI Zero3 because people are already prompted to disable it if they aren't using it. I also thing its supported over USB but regardless, prob just leave it on.

srimanachanta, 2024-03-30:
@MichaIng Seems the [build](https://github.com/MichaIng/DietPi/actions/runs/8491738902/job/23264002541) failed, I don't think an artifact was uploaded either.

MichaIng, 2024-03-30:
Yes, I forgot to move the bootloader build in place. It is rerunning.

srimanachanta, 2024-03-30:
Not sure why but I don't have an HDMI output nor can I SSH or ping it. For context, I am setting up my OPI headlessly by preseeding the network info according to [this](https://www.youtube.com/watch?v=vlMpn9u0Y4o) guide. I went back and reinstalled the DietPi OPI Zero3 image and can confirm that still works (minus HDMI) in terms of immediately connecting and me being able to ping and SSH into it.

MichaIng, 2024-03-30:
You tried it with this image? https://dietpi.com/downloads/images/testing/DietPi_OrangePiZero2W-ARMv8-Bookworm.img.xz
And configured WiFi for first boot via `dietpi.txt` and `dietpi-wifi.txt`?

srimanachanta, 2024-03-30:
> You tried it with this image? https://dietpi.com/downloads/images/testing/DietPi_OrangePiZero2W-ARMv8-Bookworm.img.xz And configured WiFi for first boot via `dietpi.txt` and `dietpi-wifi.txt`?

Yes

MichaIng, 2024-03-30:
Maybe the bootloader does not pick the correct device tree automatically. I'll redo a build later with this defined explicitly.

el-bakkali, 2024-03-30:
thank you Michael for your work @MichaIng 
its much appreciated .

MichaIng, 2024-03-30:
Build is rerunning, with UART enabled, SPI toggle and device tree set explicitly in boot config: https://github.com/MichaIng/DietPi/actions/runs/8491738902/job/23267416633

srimanachanta, 2024-03-30:
> Build is rerunning, with UART enabled, SPI toggle and device tree set explicitly in boot config: https://github.com/MichaIng/DietPi/actions/runs/8491738902/job/23267416633

Still doesn't work. Same issues as previously mentioned

MichaIng, 2024-03-30:
Not sure then. Looks like we need someone with a GPIO header and UART adapter to check serial console logs. The PR at Armbian is also still in draft, quite possible that it is not yet fully functional. In case it is just the bootloader, since the Zero 3 image did boot for @jamsinclair, we could try it with a new Zero 3 image: https://github.com/MichaIng/DietPi/actions/runs/8493763090
Once finished, it can be found here: https://dietpi.com/downloads/images/testing/
Flash it and then you just manually change the device tree (which is present in this kernel build):
```sh
fdtfile=allwinner/sun50i-h616-orangepi-zero2w.dtb
```
Change this in `dietpiEnv.txt` on the FAT partition, prior to first boot.

Ah, btw @srimanachanta do you have a variant with 1.5 GB RAM? Because that one won't boot with the Armbian bootloader build, same as on Zero 3. Then we could use the bootloader build from Orange Pi. But since I have the board here now, I am working with someone at OpenWRT to test and in case add a patch to add support for 1.5 GB RAM to mainline U-Boot as well. At least we already know how do hardcode it, so there is still a dedicated U-Boot build required, but at least it can be mainline U-Boot.

srimanachanta, 2024-03-30:
> Not sure then. Looks like we need someone with a GPIO header and UART adapter to check serial console logs. The PR at Armbian is also still in draft, quite possible that it is not yet fully functional. In case it is just the bootloader, since the Zero 3 image did boot for @jamsinclair, we could try it with a new Zero 3 image: https://github.com/MichaIng/DietPi/actions/runs/8493763090 Once finished, it can be found here: https://dietpi.com/downloads/images/testing/ Flash it and then you just manually change the device tree (which is present in this kernel build):
> 
> ```shell
> fdtfile=allwinner/sun50i-h616-orangepi-zero2w.dtb
> ```
> 
> Change this in `dietpiEnv.txt` on the FAT partition, prior to first boot.
> 
> Ah, btw @srimanachanta do you have a variant with 1.5 GB RAM? Because that one won't boot with the Armbian bootloader build, same as on Zero 3. Then we could use the bootloader build from Orange Pi. But since I have the board here now, I am working with someone at OpenWRT to test and in case add a patch to add support for 1.5 GB RAM to mainline U-Boot as well. At least we already know how do hardcode it, so there is still a dedicated U-Boot build required, but at least it can be mainline U-Boot.

I don't have a 1.5GB variant, sorry. I will try the Zero3 with the changes sometime later.

MichaIng, 2024-03-30:
Okay, I'll also redo a Zero 2W image with the Zero 3 bootloader tomorrow, but need to sleep now 🙂.

srimanachanta, 2024-03-30:
> Not sure then. Looks like we need someone with a GPIO header and UART adapter to check serial console logs. The PR at Armbian is also still in draft, quite possible that it is not yet fully functional. In case it is just the bootloader, since the Zero 3 image did boot for @jamsinclair, we could try it with a new Zero 3 image: https://github.com/MichaIng/DietPi/actions/runs/8493763090 Once finished, it can be found here: https://dietpi.com/downloads/images/testing/ Flash it and then you just manually change the device tree (which is present in this kernel build):
> 
> ```shell
> fdtfile=allwinner/sun50i-h616-orangepi-zero2w.dtb
> ```
> 
> Change this in `dietpiEnv.txt` on the FAT partition, prior to first boot.
> 
> Btw @srimanachanta do you have a variant with 1.5 GB RAM? Because that one won't boot with the Armbian bootloader build, same as on Zero 3. Then we could use the bootloader build from Orange Pi. But since I have the board here now, I am working with someone at OpenWRT to test and in case add a patch to add support for 1.5 GB RAM to mainline U-Boot as well. At least we already know how do hardcode it, so there is still a dedicated U-Boot build required, but at least it can be mainline U-Boot.
> 
> EDIT: Ah, @srimanachanta with the Zero 3 image, at least WiFi worked well for you as well? Then I am more certain that the bootloader is the issue, and hence using the Zero 3 image with Zero 2W device tree hopefully works. No big issue, in this case, to switch the bootloader for our actual Zero 2W builds.

Ok, tried the new Zero3 image with your recommendations for the device tree and it still doesn't work. I went back and used the same image but didn't set that line and that worked. The base (current release) image for OPI3 does work still so my board isn't bricked 😀 .

MichaIng, 2024-03-30:
The bootloader for the Zero 3 has not changed, it is still the same build from January. I redid only the kernel 🤔. Let's wait for someone else to verify, and in case for someone with GPIO header and UART adapter.

... oh good point with the device tree name, it should be:
```sh
fdtfile=allwinner/sun50i-h618-orangepi-zero2w.dtb
```
However, the Zero 2W image has this set correctly. But please try to adjust that one, as it could be still a bootloader issue. Should be still possible via FAT partition.

srimanachanta, 2024-03-31:
@el-bakkali Can you test out the new Orange Pi Zero 3 image from the testing downloads section?

MichaIng, 2024-03-31:
New image up which uses the Orange Pi Zero 3 bootloader instead, but Orange Pi Zero 2W device tree: https://dietpi.com/downloads/images/testing/

srimanachanta, 2024-03-31:
> New image up which uses the Orange Pi Zero 3 bootloader instead, but Orange Pi Zero 2W device tree: https://dietpi.com/downloads/images/testing/

Doesn't work 😔

RootLUG, 2024-04-02:
I will try to help with testing, my 4G and 1G version just arrived, also with the 24pin extension board. I have already soldered the 40pin header on the 1G version so I can try to test that but I am not sure how to do that properly? 

The last image for me boot fines, on the 4G version , for some reason it didn't boot occasionally between reboots, but always did after re-plugging the power. haven't observed/replicated this problem at 1G.

The 24pin extension board does not work fully. The usb ports are working as I see logs in dmesg when inserting usb but for example the eth port appears to not be powered (it is powered when power is applied but then it is disabled) I have not found a way yet to test the buttons or IR receiver.
On-board wlan0 worked fine after enabling it in the dietpi-config, didn't had a chance to test bluetooth yet. this is all on the 1G version, I will try to play more with 4G to see if I can replicate the problem with not booting up.

MichaIng, 2024-04-02:
> Edit: I built and flashed the upstream Armbian image and can confirm that it boots and I can setup networking and stuff. The issue for DietP is isolated here.

Since there isn't any Orange Pi Zero 2W image from Armbian, the PR has not even been merged, which image do you mean? The community build for the Zero 3? We use the same bootloader and kernel, and the bootloader did not change since it worked for you in the past, so it looks a little random to me 🤔.

@RootLUG 
Many thanks for testing and verifying that it does boot in general. You tried this image, right? https://dietpi.com/downloads/images/testing/DietPi_OrangePiZero2W-ARMv8-Bookworm.img.xz
I'll revert to using the actual Zero 2W bootloader, as I think srimanachanta was/is suffering from an isolated issue (which requires a serial console to investigate), while the Zero 2W bootloader might actually work. Does HDMI work for you?

About Ethernet, is the adapter detected at all?
```sh
ip l
```
And do you try to use onboard Ethernet via 20pin extension interface? If so, how is this accessible, actually? Via ribbon cable? Or do you mean a USB Ethernet adapter?

The PR at Armbian is still WIP and received some commits recently. Rebuilding kernel (and bootloader) might change/fix things. At least I see USB via 24pin interface was tested.

MichaIng, 2024-04-02:
This is exactly what I did for our first Zero 2W image with Zero 2W bootloader. But as said, there were commits in between, so I'll redo that.

HeikoGr, 2024-04-09:
To all the users with ethernet problems.
I poked around at the orangepi github source (https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/packages/raspi/orangepi/common/files/orangepi-hardware-optimization) and found an interesting part where the ethernet interface for bookworm based kernels maybe is named „end0“ instead of „eth0“.

RootLUG, 2024-04-10:
Makes sense,
I tried to run it manually, but I got permission denied:
```
root@DietPi:~# echo "end0"> /sys/class/leds/100m_act/device_name
-bash: /sys/class/leds/100m_act/device_name: Permission denied
root@DietPi:~# echo "end0"> /sys/class/leds/100m_link/device_name
-bash: /sys/class/leds/100m_link/device_name: Permission denied

root@DietPi:~# find / -name "orangepi-hardware-optimization" 2>/dev/null
root@DietPi:~# echo netdev > /sys/class/leds/100m_act/trigger
-bash: echo: write error: Invalid argument
root@DietPi:~# cat /sys/class/leds/100m_act/trigger
[none] usb-gadget usb-host kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock usbport disk-activity disk-read disk-write mtd nand-disk heartbeat cpu cpu0 cpu1 cpu2 cpu3 activity default-on panic mmc0 mmc1 rfkill-any rfkill-none rc-feedback rfkill0
```

The file (hw optimizations) also doesn't appear to exists on dietpi? Does it need to run at specific time during boot or something I can try/test?

MichaIng, 2024-04-20:
The interface is always named `eth0`. `end0` is what systemd renames it to, in case this is not prevented via kernel command-line options, and we do prevent it.

@RootLUG 
You need to load the kernel module for interface-triggered LEDs:
```sh
modprobe ledtrig-netdev
echo netdev > /sys/class/leds/100m_act/trigger
echo eth0 > /sys/class/leds/100m_link/device_name
echo 1 > /sys/class/leds/100m_link/link
```
However, there is no point to deal with the network adapter LEDs, as long as the network interface itself is not available. Note that these are not the LEDs at the adapter side, but an additional LED at the PCB or case. See e.g. the WAN/LAN1/LAN2 LEDs at the NanoPi R6C case here:

[image]

The LEDs right at the adapter socket are not controlled via these sysfs LED devices.

I generated a new image: https://dietpi.com/downloads/images/testing/
This is again using the Zero 2W bootloader. Hopefully it fixes Ethernet for you.

RootLUG, 2024-04-24:
Thanks for the info, just a quick update as I had only few minutes to play with it, the ethernet interface is still not visible in the latest image. I would try to get more time this week to pull out some logs as I saw some pin setup errors in dmesg. Is there anything besides dmesg logs that I should copy over that could help with debugging the issue?

RootLUG, 2024-04-25:
here are the dmesg logs from booting up:

[logs.txt](https://github.com/MichaIng/DietPi/files/15112216/logs.txt)

MichaIng, 2024-04-25:
Thanks. Okay, so the Ethernet driver starts to load, throws some "not found" errors which I do not fully understand, and finally probing the adapter fails with error 110, which seems to be some generic transfer timeout error.

alexl83, 2024-04-27:
dear @MichaIng ,
I reported the same issue to Armbian in [https://github.com/armbian/build/pull/6451#issuecomment-2080513242](https://github.com/armbian/build/pull/6451#issuecomment-2080513242): unfortunately there's no solution yet and it's a community maintained board, so no assurance about it

 It seems that ethernet PHY is not getting powered, output dmesg:
```
[    2.119267] dwmac-sun8i 5020000.ethernet: IRQ eth_wake_irq not found
[    2.119296] dwmac-sun8i 5020000.ethernet: IRQ eth_lpi not found
[    2.119502] dwmac-sun8i 5020000.ethernet: supply phy-io not found, using dummy regulator
[    2.119726] dwmac-sun8i 5020000.ethernet: PTP uses main clock
[    2.119757] dwmac-sun8i 5020000.ethernet: Current syscon value is not the default 58000 (expect 0)
[    2.139330] dwmac-sun8i 5020000.ethernet: No HW DMA feature register supported
[    2.139363] dwmac-sun8i 5020000.ethernet: RX Checksum Offload Engine supported
[    2.139371] dwmac-sun8i 5020000.ethernet: COE Type 2
[    2.139379] dwmac-sun8i 5020000.ethernet: TX Checksum insertion supported
[    2.139387] dwmac-sun8i 5020000.ethernet: Normal descriptors
[    2.139395] dwmac-sun8i 5020000.ethernet: Chain mode enabled
[    2.139415] dwmac-sun8i 5020000.ethernet: device MAC address da:90:21:c8:74:0c
[    2.300160] dwmac-sun8i 5020000.ethernet: EMAC reset timeout
[    2.300192] dwmac-sun8i 5020000.ethernet eth0: stmmac_dvr_remove: removing driver
[    2.316507] dwmac-sun8i: probe of 5020000.ethernet failed with error -110
```

reference info link: [https://linux-sunxi.org/Sun8i_emac](https://linux-sunxi.org/Sun8i_emac)

I came accross a possible kernel patch I cannot evaluate relevancy for: [https://github.com/jernejsk/linux-1/commit/25b44143ea8162209beb02759ca3ea3bd3be7a74](https://github.com/jernejsk/linux-1/commit/25b44143ea8162209beb02759ca3ea3bd3be7a74)

Hope it helps
Thanks 🙏🏻

alexl83, 2024-04-27:
One way of solving would be to adapt [this patch](https://github.com/orangepi-xunlong/linux-orangepi/commit/97b476246bc79756423cf6c1d4907606c9633d4b.patch) (and more perhaps) to recent current/edge kernels

orangepizero2w emac uses a different kernel driver CONFIG_SUNXI_GMAC and `ethernet@5030000` entry in DTB configured as per their code

Their forked 6.1.39-sun50iw9 version works, I built a test bookworm release using their outdated and plagiarized armbian framework

hope it helps

Ale

MichaIng, 2024-05-02:
Hmm part of this patch is already done at Armbian: https://github.com/armbian/build/blob/main/patch/kernel/archive/sunxi-6.1/patches.armbian/mfd-Add-support-for-X-Powers-AC200.patch

You could apply it on top of all Armbian patches? I'll test it now.

MichaIng, 2024-05-04:
I see you are already one step further: https://github.com/alexl83/orangepizero2w_patch
- The first error is strange:`mdiobus_scan` is declared in `phy.h`, which is included.
- The 2nd is indeed due to an upstream change: `of_get_named_gpio_flags` **was** declared in `of_gpio.h`, which is included. But this API has been removed upstream: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/include/linux/of_gpio.h?id=40fc56ee608cdb20022c225ac6f1e4b7ba63f8f1
    I could not find a replacement at quick view.

RootLUG, 2024-05-10:
I assume the patch is not yet merged as the behaviour is the same on latest testing image & then switching to beta branch per instructions. I also noticed that occasionally it kernel panic on shutdown, screenshot below:
[image]

I've been trying to test also bluetooth but got error connecting to the hci socket: `Can't open HCI socket: Address family not supported by protocol` when trying to interact with hcitool and hciconfig. No message in dmesg afterwards, that's about as far as I got with testing ble

MichaIng, 2024-05-10:
A patch is being worked on, basically done. I'll do a test build tomorrow morning. Your kernel panic on shutdown however does not look related. It prevents shutdown/reboot, right? I guess this will become better with Linux 6.8, which has native Orange Pi Zero 2W support, just without Ethernet, i.e. once Armbian sunxi64 edge has been moved to Linux 6.8 (or newer).

Btw, @alexl83 in the context of Igor's post, blaming us for not contributing to Armbian: I did, I would love to continue doing so, like the sunxi64 edge migration + test on all SBCs I have. But Igor put me onto the GitHub orga's blacklist, so I can't. I.e. he is blaming us for not contributing, but preventing it by himself. So I am forced to carry an increasing number of Armbian build system patches/commits on our fork, which I would love to open PRs with. I could expand the long story, if interested (while there is more before I joined the DietPi project), but TL;DR is that Igor just wants money from us, no time donation/contributions. But we do not earn money with DietPi, especially not with the SBCs we use the Armbian build system for kernel and U-Boot builds for: DietPi is mostly RPi and x86_64, in user numbers, hence donations.

alexl83, 2024-05-10:
Hi @MichaIng, I'm late answering - been a bit taken by work and family things
my testing of opi02w test branch armbian has been limited, perhaps I missed the kernel panic(s) due to my sessions being quite short - will expand hopefully starting next week

I understand that FOSS world is complicated, relationships perhaps evolve over time and can get complicated
I suppose both communities and LEaders can try and rebuild a bit of rapport / collaboration over time in small steps - I imagine pull requests might represent a possible starting point

PAssion in tech, humble opinion ,should unite people: but again, world is now messier than ever, so complications can build up over time - I sincerely hope a bit of common ground to re-start constructive dialogue could pop up soon 
a good spark is all that's needed :)

MichaIng, 2024-05-10:
> I suppose both communities and LEaders can try and rebuild a bit of rapport / collaboration over time in small steps - I imagine pull requests might represent a possible starting point

As said, I am on the block list of `github.com/armbian` orga, which prevents me from interacting with any of their repos, i.e. no PRs, issues, comments, reactions, even despite past contributions, I do not appear in the contributors list anymore (of course in my commit messages, else it would be illegal). So I would love to contribute and give something back to Armbian, but Igor obviously does not want me to 🙈. All I can do is offering others on Armbian Discord and forum to use our commits, where it fits.

Romaster1985, 2024-05-25:
Hello people's pi 😜. I'm looking for a image of Android 12 (not TV) for this Orange Pi Zero 2W 4GB RAM and not found in any web site but y trie with image for Banana Pi M4 Zero and boot Android 12 and recognize 4GB RAM but the H618 processor is too hot and it does not work the Wifi and Bluetooth.

https://wiki.banana-pi.org/Banana_Pi_BPI-M4_Zero

Me gustaría que alguien me ayude cocinando una ROM de Android 12 (no TV) para ésta SBC que funcione bien o ver la manera para que la ROM de la Banana Pi funcione con el kernel y todo lo que necesita la Orange Pi Zero 2W para funcionar por completo sin problemas.

Tks. ;)

regrepl, 2024-05-25:
@ Romaster1985  I don’t know about android. I tried androidtv image that is linked in orangepi page, but that even didn’t boot.

I tried dietpi version 9.4.2, unfortunately still some problems. I boots but it gives a lot of errors. No ethernet and wifi at first boot and a very nasty kernel panic when trying to do a soft reboot as on above picture.
After hard reboot I getting wifi and can configure it, however still no ethernet. After configuration there is no longer kernel panic with soft reboot, but also doesn't do a reboot just getting off. 

I am attaching logs.
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/15444397/dietpi-firstrun-setup.log)
[dmesg.txt](https://github.com/MichaIng/DietPi/files/15444398/dmesg.txt)
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/15444399/dietpi-firstboot.log)

MichaIng, 2024-05-25:
@regrepl 
Since you did not enable WiFi via `dietpi.txt` (and `dietpi-wifi.txt` if you want to have it connecting to an SSID), it was disabled on first boot. However, it should be possible to re-enable it via `dietpi-config`. As far as I can see from your logs, you indeed re-enabled WiFi (the module/driver), but did not configure an SSID. I guess it said something like "no adapter found"? The kernel logs indeed show an error loading the driver:
```
[  107.568290] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  107.569051] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  107.569672] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[  107.607743] [ADDR_MGT] addr_mgt_probe: module version: v1.0.9
[  107.607847] [ADDR_MGT] addr_mgt_probe: success.
[  107.612943] WCN: marlin_init entry!
[  107.613487] WCN: wcn config bt wake host
[  107.613729] WCN_ERR: dts node for bt_wake not found
[  107.613733] WCN: marlin2 parse_dt some para not config
[  107.620540] sdiohal:sdiohal_parse_dt adma_tx:1, adma_rx:1, pwrseq:0, irq type:data, gpio_num:0, blksize:840
[  107.622759] sdiohal:sdiohal_init ok
[  107.623348] WCN: marlin_probe ok!
[  107.631026] WCN: start_marlin [MARLIN_WIFI]
[  107.631052] WCN: marlin power state:0, subsys: [MARLIN_WIFI] power 1
[  107.631058] WCN: the first power on start
[  107.736038] WCN: marlin chip en dummy pull up -- need manually set GPIO 
[  107.764041] sdiohal:sdiohal_scan_card
[  107.764074] Error: Driver 'sdiohal' is already registered, aborting...
[  107.773602] sdiohal err:sdio_register_driver error :-16
[  110.820073] WCN_ERR: wait SDIO rescan card time out
[  110.829104] WCN_ERR: chip power on fail
[  110.836725] sdiohal:sdiohal_runtime_put entry
[  110.836734] sdiohal err:sdiohal_runtime_put line 1756 not have card
[  110.867413] sdiohal err:sdiohal_tx_thread line 152 not have card
[  113.892085] sprdwl:[WIFI_CMD_SYNC_VERSION]timeout
[  113.902272] sprdwl:ctx_id:0 cmd: WIFI_CMD_SYNC_VERSION[9] rsp timeout (mstime = 110863), num=1
[  113.916481] sprdwl:SYNC CMD ERROR!!
[  113.926257] sprdwl:sprdwl_tx_deinit cmd not yet transmited, cmd_send:1, cmd_poped:0
[  123.168445] sprdwl:sprdwl_msg_deinit ref not ok! wait for pop!
[  126.199219] unisoc_wifi: probe of unisoc_wifi failed with error -1
```
Looks like this one has an issue to load when it was unloaded first. To test this explicitly (when being connected via Ethernet or HDMI + keyboard):
1. Enable WiFi via `dietpi-config`, but do not configure an SSID. Verify that an adapter has been found, i.e. the menu to configure an SSID is shown.
```sh
ip l
modprobe -r sprdwl_ng
sleep 1
modprobe sprdwl_ng
sleep 1
ip l
```
The first command should then list `wlan0`, the last command probably not anymore. If this appears to be true, we should avoid to unload and blacklist this module, at least on first boot. Later, we can special-handle this SBC to show a prompt and offer a reboot when WiFi was enabled. AFAIK, unloading and re-loading the WiFi related kernel module works on all other SBCs, but I'll re-test on Orange Pi Zero 3, since it is so similar.

Regarding WiFi, I am currently doing a kernel build based on the WIP PR to fix Ethernet: https://github.com/MichaIng/DietPi/actions/runs/9238103446
Once it has (successfully) finished, test this:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-sunxi64.deb
dpkg -i linux-{image,dtb}-current-sunxi64.deb
reboot
```
And see whether `eth0` does now show up:
```sh
ip l
```

[55 more replies]

closed 2025-06-09 by MichaIng