MichaIng/DietPi #6828: Kodi fails to start

oradke, 2023-12-30:
### Creating a bug report/issue
- [ X ] I have searched the existing open and closed issues

#### Required Information
- DietPi version G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version bookworm 0
- Kernel version | ``
- SBC model RPi 4 Model B (aarch64)
- Power supply used: original Raspberry
- SD card used Sandisk Ultra 64GB

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? Clean install
- Can this issue be replicated on a fresh installation of DietPi?
yes

#### Steps to reproduce

- Create dietpi image
- indicate installation of Kodi in dietpi.txt
- indicate autostart to Kodi
- boot SD card
- wait for installation to finish and reboot
#### Expected behaviour

- pi should start intop Kodi

#### Actual behaviour

- after installation finishes, screen stops at root prompt
- after a reboot, screen stays black
- journalctl revelas an error:
```
Dez 30 14:46:36 Kodi kodi.bin[12312]: pw.conf: can't load config client.conf: No such file or directory
Dez 30 14:46:36 Kodi kodi.bin[12312]: pw.conf: can't load default config client.conf: No such file or directory

```

#### Extra details

- using original Raspberry TFT touchscreen and on-board 3.5mm headphone jack

Joulinar, 2023-12-30:
Found something similar on Kodi forum https://forum.kodi.tv/showthread.php?tid=371581

MichaIng, 2023-12-30:
Strange error. I never heard that Kodi requires any config file, aside of the ones it auto-generates in the user's home dir. Do you as well get this segmentation fault from the Kodi forum?
```sh
dmesg -l 0,1,2,3
```
And do you have a crash log as well, either in your home dir, or in its `.kodi` sub directory somewhere.

oradke, 2023-12-30:
Here's the log:
[kodi.log](https://github.com/MichaIng/DietPi/files/13799211/kodi.log)

after trying

```
sudo apt install pipewire pipewire-pulse
sudo apt install pipewire-audio-client-libraries
sudo apt install gstreamer1.0-pipewire libspa-0.2-bluetooth libspa-0.2-jack
```

the error disappeared from journalctl, but Kodi still won't start.

MichaIng, 2023-12-30:
I had such a feeling that those two errors were unrelated. `pw` for PipeWire makes sense, which is the new default audio stack on RPi OS. But I think we would have seen a lot more reports if it would not also just work with ALSA.

This will be the major issue:
```
error <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
```
KMS/DRM is not enabled. This does not exist, right?
```sh
ls -l /dev/dri
```
This should have been enabled automatically on Kodi installation. Can you check whether the overlay was added:
```sh
grep vc4-kms-v3d /boot/config.txt
```
If this does not give a result, can you run this command and check again:
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d 512
```
And in case the overlay was now added, reboot and retry.

Did you add the TFT via `dietpi-config` after Kodi installation? And/or set another resolution there? All these options make use of the legacy graphics stack, while Kodi requires the new one (KMS). Also changing the resolution to one other option than the KMS one(s), will disable it. We need to switch all this to the new graphics stack anyway, now more then before since the RPi 5 compatible firmware packages removed vast parts of the legacy stack. Until then, better follow the official docs to setup the display: https://www.raspberrypi.com/documentation/accessories/display.html
In case the resolution is wrong, you can remove the two `framebuffer_width`/`framebuffer_height` settings from `/boot/dietpi.txt` or set them to the correct values.

oradke, 2023-12-31:
Thanks for the reply! Let me go through all the items:
- /dev/dri does NOT exist
- the grep command returns NOTHING, so overlay way NOT added!
- command ran fine, afterwards grep found the overlay
- after reboot, screen was dark, but Kodi log had no errors
- I remember this happened after the very first install. I had actually used dietpi-config. Perhaps that threw KODI off
- dietpi.txt does not have framebuffer settings
- I edited boot.txt and the screen changed, but Kodi still crashes with 
`critical <general>: CApplication::CreateGUI - unable to init windowing system`
Have attached 
[kodi.log](https://github.com/MichaIng/DietPi/files/13800440/kodi.log)
new log file.

oradke, 2023-12-31:
Adding
`display_auto_detect=1`
too config.txt did the trick. Now Kodi works!

MichaIng, 2023-12-31:
Right, that is the way to enable LCD support with KMS 👍. Alternatively one could add a specific overlay in some cases, but the auto-detection usually just works.

closed 2023-12-31 by MichaIng