MichaIng/DietPi #6829: Services on Dietpi Dashboard not working anymore.

sagecandida, 2023-12-30:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `bookworm 1`
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
- SBC model | `RPi Zero 2 W (armv7l)` or (EG: RPi3)
- Power supply used | 5V3A
- SD card used | Samsung

#### Additional Information (if applicable)
- Software title | Dietpi Dashboard
- Was the software title installed freshly or updated/migrated? freshly installed
- Can this issue be replicated on a fresh installation of DietPi? didn't try it

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

-  Services on Dietpi Dashboard not working anymore.

#### Extra details

- ...
[image]

MichaIng, 2023-12-30:
Can you check the service logs?
```sh
journalctl -u dietpi-dashboard
```

sagecandida, 2023-12-30:
> Can you check the service logs?
> 
> ```shell
> journalctl -u dietpi-dashboard
> ```
```

Dec 30 21:42:12 DietPi systemd[1]: Started dietpi-dashboard.service - Web Dashboard (DietPi).
Dec 30 21:42:19 DietPi dietpi-dashboard[397]:    6.568805831s  INFO request{remote_addr=[::ffff:192.168.1.223]:50076}: dietpi_dashboard: Request to /ws
Dec 30 21:42:21 DietPi dietpi-dashboard[397]:    8.704511091s  WARN request{remote_addr=[::ffff:192.168.1.223]:50076}:socket_handler:service_handler: dietpi_dashboard::page_handlers: Couldn't get failed service name
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    8.971355674s  INFO request{remote_addr=[::ffff:192.168.1.223]:59570}: dietpi_dashboard: Request to /service
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.012932132s  INFO request{remote_addr=[::ffff:192.168.1.223]:59570}: dietpi_dashboard: Request to /assets/index.49a5df62.js
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.025539841s  INFO request{remote_addr=[::ffff:192.168.1.223]:59580}: dietpi_dashboard: Request to /assets/index.b578e7cd.css
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.026247080s  INFO request{remote_addr=[::ffff:192.168.1.223]:59590}: dietpi_dashboard: Request to /assets/xterm.bd908267.js
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.089983695s  INFO request{remote_addr=[::ffff:192.168.1.223]:59604}: dietpi_dashboard: Request to /ws
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.096219320s  INFO request{remote_addr=[::ffff:192.168.1.223]:59590}: dietpi_dashboard: Request to /assets/dietpi.4cc58251.png
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.096732810s  INFO request{remote_addr=[::ffff:192.168.1.223]:59570}: dietpi_dashboard: Request to /assets/github-mark.31287333.svg
Dec 30 21:42:22 DietPi dietpi-dashboard[397]:    9.113915778s  INFO request{remote_addr=[::ffff:192.168.1.223]:59570}: dietpi_dashboard: Request to /favicon.png
Dec 30 21:42:23 DietPi dietpi-dashboard[397]:   10.668739059s  WARN request{remote_addr=[::ffff:192.168.1.223]:59604}:socket_handler:service_handler: dietpi_dashboard::page_handlers: Couldn't get failed service name

```

MichaIng, 2023-12-30:
```
socket_handler:service_handler: dietpi_dashboard::page_handlers: Couldn't get failed service name
```
Hmm it fails here: https://github.com/ravenclaw900/DietPi-Dashboard/blob/v0.6.2/src/page_handlers.rs#L245-L250
In this part of the code, not much changed since last release, but could you still test it with the nightly (if not done already):
```sh
dietpi-software reinstall 200
systemctl restart dietpi-dashboard
```
and when asked, select nightly build.

sagecandida, 2023-12-30:
> ```
> socket_handler:service_handler: dietpi_dashboard::page_handlers: Couldn't get failed service name
> ```
> 
> Hmm it fails here: https://github.com/ravenclaw900/DietPi-Dashboard/blob/v0.6.2/src/page_handlers.rs#L245-L250 In this part of the code, not much changed since last release, but could you still test it with the nightly (if not done already):
> 
> ```shell
> dietpi-software reinstall 200
> systemctl restart dietpi-dashboard
> ```
> 
> and when asked, select nightly build.

I just did it but still not working

[image]

MichaIng, 2023-12-30:
I am not vised enough to understand why the code could fail. Pinging @ravenclaw900, probably you have an idea.

It seems to fail on a failed service (lol). Let's check which one this could be. Anyway good to check which other service is failing why:
```sh
systemctl | grep failed
```

sagecandida, 2023-12-30:
> I am not vised enough to understand why the code could fail. Pinging @ravenclaw900, probably you have an idea.
> 
> It seems to fail on a failed service (lol). Let's check which one this could be. Anyway good to check which other service is failing why:
> 
> ```shell
> systemctl | grep failed
> ```

`● isc-dhcp-server.service    loaded failed     failed          LSB: DHCP server`

MichaIng, 2023-12-30:
Do you run a WiFi hotspot or something similar so that you need a DHCP server?

Try to stop and disable it, and check whether this fixes the dashboard service list:
```sh
systemctl disable --now isc-dhcp-serer
```

To check why it actually failed:
```sh
journalctl -u isc-dhcp-server
```

sagecandida, 2023-12-30:
> Do you run a WiFi hotspot or something similar so that you need a DHCP server?
> 
> Try to stop and disable it, and check whether this fixes the dashboard service list:
> 
> ```shell
> systemctl disable --now isc-dhcp-serer
> ```
> 
> To check why it actually failed:
> 
> ```shell
> journalctl -u isc-dhcp-server
> ```

I installed the WiFi Hotspot initially but later removed it. However, an unexpected conflict arose, leading to the services not appearing on the DietPi dashboard. Fortunately, after disabling it, everything is functioning properly again. Thank you for your assistance!

closed 2023-12-30 by sagecandida

MichaIng, 2023-12-30:
Okay, you could then also remove the DHCP server:
```sh
apt autopurge isc-dhcp-server
```
We still need to find out why the dashboard fails to list services if one of them failed, or whether this particular one is/was a special case. But can do that on the dashboard repo next year, when I find time to test/replicate.

ravenclaw900, 2023-12-31:
If you haven't already removed the DHCP server, the output of
```
dietpi-services list
```
would be extremely helpful.

sagecandida, 2024-01-02:
> If you haven't already removed the DHCP server, the output of
> 
> ```
> dietpi-services list
> ```
> 
> would be extremely helpful.

unfortunately i already removed the DHCP server