MichaIng/DietPi #6831: DietPi becomes false LAN IP after restart

sagecandida, 2024-01-01:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `bookworm 1`
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- SBC model | `RPi Zero 2 W (armv7l)` or (EG: RPi3)
- Power supply used | 5V3A
- SD card used | Samsung

#### Additional Information (if applicable)
- Software title | Dietpi Dashboard
- Was the software title installed freshly or updated/migrated? freshly installed
- Can this issue be replicated on a fresh installation of DietPi? didn't try it

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

-  After rebooting DietPi, the LAN IP address of WireGuard occasionally takes precedence over the static IP address assigned to Ethernet. Is there a solution to prevent this issue?

#### Extra details

- I connected my monitor to see the problem because i couldn't connect to my dietpi through ssh
[image]

sagecandida, 2024-01-01:
> Are you able to connect a keyboard to login locally? Probably your system did not get a DHCP address assigned at all

yes i can

Joulinar, 2024-01-01:
Once logged in, can you share

`ip a`

sagecandida, 2024-01-01:
> Once logged in, can you share
> 
> `ip a`

[image]

Joulinar, 2024-01-01:
as you can see, no IP address has been assigned to your `eth0` interface and it is in state `DOWN`. Can you share following

`journalctl -u ifup@eth0.service`

sagecandida, 2024-01-01:
> as you can see, no IP address has been assigned to your `eth0` interface and it is in state `DOWN`. Can you share following
> 
> `journalctl -u ifup@eth0.service`

[image]

Joulinar, 2024-01-01:
reboot your system and check again. There should be something. Otherwise it looks Ethernet is not recognized. Check cabling aso.

sagecandida, 2024-01-01:
> reboot your system and check again. There should be something. Otherwise it looks Ethernet is not recognized. Check cabling aso.

I rebooted it several times and didn't work. I just format the sd card and installed the dietpi again and everything worked again but after i restore my backup, same problem. I broked something somehow but idk

sagecandida, 2024-01-01:
i just checked that php8.2-fpm failed to start and i can't enable it again

Joulinar, 2024-01-01:
you should fix your network issue first before taking care on other issues. Can you share following on a system that has been restored?

`cat /etc/network/interfaces`

sagecandida, 2024-01-02:
> you should fix your network issue first before taking care on other issues. Can you share following on a system that has been restored?
> 
> `cat /etc/network/interfaces`

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.1.22
netmask 255.255.255.0
gateway 192.168.1.1
#dns-nameservers 192.168.1.1

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 192.168.1.1
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

sagecandida, 2024-01-02:
i just figured out that it becomes sometimes the ip adress of wireguard. I just delete the pivpn and all the wireguard configurations and reboot several times and works just fine. But i would like to use wireguard. Is there any chance to prevent dietpi becomes the IP adress from the wireguard? static ip adress is already set.

Joulinar, 2024-01-02:
There is something wrong on your setup. Personally I use Wireguard since years without issues. As well we have a lot of users running it. How're you are the first who has this issue. Strange is, your Ethernet network interface seems to be down. Wireguard IP is shown only, because there is no Ethernet address assigned. 

You could try following if hit by the issue

```
ifup eth0
```

sagecandida, 2024-01-02:
> There is something wrong on your setup. Personally I use Wireguard since years without issues. As well we have a lot of users running it. However you are the first who has this issue. Strange is, your Ethernet network interface seems to be down. Wireguard IP is shown only, because there is no Ethernet address assigned.
> 
> You could try following if hit by the issue
> 
> ```
> ifup eth0
> ```

idk what caused the internet problem but at the end i just installed the dietpi new and configured everything again now internet is working but then i find out wireguard causing the problem when i reboot dietpi. 

```
root@DietPi:~# ifup eth0
ifup: interface eth0 already configured
```

sagecandida, 2024-01-02:
is there any option that i can only use eth0 as a network interface so when i install wireguard again that the dietpi becomes its LAN IP from eth0? 

```
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:e0:4c:68:05:a1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.22/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd00::2e0:4cff:fe68:5a1/64 scope global dynamic mngtmpaddr 
       valid_lft 6845sec preferred_lft 3245sec
    inet6 2001:9e8:e8b:e600:2e0:4cff:fe68:5a1/64 scope global dynamic mngtmpaddr 
       valid_lft 6845sec preferred_lft 3245sec
    inet6 fe80::2e0:4cff:fe68:5a1/64 scope link 
       valid_lft forever preferred_lft forever

```

Joulinar, 2024-01-02:
I guess you misunderstood the issue. Wireguard is not assisting an IP address to the `eth0` interface. Because Wireguard is managing `wg0` interface only. Your problem is, that sometimes your `eth0` interface is not getting an IP address assigned at all. And this is the real issue than. As a side effect you see the information from `wg0` within login screen because there is no `eth0` interface information to be displayed.

sagecandida, 2024-01-02:
> I guess you misunderstood the issue. Wireguard is not assisting an IP address to the `eth0` interface. Because Wireguard is managing `wg0` interface only. Your problem is, that sometimes your `eth0` [...]

Is there any way to prevent it?

Joulinar, 2024-01-02:
we need to find out why your system is not able to start the Ethernet interface. There it would be important to check as soon as the issue happen.

```
journalctl -u ifup@eth0.service
cat /etc/network/interfaces
ip a
ifup eth0
```

closed 2025-05-31 by MichaIng