MichaIng/DietPi #6832: What program is using 8080?

RRIoT, 2024-01-01:
Hi.

What program is using 8080?

When you remote into ip address of Dietpi, you get  "404 page not found"
I have an irrigation program that I would like to use that uses 8080

Info is:

- DietPi Version:

root@RR-DEITPI-IRRIGATION:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro Version:

root@RR-DEITPI-IRRIGATION:~# echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0

- Kernel version:

root@RR-DEITPI-IRRIGATION:~# uname -a
Linux RR-DEITPI-IRRIGATION 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model:

root@RR-DEITPI-IRRIGATION:~# echo $G_HW_MODEL_NAME
RPi 3 Model B+ (aarch64)

- Power supply used:

RS Components Ltd
+5.1v  2.5A

- SD card used:

Samsung 128G

Additional Information:
Software installed so far on fresh install

200
76
0
1
110
5
6
170
88
90
152
8
9
17
34
130
150
188
189
196

root@RR-DEITPI-IRRIGATION:~# dietpi-bugreport
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: a3f1e73a-ca9e-477d-bcfd-e5c5007cfe19

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

root@RR-DEITPI-IRRIGATION:~# echo $G_HW_UUID
a3f1e73a-ca9e-477d-bcfd-e5c5007cfe19

Irrigation Software to install is https://github.com/Dan-in-CA/SIP/

Love this OS.
Only OS with Node-Red working correctly!
Thank you very much
Have a great day

Kind Regards
Craig

Joulinar, 2024-01-01:
you can check available ports as follow

`ss -tulpn | grep LISTEN`

RRIoT, 2024-01-01:
Hi Joulinar

Thank you so much for the quick reply.

Results are below

***@***.***:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      4096           0.0.0.0:111        0.0.0.0:*
 users:(("rpcbind",pid=320,fd=4),("systemd",pid=1,fd=33))
tcp   LISTEN 0      80           127.0.0.1:3306       0.0.0.0:*
 users:(("mariadbd",pid=492,fd=21))
tcp   LISTEN 0      4096           0.0.0.0:41007      0.0.0.0:*
 users:(("rpc.mountd",pid=442,fd=9))
tcp   LISTEN 0      64             0.0.0.0:2049       0.0.0.0:*

tcp   LISTEN 0      128            0.0.0.0:22         0.0.0.0:*
 users:(("sshd",pid=465,fd=3))
tcp   LISTEN 0      50             0.0.0.0:139        0.0.0.0:*
 users:(("smbd",pid=508,fd=49))
tcp   LISTEN 0      4096           0.0.0.0:51559      0.0.0.0:*
 users:(("rpc.mountd",pid=442,fd=17))
tcp   LISTEN 0      100            0.0.0.0:1883       0.0.0.0:*
 users:(("mosquitto",pid=434,fd=4))
tcp   LISTEN 0      50             0.0.0.0:445        0.0.0.0:*
 users:(("smbd",pid=508,fd=48))
tcp   LISTEN 0      4096           0.0.0.0:39321      0.0.0.0:*
 users:(("rpc.mountd",pid=442,fd=13))
tcp   LISTEN 0      64             0.0.0.0:33173      0.0.0.0:*

tcp   LISTEN 0      4096              [::]:111           [::]:*
 users:(("rpcbind",pid=320,fd=6),("systemd",pid=1,fd=35))
tcp   LISTEN 0      64                [::]:39543         [::]:*

tcp   LISTEN 0      4096              [::]:37465         [::]:*
 users:(("rpc.mountd",pid=442,fd=11))
tcp   LISTEN 0      511                  *:80               *:*
 users:(("apache2",pid=639,fd=4),("apache2",pid=638,fd=4))
tcp   LISTEN 0      4096              [::]:55845         [::]:*
 users:(("rpc.mountd",pid=442,fd=19))
tcp   LISTEN 0      64                [::]:2049          [::]:*

tcp   LISTEN 0      128               [::]:22            [::]:*
 users:(("sshd",pid=465,fd=4))
tcp   LISTEN 0      50                [::]:139           [::]:*
 users:(("smbd",pid=508,fd=47))
tcp   LISTEN 0      1024                 *:5252             *:*
 users:(("dietpi-dashboar",pid=430,fd=9))
tcp   LISTEN 0      100               [::]:1883          [::]:*
 users:(("mosquitto",pid=434,fd=5))
tcp   LISTEN 0      4096              [::]:40445         [::]:*
 users:(("rpc.mountd",pid=442,fd=15))
tcp   LISTEN 0      50                [::]:445           [::]:*
 users:(("smbd",pid=508,fd=46))
***@***.***:~#

Hope you are having a great day

Kind Regards
Craig
__________________________________

Craig Riddle

Riddle's Ranch.

Advanced IoT NZ (Internet of Things).
(Advanced Weather Stations, Raspberry Pi's, ESP32's)

Pivot Service Technician
Rainer Irrigation Limited
Ashburton

103 Albert Street
Elgin
Ashburton 7700
New Zealand

C. 027 7190715
E. ***@***.***
L. https://www.linkedin.com/in/craig-riddle-5756ba47

Joulinar, 2024-01-01:
If possible, try to answer directly on GitHub with such replays. Answering per mail did not allow cool formation of your output :wink:

Ok looks like there is nothing running on port 8080. Means it should be free to use. If you just enter IP address., your browser will use port 80 by default and you will reach your Apache2 web server on this.

closed 2024-07-11 by MichaIng