MichaIng/DietPi #6833: rpimonitor not starting automatically

schlomo, 2024-01-02:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
buster 1
```
- Kernel version | `uname -a`
```
Linux emby 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi 4 Model B (armv7l)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
5V 2A
```

#### Additional Information (if applicable)
- Software title | **rpimonitor**
- Was the software title installed freshly or updated/migrated? **updated**
- Can this issue be replicated on a fresh installation of DietPi? **don't have other system to test, sorry**

#### Steps to reproduce
1. didn't update DietPi for several months
2. ran `dietpi-update`
3. `reboot`

#### Expected behaviour

rpimonitor should auto-start upon booting

#### Actual behaviour

rpimonitor doesn't start after reboot

#### Extra details

Seems like auto start is not supported?

[image]

[image]

Also enabling it via `dietpi-services` didn't help:

[image]

#### Potential fix

I "fixed" this problem by creating a `/etc/systemd/system/rpimonitor.service.d/override.conf` file with this content:

```
[Install]
WantedBy=multi-user.target
```

Joulinar, 2024-01-02:
> buster 1

Your system is extremely old as you are running Debian Buster still. From our end we will drop Buster support on next DietPi release v9. You should consider upgrading to Debian Bullseye at least. https://dietpi.com/blog/?p=811

As well we did not change anything on this software title since a year at least.

Next, RPIMonitor has software has not been updated since ages as the project stopped somewhere 2017. https://github.com/XavierBerger/RPi-Monitor-deb/issues/25 Probably we should remove the title from our list. 

Personally, I tested RPIMonitor on my RPi3B+ running Debian Bookworm without issues. Services is starting after reboot.

```
root@DietPi3:~# systemctl status rpimonitor
● rpimonitor.service - LSB: RPi-Monitor daemon
     Loaded: loaded (/etc/init.d/rpimonitor; generated)
     Active: active (running) since Tue 2024-01-02 20:18:04 CET; 1min 47s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 290 ExecStart=/etc/init.d/rpimonitor start (code=exited, status=0/SUCCESS)
      Tasks: 3 (limit: 2176)
        CPU: 1.160s
     CGroup: /system.slice/rpimonitor.service
             ├─392 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log
             ├─393 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log
             └─394 /usr/bin/perl /usr/bin/rpimonitord -b /var/run/rpimonitord.pid -l /var/log/rpimonitor.log

Jan 02 20:18:02 DietPi3 systemd[1]: Starting rpimonitor.service - LSB: RPi-Monitor daemon...
Jan 02 20:18:04 DietPi3 rpimonitor[290]: Starting RPi-Monitor:.
Jan 02 20:18:04 DietPi3 systemd[1]: Started rpimonitor.service - LSB: RPi-Monitor daemon.
root@DietPi3:~#
```

schlomo, 2024-01-03:
Thanks @Joulinar I'll upgrade and report back. Very good point indeed, and high time for me to do so.

MichaIng, 2025-05-31:
Closing this issue. Feel free to reopen if the issue exists on recent image/Debian version.

closed 2025-05-31 by MichaIng