MichaIng/DietPi #6834: Add required eBPF kernel configurations

kbrighton, 2024-01-02:
I've been attempting to run the cilium CNI plugin (as well as the Calico CNI plugin) via rke2, and it's not been working correctly, due to missing eBPF kernel configurations:

The requirements for these CNI plugins are:
CONFIG_BPF=y
CONFIG_BPF_SYSCALL=y
CONFIG_NET_CLS_BPF=y
CONFIG_BPF_JIT=y
CONFIG_NET_CLS_ACT=y
CONFIG_NET_SCH_INGRESS=y
CONFIG_CRYPTO_SHA1=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CGROUPS=y
CONFIG_CGROUP_BPF=y
CONFIG_PERF_EVENTS=y
CONFIG_SCHEDSTATS=y

We seem to be missing:
CONFIG_NET_CLS_BPF=y
CONFIG_CRYPTO_USER_API_HASH=y
CONFIG_CGROUPS=y
CONFIG_PERF_EVENTS=y
CONFIG_SCHEDSTATS=y

If those could be added to the quartz64_defconfig file, at the very least, it would be very helpful.

Reference here: https://docs.cilium.io/en/stable/operations/system_requirements/

MichaIng, 2024-01-19:
Which SBC are you referring to? We control the kernel configuration for Quartz64, VisionFive 2 and Star64 only. For all other SBC/PC images we use externally managed kernel packages, from Debian, Armbian or the respective vendors.

kbrighton, 2024-01-19:
It would be the quartz64, I believe.

MichaIng, 2024-06-15:
Does this work on Raspberry Pi (RPi OS)? It misses `CONFIG_NET_CLS_BPF` as well. Debian and Armbian kernels however ships with it, hence I think it is common enough for us to add it as well.

MichaIng, 2024-06-15:
Done with: https://github.com/MichaIng/DietPi/commit/8d572699
Kernel builds running: https://github.com/MichaIng/DietPi/actions/runs/9530987096
I'll do the same for VisionFive 2 and Star64.

MichaIng, 2024-06-15:
All done, please try:
```sh
apt update
apt upgrade
reboot
```

closed 2024-07-02 by MichaIng