MichaIng/DietPi #6835: diet-update

GHTho, 2024-01-03:
#### Details:
- Date           | Wed Jan  3 11:25:58 CST 2024
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. diet-update
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Ign:1 https://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.6 InRelease
```

Joulinar, 2024-01-03:
You somehow manually added a 3rd party repository for MongoDB Ubuntu to your system. You might need to update your source file.

GHTho, 2024-01-03:
Ah, Unifi controller

Joulinar, 2024-01-03:
you need to check the whole error message if your run `apt update`. This should indicate what's wrong.

GHTho, 2024-01-03:
I will thx; weird; been running this for over a year without a conflict
Thx for you help

Joulinar, 2024-01-03:
Maybe the repository has changed or removed. MongoDB v3.6 is extremely old.

Joulinar, 2024-01-03:
don't use latest 4.4 as they stop supporting ARM somewhere in the release. https://dietpi.com/forum/t/how-to-run-the-unifi-console-on-a-dietpi-raspi/18552/8?u=joulinar

68wooley, 2024-01-04:
You can use MongoDB 4.4 up to release 18 on Pi 4. Everything after that requires the ARMv8.2-A microarchitecture. Pi 5 will run the latest releases.

Discussion here:

https://www.mongodb.com/community/forums/t/mongodb-community-6-0-5-illegal-instruction-core-dumped-ubuntu-18-04-on-cortex-a72-aarch64/223970