MichaIng/DietPi #6845: upgrade php failed

Drajcoshi, 2024-01-09:
#### Details:
- Date           | Tue Jan  9 11:45:31 GMT 2024
- DietPi version | v8.25.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux pi4B-Dietpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install php8.2-fpm php8.2-apcu php8.2-curl php8.2-gd php8.2-mbstring php8.2-xml php8.2-zip php8.2-mysql php8.2-redis`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 php8.2-common : Depends: libffi6 (>= 3.2) but it is not installable
E: Unable to correct problems, you have held broken packages.

Joulinar, 2024-01-09:
> The following packages have unmet dependencies:
> php8.2-common : Depends: libffi6 (>= 3.2) but it is not installable

This sound strange because `php8.2-common` depends on `libffi8` and not `libffi6` https://packages.debian.org/bookworm/php8.2-common

MichaIng, 2024-01-12:
Probably the Bullseye branch of Ondrej's PHP repo is installed? Cam you check:
```sh
for i in /etc/apt/sources.list.d/*.list; do echo "$i:"; cat "$i"; done
```