MichaIng/DietPi #6846: shairport-sync installation is broken

keesma, 2024-01-10:
Dietpi issue

#### Details:
- Date           | Wed Jan 10 13:27:13 CET 2024
- DietPi version | v8.24.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux werkkamer 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -y --allow-change-held-packages install ./package.deb`
- Exit code      | 100
- Software title | DietPi-Patch

#### Steps to reproduce:

1. Dietpi-update
2. Wait until error

#### Expected behaviour:

- It should finish without errors

#### Actual behaviour:

- See error below

#### Extra details:

- I installed a newer version of shairport some time ago, after the previous diet update

-  I tried to reinstall shairport but that did not resolve the issue

#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 shairport-sync : Depends: libglib2.0-0 (>= 2.58.3-2+deb10u5) but 2.58.3-2+deb10u4 is to be installed
E: Unable to correct problems, you have held broken packages.
```
I

keesma, 2024-01-10:
I solved it with:

sudo dpkg --configure -a
sudo apt-get install -f 
apt --fix-broken install

and a reboot

Joulinar, 2024-01-10:
hmm you are running quite old Debian Buster system, which we will drop support on upcoming release. 

Anyway, maybe we need to check dependency as we build the package ourselves. 

@MichaIng I tested on ARMv6 RPi image on my RPi3B+ and it seems RPi repository is behind Debian repository

Bookwoorm

```
The following packages have unmet dependencies:
 shairport-sync : Depends: libglib2.0-0 (>= 2.74.6-2) but it is not going to be installed
```

```
root@DietPi3:~# apt show libglib2.0-0
Package: libglib2.0-0
Version: 2.74.6-1
```

On Bullseye it seems to be fine. For Buster I don't have a test system anymore.

MichaIng, 2024-01-11:
We can test this nicely here 😉: https://github.com/MichaIng/DietPi/actions/runs/7493666412

I wonder whether we should just remove the dependencies' version requirements for our packages. In theory, library versions need to match those the package was compiled against, but practically Debian very rarely ships breaking changes on a stable version, and the other way round we regularly face(d) issues between Debian and Raspbian with the versions added. We do already remove `+bN`, `+rptN` and `+rpiN` suffixes for this reason, but as can be seen, we'd need to remove `debNuN` and even `-N` as well, and still won't be sure to catch all cases.

The alternative would be to strictly install only ARMv6 packages on Raspbian, so the versions are assured to be available, since we compile ARMv6 packages on Raspbian containers (of course). But the benefit of smaller binaries and others thanks to NEON would be lost on ARMv7 capable RPi models.

In a perfect world, we would define exactly those minimum versions which really are required based on the library functions used and when breaking changes affected them, but this is something we can impossibly maintain.

MichaIng, 2024-01-11:
EDIT: Ah dammit, not so easy to test on GitHub, since these are either Raspbian containers with ARMv6 arch hardcoded (hence the ARMv7 packages are not installed) or ARMv7 arch on Debian, where the latest library versions are available 🙈. At least we can see in the logs which library version was installed. I'll patch the packages on the server for now.

MichaIng, 2024-01-11:
Should be fixed for Buster and Bookworm (no issue on Bullseye), please try again.

closed 2024-01-16 by MichaIng