MichaIng/DietPi #6848: mjpg-streamer broken ?

supertevran, 2024-01-11:
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
bookworm 1
Linux Therme 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
RPi B (armv6l)
944f7361-e48c-4480-9b04-6f8c96ec6d09

Fresh Install. Pi Camera Module.
vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0
/dev/video0:
 HDIO_GET_IDENTITY failed: Inappropriate ioctl for device

Unfortunately not working. Thx. for any help.

MichaIng, 2024-01-13:
Indeed, with Bookworm, the legacy camera interface has been removed. See also here: https://dietpi.com/forum/t/mjpg-streamer-on-rpi0/18843

Can you try this:
```sh
G_AGP libraspberrypi-dev
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d
G_CONFIG_INJECT 'camera_auto_detect=' 'camera_auto_detect=1' /boot/config.txt
sed -i -e '/libraspberrypi-dev/d' -e '/input_raspicam.so/d' /boot/dietpi/dietpi-software
dietpi-software reinstall 137
```
This enables the modern KMS video and camera stack and compiles mjpg-streamer without the legacy RPi camera module support. Instead it will try to grab the stream from `/dev/video0` with the normal UVC module. Let's see whether this works.

supertevran, 2024-01-13:
Hi.Thx. for your reply. Unfortunately no change....

MichaIng, 2024-01-13:
But `vcgencmd get_camera` should show now:
```sh
supported=1 detected=1, libcamera interfaces=1
```
(1 at the end), isn't it?

Generally to test the functionality of the camera with the modern driver stack, you can use this package and command:
```sh
apt install rpicam-apps-lite
rpicam-hello
```

supertevran, 2024-01-14:
vcgencmd get_camera
supported=1 detected=0, libcamera interfaces=1

`[15:07:33.786468000] [6565]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[15:07:36.212857000] [6570]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[15:07:36.238956000] [6570]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c@7e804000/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
[15:07:36.241038000] [6570]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[15:07:36.393557000] [6565]  INFO Camera camera.cpp:1183 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[15:07:36.397122000] [6570]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c@7e804000/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
#0 (0.00 fps) exp 33222.00 ag 5.50 dg 1.00
#1 (30.03 fps) exp 33222.00 ag 5.50 dg 1.00
#2 (30.01 fps) exp 33222.00 ag 5.88 dg 1.00
#3 (30.01 fps) exp 33222.00 ag 5.38 dg 1.00
#4 (30.02 fps) exp 33222.00 ag 5.38 dg 1.00
#5 (15.00 fps) exp 33222.00 ag 5.25 dg 1.02
#6 (30.03 fps) exp 33222.00 ag 5.25 dg 1.02
#7 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#8 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#9 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#10 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#11 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#12 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#13 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#14 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#15 (15.00 fps) exp 33222.00 ag 5.25 dg 1.02
#16 (30.04 fps) exp 33222.00 ag 5.25 dg 1.02
#17 (30.00 fps) exp 33222.00 ag 5.25 dg 1.02
#18 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#19 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#20 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#21 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#22 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#23 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#24 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#25 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#26 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#27 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#28 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#29 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#30 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#31 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#32 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#33 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#34 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#35 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#36 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#37 (30.03 fps) exp 33222.00 ag 5.25 dg 1.02
#38 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#39 (29.95 fps) exp 33222.00 ag 5.25 dg 1.02
#40 (30.07 fps) exp 33222.00 ag 5.25 dg 1.02
#41 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#42 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#43 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#44 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#45 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#46 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#47 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#48 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#49 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#50 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#51 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#52 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#53 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
[... 44 lines cut ...]
#98 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#99 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#100 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#101 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#102 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#103 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#104 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#105 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#106 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#107 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#108 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#109 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#110 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#111 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
#112 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#113 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#114 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#115 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#116 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#117 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#118 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#119 (30.02 fps) exp 33222.00 ag 5.25 dg 1.02
#120 (30.01 fps) exp 33222.00 ag 5.25 dg 1.02
#121 (15.01 fps) exp 33222.00 ag 5.25 dg 1.02
`

MichaIng, 2024-01-14:
Seems to work. I guess you need to run this from within a desktop/X session to have an actual preview window show up, but the frames are collected successfully.

Interesting is this:
```sh
Unicam device /dev/media2 and ISP device /dev/media0
```
We were always looking for `/dev/video*` devices, but probably those `/dev/media*` devices need o be used.

But I was looking around a little more, and it seems that is not gonna work either, at least not without `libcamerify`, which I do not want to spend much time with for now.

But, there is a fork with libcamera support: https://github.com/ArduCAM/mjpg-streamer
Also discussed here: https://forums.raspberrypi.com/viewtopic.php?t=346053
Let me create a PR with this fork and its `input_libcamera` module compiled. This sounds promising.

hellvesper, 2025-01-31:
> Indeed, with Bookworm, the legacy camera interface has been removed. See also here: https://dietpi.com/forum/t/mjpg-streamer-on-rpi0/18843
> 
> Can you try this:
> 
> G_AGP libraspberrypi-dev
> /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d
> G_CONFIG_INJECT 'camera_auto_detect=' 'camera_auto_detect=1' /boot/config.txt
> sed -i -e '/libraspberrypi-dev/d' -e '/input_raspicam.so/d' /boot/dietpi/dietpi-software
> dietpi-software reinstall 137
> This enables the modern KMS video and camera stack and compiles mjpg-streamer without the legacy RPi camera module support. Instead it will try to grab the stream from `/dev/video0` with the normal UVC module. Let's see whether this works.

I think this one is helped to make work `rpicam-apps-lite` which is enough for my Timelapse project. I suggest somehow mark this as temporal workaround. Without that snipped `rpicam-apps-lite` can't find connected camera

hellvesper, 2025-02-03:
Hi, I also have issue with camera on `RPi Zero 2 W`. 

A make fresh install of `DietPi v9.9.0` image for `Raspberry Pi 2/3/4/Zero 2`

Then I've installed `rpicam-apps-lite` from apt. Make powercycle (with proper shutdown). But libcamera didn't see any camera:
```bash
root@DietPi:~# rpicam-hello 
[0:00:21.160702230] [1132]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
Preview window unavailable
ERROR: *** no cameras available ***
```

Then I tried to apply that shell snippet and reboot:
```bash
root@DietPi:~# G_AGP libraspberrypi-dev
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d
G_CONFIG_INJECT 'camera_auto_detect=' 'camera_auto_detect=1' /boot/config.txt
[ INFO ] APT purge libraspberrypi-dev, please wait...
[ INFO ] None of the packages are currently installed. Aborting...
[  OK  ] APT purge libraspberrypi-dev

 DietPi-Set_hardware
─────────────────────────────────────────────────────
 Mode: rpi-opengl (vc4-kms-v3d)

[  OK  ] DietPi-Set_hardware | sed --follow-symlinks -Ei /^[[:blank:]]*dtoverlay=vc4-f?kms-v3d/d /boot/config.txt
[  OK  ] DietPi-Set_hardware | Added setting dtoverlay=vc4-kms-v3d,noaudio to end of file /boot/config.txt
[  OK  ] rpi-opengl vc4-kms-v3d | Completed
[  OK  ] G_CONFIG_INJECT | Added setting camera_auto_detect=1 to end of file /boot/config.txt
```

And got this:
```bash
root@DietPi:~# rpicam-hello 
ERROR: rpicam-apps currently only supports the Raspberry Pi platforms.
Contributions for other platforms are welcome at https://github.com/raspberrypi/rpicam-apps.
```
also `get_camera` command returns zero for all items:
```bash
root@DietPi:~# vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0
```

but device is exist in system:
```bash
root@DietPi:~# ls /dev/video*
/dev/video0
root@DietPi:~# ls /dev/media*
/dev/media0
root@DietPi:~# vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0
```

I don't know how to fix this. 

So tried to check is it working on `Raspberry Pi OS Lite (Bookworm)` to exclude hardware issues and it does:
```bash
vesper@pizero:~ $ rpicam-hello 
[0:01:53.583108145] [679]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:01:53.891556160] [684]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:01:53.901240138] [684]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media0
[0:01:53.901444330] [684]  INFO RPI pipeline_base.cpp:1126 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:01:53.907837235] [679]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-SGBRG10_CSI2P
[0:01:53.909020346] [684]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
#0 (0.00 fps) exp 33222.00 ag 5.38 dg 1.00
#1 (30.01 fps) exp 33222.00 ag 4.12 dg 1.00
#2 (30.01 fps) exp 27859.00 ag 2.00 dg 2.32
#3 (30.01 fps) exp 32317.00 ag 4.00 dg 1.00
#4 (30.01 fps) exp 32270.00 ag 4.00 dg 1.00
#5 (30.01 fps) exp 32247.00 ag 4.00 dg 1.00
```

but strange that for RPi OS `get_camera` also return nothing:
```bash
vesper@pizero:~ $ vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0
```

chri-kai-in, 2025-04-07:
Hello,

not sure whether this is of relevance, but i faced harsh removal of /dev/video0 interface after updating to kernel 6.12.20 with latest dietpi "apt update" command on two RBPi 3 model A. Both devices running motioneye with a local CSI camera attached. Same setup worked flawless until update. I found the following issue in the net describing the exact same symptoms that i faced, with a solution cited for one of the next kernel/firmware releases.

https://github.com/raspberrypi/linux/issues/6753

Best Regards

MichaIng, 2025-04-10:
But this does not restore legacy camera API support, does it? I mean which tool/software/library you use to get the camera module stream?

MichaIng, 2025-04-19:
Interesting, so it really seems MMAL library is missing only, while the kernel still supports it, or probably it is really only the camera module itself which needs to support it, like the module 3 does not.

Generally, `libcamerify` can be used as shim to provide legacy API on top of modern API. There are some issues at the motionEye repo on how to do that. When I find time, I want to implement this sort of natively into motionEye as part of the `motioneye_init` command.

closed 2025-05-31 by MichaIng