MichaIng/DietPi #6850: DietPi update

macraj, 2024-01-11:
#### Details:
- Date           | Thu Jan 11 17:07:35 CET 2024
- DietPi version | v9.-1.0 (MichaIng/dev)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | `Linux DatsPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `rm dev.tar.gz DietPi-dev/dietpi/pre-patch_file DietPi-dev/dietpi/server_version-6`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. sudo dietpi-update
#### Expected behaviour:

- update
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
rm: cannot remove 'DietPi-dev/dietpi/pre-patch_file': No such file or directory
rm: cannot remove 'DietPi-dev/dietpi/server_version-6': No such file or directory
```

Trimble-tech, 2024-01-11:
I had the same issue on a RPi4B; opening a sub-shell in the `/tmp` directory shows the `pre-patch_file`,  `server_version-6`, and the `dev.tar.gz` not being present. Also, when running `DietPi-BugReport` the process was unable to finish after extended wait. 

Below is the details output by the update error:
- Date           | Thu Jan 11 11:29:39 EST 2024
- DietPi version | v9.-1.0 (MichaIng/dev)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Piper2 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `rm dev.tar.gz DietPi-dev/dietpi/pre-patch_file DietPi-dev/dietpi/server_version-6`
- Exit code      | 1
- Software title | DietPi-Update
Based on the details I see here and in the original post, I believe the issue is due to the test images getting upgraded to Bookworm earlier; the updater is looking for files that don't exist due to it. In particular, they are the following in my case:
- `/tmp/DietPi-Update/DietPi-dev/dietpi/pre-patch_file`
- `/tmp/DietPi-Update/DietPi-dev/dietpi/server_version-6`
- `/tmp/Dietpi-Update/dev.tar.gz`

I opened a sub-shell and touched all of these files (`touch filename`) to create them, then the update proceeded as normal. **_So far my system works and reports version 9.0.0 (dev) but I am unsure if this is a reasonable fix for the problem_.**

Joulinar, 2024-01-11:
Guys this is a development branch. Not even in Beta status. But thx for reporting.

MichaIng, 2024-01-11:
Generally, if you are not aware that you were on a (naturally unstable) development branch:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```

I was tinkering with removing the obsolete DietPi v6 files from the repo last night. I thought I had fixed it, worked on my VM finally, but seems it was too late and I was too tired.

> DietPi version | v9.-1.0 (MichaIng/dev)

Hmm, that alone is strange already, it should be `v9.0.-1` 🤔.

MichaIng, 2024-01-11:
Fixed with: https://github.com/MichaIng/DietPi/commit/7badaa5

closed 2024-01-13 by MichaIng