MichaIng/DietPi #6853: Proxmox 8 installation on  Raspberry Pi 4 (with 4 or 8 GB RAM) andd using DietPi as VM in arm64 Architectur

LudgerKuehm, 2024-01-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

VM on Raspberry Pi4 in arm64 Architektur
===========================

I'm using Proxmox 8 on a Raspberry 4 with 8 GB RAM, but Proxmox 8 is ready to run on a Raspberry Pi 4 with 4GB RAM, too.

Proxmox 8 installation on  Raspberry Pi 4 (with 4 or 8 GB RAM)
=======================================
I has used this manual (sorry, it's in german) :
[https://www.bachmann-lan.de/proxmox-8-auf-dem-raspberry-pi-4-installieren/](url)

A Debian (bookworm) or even better a DietPi (bookworm) installation is suitable as a host system.

Limitations:
However, there are a few limitations with Proxmox 8 on the Raspberry Pi 4. As far as I know these are:

- the VMs must be set up as EFI systems.
- No third-party CPU emulation is possible, only the ARM64 architecture of the RPi3 can be used (CPU selection: HOST; 1 socket, 4 cores)
- There are generally no ready-made VMs on the internet, you have to create them yourself
- The same applies to live CD/DVD images for the ARM architecture. So far I have only found a suitable and bootable LiveDVD ISO from Kali Linux.
Creating a DietPi VM:
=============
If you want to use DietPi as a VM, you first have to set up a Debian system from the Debian-Netinst-ARM64 ISO and then use the “dietpi-installer” to convert Dietpi as a "Generic Device".

Problem that can occur:  It can happen that the LAN device "eth0" is lost.

Remedy:
- Find the current LAN device with 'sudo dhclient -v'
- Enter the identified LAN device, e.g. enp0s11, in '/etc/network/interfaces':

allow-hotplug enp0s11
iface enp0s11 inet dhcp
dns-nameservers 9.9.9.9 149.112.112.112

- then reboot

Rename 'enp0s11' to 'eth0' :
============================
- This works best with raspi-config
- add the raspberrypi repo 'https://archive.raspberrypi.org' for bookworm with the correct key and install raspi-config with 'sudo apt install -y raspi-config'
- start raspi-config and use '6 Advanced Oprions' and then use the function 'A2 Network interface Names' and reboot.
- change entries in '/etc/network/interfaces' from 'enp0s11' to 'eth0' and reboot.

Thats all, enjoy ;-) .

LudgerKuehm, 2024-01-12:
You can make good use of a Raspberry Pi 4 with this:
On the Raspberry Pi4 Host: xfce desktop system with vnc-remote; Proxmox 8-host, transmission-daemon, Jellyfin Mediacenter and Komga book archive
additionally 3 VMs at the same time, 1 Dietpi VM with openmediavault, 1 Dietpi VM as a Wordpress server, 1 Dietpi VM as an xfce desktop system.
With active aluminum cooler the Pi is around 40 degrees Celsius

MichaIng, 2024-05-07:
While it makes sense that DietPi can be used as Proxmox host, like every other Debian, what I take from this is that there is a need for ARMv8 DietPi VM images. We have an open issue/request for this: #5637

closed 2024-05-07 by MichaIng

MichaIng, 2024-05-07:
We could add it as QCOW2 image, to be generated a VM manually from, or via installer script: https://github.com/dazeb/proxmox-dietpi-installer

LudgerKuehm, 2024-05-13:
I have made a finished default VM for the aarch64 architecture on the Raspberry Pi 4 or 5 
available for download on Onedrive:
[https://1drv.ms/f/c/23eeaa1af7c10252/EuuB1zLqLFRApucLuaZ-NWIBRaN0FJ-XwzHtb0rPEIALcQ?e=WH2iQ6](url)

for more info read vm_info.txt

ZeroZorro000, 2025-07-27:
Although it is an older post, I got this working on Proxmox 8 on my RPi5.
I had to go into the Bios and add the created VM to the Boot options and put it on top of the Boot order. That and let Dietpi use all of the space on the disk /sda.
Updating/upgrading after that.
Thanks!