MichaIng/DietPi #6854: DietPi-Software | Sonarr: Migrate to v4

MichaIng, 2024-01-12:
Installer: https://raw.githubusercontent.com/Sonarr/Sonarr/develop/distribution/debian/install.sh

While the download page states that the v3 package will be overwritten, the new installer actually does not install a (DEB) package but downloads and extracts an archive. The DEB package and APT repository would just remain. This is pretty unclean and I am hence not sure whether this is the intended way of migration.

We can purge the package and do the migration via `dietpi-update`, but we'd need to make sure no data/configs are lost, hence needs some testing.

miguelpruivo, 2024-01-18:
Looking forward it!

tundrablasen, 2024-02-10:
Off topic: 
Just wanted to thank you all for your dietpi work. Take your time and don’t stress.

MW420BLAZEIT, 2024-04-15:
One day 🙏

tundrablasen, 2024-04-15:
> One day 🙏

Just upgraded to dietpi 9.3. does this mean I can upgrade now easy to sonarr v4?

Joulinar, 2024-04-15:
No

tundrablasen, 2024-05-02:
@MichaIng @Joulinar 
Would it make sense to get in contact with the people over at the sonarr GitHub? https://github.com/Sonarr/Sonarr/issues

MichaIng, 2024-05-02:
Not needed. I actually know what to do, just need to find the time to actually test/verify and implement it.

tonystower, 2024-07-06:
Hello Michael. Is there any update on this?
I'm run into some difficulties with v3, which apparently are dealt with in v4, so it would be very useful for me.

Thanks v much.

tundrablasen, 2024-07-06:
> Hello Michael. Is there any update on this? I'm run into some difficulties with v3, which apparently are dealt with in v4, so it would be very useful for me.
> 
> Thanks v much.

Same for me, would also be interested.

roni-estein, 2024-08-20:
@MichaIng Following the threads brought me here, is there anything any of us can do to help that is actually useful to you? if you had instructions you wanted some people to run through and test on various platforms or chipsets and upload logs? You are maintaining this for tons of people, we can chip in and help.

tundrablasen, 2024-08-20:
Yes, I’m also volunteering. This issue becomes now very urgent since sonarr v3 has been deprecated since November 2023 and doesn’t get any (security) fixes. 

Also documentation for v3 has been taken offline.

roni-estein, 2024-10-19:
I'd be happy with a workable instruction list we can then test out and post results to help people who are coming to this. It seems like there is clearly too much work for just a small group to maintain. For those of us who are not specifically network or programming guys on linux but know just enough to be helpful, and are willing with a spare pi, a good write up would go a long way. You should be letting the community help where possible even though we may not be able to just fork dietpi and make send back a working PR we could probably be more helpful than we are.

MichaIng, 2024-10-19:
The migration should be easier than a parallel implantation. I was just to busy the last weeks to catch up.

tonystower, 2024-10-19:
Thank you @MichaIng we appreciate all the work you to develop DietPi!

tundrablasen, 2024-10-19:
First of all thanks @MichaIng for maintaining this software. You do it for free and we don’t really have anything to say or request. 

With that being said it wanted to just share my thoughts:
I think a big part of people setting up a SBC is for the purpose of a homeserver with the *arr stack. 

I would assume sonarr/radarr etc is one of the most installed softwares on dietpi. 

I see now on Reddit/discord/youtube people actively lobbying against dietpi because of the sonarr v3 issue. The recommend against dietpi because of that. 

I think this issue should get a higher priority because of that reason. 

Thanks again for maintaining dietpi.

Joulinar, 2024-10-19:
well everybody is free to install Sonarr v4 on it's own. Same as you would do on every other SBC. Quite a strange argument just because our installer is not support v4 yet. How would people install it on other systems?

And just to get an idea on what people DietPi using for 😉 https://dietpi.com/survey/#software

tundrablasen, 2024-10-19:
> well everybody is free to install Sonarr v4 on it's own. Same as you would do on every other SBC. Quite a strange argument just because our installer is not support v4 yet. How would people install [...]

I disagree with this argument. Why do you even then provide dietpi-Software with a GUI and one-click install?

majority of the users have no experience with Debian or the terminal. That’s why dietpi is so attractive. Have an automated service/software repo which can be installed just with clicks.

as I said. I have no right to request anything. Just mirroring feedback from homeserver community

Joulinar, 2024-10-19:
well I was following this argument

> I see now on Reddit/discord/youtube people actively lobbying against dietpi because of the sonarr v3 issue. They recommend against dietpi because of that.

Soo if people `recommend against dietpi`, how would they install Sonarr on a different platform?

tundrablasen, 2024-10-19:
> well I was following this argument
> 
> > I see now on Reddit/discord/youtube people actively lobbying against dietpi because of the sonarr v3 issue. They recommend against dietpi because of that.
> 
> Soo if people `recommend against dietpi`, how would they install Sonarr on a different platform?

I see recommendations to use proxmox with tt helper scripts
https://tteck.github.io/Proxmox/

Or using AppStore of casaOS (yes I know casaOS can run on dietpi, but it’s not mentioned)

Openmediavault

I just want to share my subjective observations. Still love dietpi but also want to keep my software secure and up to date

Joulinar, 2024-10-19:
CasaOS is nothing else than a Docker frontend. You can get a Sonarr Docker container up on DietPi in 1 minute.
Proxmox is quite on overkill for an SBC, same as OMV.

And if someone really like to switch and test, Sonarr offers a migration script that can be used with minor adjustments.

I did a very quite test and it seems to be working

```
root@DietPiRS6:~# systemctl status sonarr.service
● sonarr.service - Sonarr Daemon
     Loaded: loaded (/etc/systemd/system/sonarr.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-10-19 13:06:33 CEST; 51s ago
   Main PID: 11021 (Sonarr)
      Tasks: 23 (limit: 201)
     Memory: 89.8M
        CPU: 13.853s
     CGroup: /system.slice/sonarr.service
             └─11021 /opt/Sonarr/Sonarr -nobrowser -data=/mnt/dietpi_userdata/sonarr/
```

roni-estein, 2024-10-19:
> And if someone really like to switch and test, Sonarr offers a migration script that can be used with minor adjustments.
>
> I did a very quite test and it seems to be working
>
> ```
> root@DietPiRS6:~# [...]

That's Awesome! Can you put down what the minor adjustments were? Even a gist of the updated script! I'm positive that would solve 100% of the concerns people have and then this becomes an very minor issue. Again, I'm only asking how we can help.  Not just on here, but in general. If there was a list of things to try I'd be happy to participate in that too.

Joulinar, 2024-10-19:
Without any guarantee https://dietpi.com/forum/t/sonarr-version-out-of-date/21729/12?u=joulinar

roni-estein, 2024-10-19:
Perfection!!

SjoerdV, 2024-10-20:
> Without any guarantee https://dietpi.com/forum/t/sonarr-version-out-of-date/21729/12?u=joulinar

Excellent step-by-step! The only thing *for my setup* is that I skipped the purging of the old sonarr (`G_ADP sonarr`) and just ran `apt remove sonarr` as this will leave current configs be.

tundrablasen, 2024-10-29:
> Without any guarantee https://dietpi.com/forum/t/sonarr-version-out-of-date/21729/12?u=joulinar

Hey @Joulinar 
thank you for this. Does this mean we might see progress for sonarr v4 within the dietpi updater soon?

Good start in the day to you all

tonystower, 2024-10-29:
> Without any guarantee https://dietpi.com/forum/t/sonarr-version-out-of-date/21729/12?u=joulinar

Thank you very much for your help!!

MichaIng, 2024-12-12:
The official script for reference, though it basically does nothing really different than our other Sarvarr installs: https://raw.githubusercontent.com/Sonarr/Sonarr/develop/distribution/debian/install.sh

tundrablasen, 2024-12-12:
> The official script for reference, though it basically does nothing really different than our other Sarvarr installs: https://raw.githubusercontent.com/Sonarr/Sonarr/develop/distribution/debian/install.sh

Does this mean in 9.9 there will be an official way to update to V4?

MichaIng, 2024-12-12:
At least I am having a serious look into it now :) .

MichaIng, 2024-12-12:
One question which remains is what we do with ARMv6 support. Sonarr v4 does not support it (as dotnet core does not), so we'd need to keep the v3 repo option for that 🤔.

tundrablasen, 2024-12-12:
> At least I am having a serious look into it now 🙂.

Love it!

MichaIng, 2024-12-15:
Migration implemented: #7307
If you want to test:
```sh
G_DEV_BRANCH dev
```

closed 2024-12-25 by MichaIng