MichaIng/DietPi #6869: Permit weekly backups in dietpi-backup

beesyrupfactory, 2024-01-20:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- For the longest time, I had no idea why my services kept going down consistently at 1:25 AM every day until I realized there was a daily cronjob running as cron.daily. Although I like the idea of having a daily backup, I think I'm in the camp where I would benefit more  from weekly backups at a scheduled time.

#### Describe the solution you'd like:
- When going into dietpi-backup, we can see that there is a Daily backup flag (On/Off). Would it be possible to allow more granularity and allow it to execute on a weekly basis?
- Or, is there an off chance that if the daily backup is disabled, a reminder is provided in the MOTD?

#### Describe alternatives you've considered:
- The only workaround I can think of is to (a) disable daily backups, or (b) keep daily backups enabled, but set it to a time that I'm not expecting to do anything. However, since dietpi-backup stops all services prior to running the backup, it feels a bit risky while I'm hosting certain services that I want to always be available such as homebridge, media servers, webserver, samba server, etc.

#### Additional context
- Nothing at the moment. Please feel free to reach out or provide alternative suggestions, or let me know if this is the wrong approach. I'm all ears!

Joulinar, 2024-01-20:
Theoretically you could exclude services from being stopped by DietPi Backup.

beesyrupfactory, 2024-01-23:
@Joulinar thanks for your response. Is there any documentation on this procedure? If this is the appropriate way to move forward, I would like to employ it rather than stopping daily backups.

Joulinar, 2024-01-23:
`dietpi-services` > select the service in question > adjust `Include/Exclude` to `Exclude` > done

MichaIng, 2024-01-23:
Hmm, I thought there was a request about this already, but probably we only internally talked about this. What we actually thought about is adding a way to not only allow to keep X daily backups, but also X weekly backups. When allowing to choose "0" for the daily backups to keep, then it would be essentially only weekly backups 🤔.

I already have an idea how to implement it, but have more high priority tasks to do. However, essentially my idea was to use the same daily cron job and rotation as well for the weekly backups: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-backup#L136-L154
The suffixes for the backup directories remain indicating the day/number of a backup, so for weekly backups there will be e.g. `data_7`, `data_14`, `data_21`, with none in between (unless >7 daily backups), rotated/increased every day/with every backup created. E.g. if you choose to keep 7 daily and 4 weekly backups, it will do a backup every and:
- Remove the 28th backup, if existent, or use it as basis for the new backup (to reduce file writes)
- Rotate all backups between 8 and 27 one up
- Rotate the 7th backup one up, if there is none between 8 and 14 (for the 2nd week) already, otherwise remove it, or use it as basis for the new backup (to reduce file writes)
- Rotate the youngest 6 backups

_Note that the rotation loop is inverse, i.e. it starts with the oldest backup to rotate._

If 0 daily backups are chosen, and we want to support this, then it would need to skip creating a new backup if there is one already between `data_2` (after rotation) and `data_7`.

Using an otherwise removed backup as basis for the new one could be done like this:
- If a last (in case weekly) backup is about to be removed, move it to `data_tmp` first.
- If a last daily backup is about to be removed, in case replace `data_tmp` with it, to have a newer basis for the new backup, hence a smaller diff and again less writes.
- If finally no new backup is to be created, because only weekly backups are wanted and a new one not yet due, finally remove `data_tmp`.

It is not so trivial to implement, but overall cleaner than having two cron jobs where we'd need to take care that they do not overlap/conflict or doubled backups, like a daily and a weekly at the same time.

tonystower, 2024-07-16:
Thanks Michael - I think your suggestion would work really well and I would love to see daily/weekly/monthly options available as you suggest.

beesyrupfactory, 2024-07-21:
> `dietpi-services` > select the service in question > adjust `Include/Exclude` to `Exclude` > done

@Joulinar - Thank you! This is exactly what I want. I'm happy to have daily backups, but I want to make sure that some of the services I want to keep alive 100% of the time were excluded. This addresses this perfectly.