MichaIng/DietPi #6870: dietpi-update fails on Quartz64 Model A from 8.xx to 9.0x

MrFr33man123, 2024-01-21:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | dietpi-update

#### Details:
- Date           | Sun Jan 21 09:10:39 CET 2024
- Bug report     | 90d9581e-e473-469e-bd34-d3bf20837ba5
- DietPi version | v9.0.0 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Quartz64 (aarch64) (ID=49)
- Kernel version | `Linux tiny-server-01 6.5.8 #1 SMP PREEMPT Sat Oct 21 20:01:59 UTC 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `dpkg -i --force-confdef,confold package.deb`
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:

1. run dietpi-update
2. wait, for above command it fails with error seen below.
#### Expected behaviour:
it should update the uboot firmware on blockdevice/sdcard in my case. /boot is mounted rw at the time of execution.
#### Actual behaviour:
- Error as seen below
#### Extra details:

- i am running dietpi on a quartz64 model A
- Boot is on SDCard all the system is on USB HDD it just boots from sd
- no further modifications to the OS only packages installed vim, nala, docker all other software is running containerd
#### Additional logs:
```
(Reading database ... 20057 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking firmware-quartz64a (6.6.12-dietpi1) over (6.6.12-dietpi1) ...
Setting up firmware-quartz64a (6.6.12-dietpi1) ...
Flashing U-Boot
lsblk: /dev/mmcblk0p1[/boot]: not a block device
ERROR: Unable to detect boot drive, aborting U-Boot flashing
dpkg: error processing package firmware-quartz64a (--install):
 installed firmware-quartz64a package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 firmware-quartz64a
```

MichaIng, 2024-01-21:
Do you have a bind mount of that drive somewhere?
```sh
lsblk
df
```

MrFr33man123, 2024-01-25:
```# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    0  1,8T  0 disk
└─sda1        8:1    0  1,8T  0 part /
mmcblk0     179:0    0 58,9G  0 disk
└─mmcblk0p1 179:1    0 58,9G  0 part /boot
                                     /sdcard
# df
Dateisystem     1K-Blöcke   Benutzt  Verfügbar Verw% Eingehängt auf
/dev/root      1923332952 368010856 1477165212   20% /
devtmpfs          3916684         0    3916684    0% /dev
tmpfs             3936492         0    3936492    0% /dev/shm
tmpfs             1574600     63900    1510700    5% /run
tmpfs                5120         0       5120    0% /run/lock
tmpfs             3936256         0    3936256    0% /tmp
tmpfs               51200         8      51192    1% /var/log
/dev/mmcblk0p1   60806420   1230216   57081416    3% /sdcard
overlay        1923332952 368010856 1477165212   20% /mnt/dietpi_userdata/docker-data/overlay2/e880648b5b6a13a21effad40f11ccdebd768be06338116a9315faad997efd0fc/merged
overlay        1923332952 368010856 1477165212   20% /mnt/dietpi_userdata/docker-data/overlay2/a73ecf477bf0a5132a317edf13e5772a963c47d46d4f48a9f3b6f1f8f8acd680/merged
overlay        1923332952 368010856 1477165212   20% /mnt/dietpi_userdata/docker-data/overlay2/9d92906492da2692b5d8e1730c0b549d06b620396ab6b33df1062a25055dda9a/merged

```
fstab:
#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
UUID=ef4dca92-a747-4854-82c2-c14828b2c08b /sdcard ext4 noatime,lazytime,rw      0 1
UUID=ef4dca92-a747-4854-82c2-c14828b21337 / ext4 noatime,lazytime,rw            0 1
/sdcard/boot            /boot               none defaults,bind                  0 0
192.168.178.50:/smarthome      /mnt/nas    nfs  defaults 0 0

I did move the OS onto the sda and just keept the bottloader on sdcard.

MichaIng, 2024-01-25:
I guess the problem is that you use a bind mount. Can you check:
```sh
findmnt -Ufnro SOURCE -T /boot
lsblk -npo PKNAME "$(findmnt -Ufnro SOURCE -T /boot)"
```
Try the following at best:
- Move everything from `/sdcard/boot` to `/boot` resp. `/sdcard`, i.e. to the root of this partition. Then mount the partition directly (no bind mount) to `/boot`.

MichaIng, 2024-01-25:
Hmm, we can try to adjust the `postinst` script to remove the trailing `[...]` part shown on bind mounts. But best would certainly be to use a method like Armbian to store the PARTUUID of the boot partition within the boot script to the kernel command-line and read that one, to be sure where the bootloader was actually loaded from 🤔. However, for now, mounting it directly to `/boot` will do as well.

MrFr33man123, 2024-01-25:
I followed your advice to move everything and remove the bind mount.
```
Setting up firmware-quartz64a (6.6.12-dietpi1) ...
Flashing U-Boot
Flashing to detected boot drive: /dev/sda
```
Looks promissing, but do i have to copy it to sdcard after each update so the actuall bootloader is used? Because that board is not able to boot from USB HDD?

MichaIng, 2024-01-25:
Hmm, now it was flashing the bootloader to the USB drive. Did you mount the partition to `/boot` as well?

MrFr33man123, 2024-01-25:
not exactly sure how to mount the directory directly.

MichaIng, 2024-01-25:
In `/etc/fstab`, change
```
UUID=ef4dca92-a747-4854-82c2-c14828b2c08b /sdcard ext4 noatime,lazytime,rw 0 1
```
to
```
UUID=ef4dca92-a747-4854-82c2-c14828b2c08b /boot ext4 noatime,lazytime,rw 0 1
```
and
```sh
systemctl daemon-reload
umount /sdcard
umount /boot
mount /boot
dpkg-reconfigure firmware-quartz64a
```
And to rule out an issue with the root partition, as the bootloader was flashed there, can you check that it is still all fine:
```sh
fdisk -l /dev/sda
```

MrFr33man123, 2024-01-25:
```
root@tiny-server-01:/# systemctl daemon-reload
root@tiny-server-01:/# mount /boot
mount: /boot: /dev/mmcblk0p1 already mounted on /boot.
       dmesg(1) may have more information after failed mount system call.
root@tiny-server-01:/# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda           8:0    0  1,8T  0 disk
└─sda1        8:1    0  1,8T  0 part /
mmcblk0     179:0    0 58,9G  0 disk
└─mmcblk0p1 179:1    0 58,9G  0 part /boot
root@tiny-server-01:/# df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root       1,8T    351G  1,4T   20% /
devtmpfs        3,8G       0  3,8G    0% /dev
tmpfs           3,8G       0  3,8G    0% /dev/shm
tmpfs           1,6G    9,0M  1,5G    1% /run
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           3,8G       0  3,8G    0% /tmp
tmpfs            50M     32K   50M    1% /var/log
overlay         1,8T    351G  1,4T   20% /mnt/dietpi_userdata/docker-data/overlay2/e880648b5b6a13a21effad40f11ccdebd768be06338116a9315faad997efd0fc/merged
/dev/mmcblk0p1   58G    1,2G   55G    3% /boot
overlay         1,8T    351G  1,4T   20% /mnt/dietpi_userdata/docker-data/overlay2/9d92906492da2692b5d8e1730c0b549d06b620396ab6b33df1062a25055dda9a/merged

root@tiny-server-01:/boot# dpkg-reconfigure firmware-quartz64a
Flashing U-Boot
Flashing to detected boot drive: /dev/mmcblk0
dd: konnte '/boot/u-boot.bin' nicht öffnen: Datei oder Verzeichnis nicht gefunden
grep: /boot/extlinux/extlinux.conf: Datei oder Verzeichnis nicht gefunden
```
The Problem is that now fstab mounts the root of the sdcard (/dev/mmcblk0p1) to /boot not /dev/mmcblk0p1/boot

MrFr33man123, 2024-01-25:
well the firmware update dpkg worked... now my system is not booting anymore. have to get the kid... so thanks for your support. Will try to fix later and do it the "right" way from the beginning. Should work then :)
Thanks for your time and thanks for maintaining dietpi!

MichaIng, 2024-01-25:
Watch out for two possible reasons that the system does not boot:
- `/boot/extlinux/extlinux.conf` points to `/vmlinuz` and `/rk3566-quartz64-a.dtb` for kernel and device tree, instead of the `/boot` sub directory. The `sed` command above should have changed this accordingly.
- That the root partition and filesystem are still fine, e.g. via `fdisk -l /dev/sda` and `blkid /dev/sda1`. Since the bootloader was flashed there, it might have overwritten the superblock, depending on where the first partition started. `fsck` from another system should be able to fix it, restoring the superblock from a backup location (which are generated and maintained automatically by ext4 filesystems).

closed 2024-01-29 by MrFr33man123