MichaIng/DietPi #6871: Failed to stop avahi while installing Kodi

GTP95, 2024-01-21:
I was having issues with Kodi (some plugins crashed on opening), so I decided to first purge it, then, since there was a dietpi update available, I updated dietpi and then tried to install Kodi again. During the installation procedure, I got an error about stopping Avahi. Likely this isn't a big issue and stooping Avahi can just be skipped, but here's my report anyway.

### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
```
$ cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
```
$ echo $G_DISTRO_NAME $G_RASPBIAN
bullseye 0
```
$ uname -a
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

- SBC model: RPi3 B+
- Power supply used: not relevant
- SD card used: not relevant

#### Additional Information (if applicable)
- Software title: Kodi
- installed freshly
- Can this issue be replicated on a fresh installation of DietPi? I don't know
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID: c5af5ca0-c0bb-4c83-9f86-135ad92165ad

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. Uninstall Kodi using dietpi-software
2. Upgrade dietpi using dietpi-update
3. Try to install kodi again

#### Expected behaviour
<!-- What SHOULD be happening? -->
- Kodi gets installed without errors

#### Actual behaviour
<!-- What IS happening? -->
- Error about stopping avahi

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

MichaIng, 2024-01-21:
> I just had to select "exit" instead of "ok" and it worked. This is likely a separate issue though.

Yes, the button labels on the dialogues are not always great. "Ok" in this case means to apply the choice but stay in the menu, and only "Exit" basically finishes the selection. We are aware of this and will go through all menus to better label those buttons soon.

In your bug report upload I do not see any stop attempt for `avahi-daemon`, so I guess this upload is from after the reboot? If it works (and stops) fine now, I see no chance to further debug the issue, so then I suggest to close it. But please reopen if you face the issue again and before rebooting, provide the output of this command:
```sh
journalctl -u avahi-daemon
```

GTP95, 2024-01-22:
I was able to install Kodi after a reboot just fine, so I guess this issue only surfaces when installing it right after a Dietpi version upgrade. Closing this issue for now.

closed 2024-01-22 by GTP95

MichaIng, 2024-01-22:
Not sure whether the DietPi update has anything to do with it. Probably the Avahi-Daemon was just in a faulty state for another reason. It got its last update April last year, so was not updated with this DietPi update.

However, we can check back when you face this again.