MichaIng/DietPi #6872: New Kodi version

GTP95, 2024-01-21:
### Creating a software request

#### Formal software information
- Software title | Kodi
- Short description | Meadi center for Linux
- Official URL (if available) | https://kodi.tv/
- GitHub URL (if available) | https://github.com/xbmc
- (Official) install documentation | https://kodi.wiki/view/HOW-TO:Install_Kodi_for_Linux
#### Are there similar/alternative software titles available with DietPi-Software?
Dietpi alreadi pakages Kodi, but it is an older version (19.4), while now the latest stable version is 20.3

#### What makes your requested software better than the above solutions, if available?
It is the latest stable version of an already packaged software.
#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

Joulinar, 2024-01-21:
We don't build own Kodi version as we install Kodi from globel Debian repository. Kodi version depends on Debian version. I guess you are running Bullseye? 

`echo $G_DISTRO_NAME`

MichaIng, 2024-01-21:
See here which Kodi versions are distributed by which Debian version: https://packages.debian.org/kodi
In case of RPi it is a little different and they are newer, as the RPi Ltd itself provides builds.

GTP95, 2024-01-22:
@Joulinar, you're right:
```
$ echo $G_DISTRO_NAME
bullseye
```
I was assuming that running `dietpi-update` would get me new releases of Dietpi, not just updates for the release I have originally installed. Is there a supported way to upgrade to the latest Dietpi release, other than re-flashing?

MichaIng, 2024-01-22:
It is not about the DietPi version, but about the Debian version. The versions of the Linux kernel, of the underlying Debian base OS, and of the set of scripts/configs we maintain as DietPi project, are not directly tied to each other 😉.

An upgrade of the underlying Debian version to a new major release is a little larger step, as it means to replace each and every installed APT package. And since there is no 100% guarantee that it will work without issues in all cases, and as some software titles and/or configs may loose support, we never do this via `dietpi-update` but instead provide an article and script to read/understand and do it as a separate better controlled step: https://dietpi.com/blog/?p=3128

GTP95, 2024-01-22:
Thanks for the link! I assumed DietPi to be a Debian derivative, I now understand it is a set of scripts on top of Debian. thanks for the clarification.

closed 2024-01-22 by GTP95