MichaIng/DietPi #6883: Upgrade to Dietpi 9 on RISC triggers cp bug

rhinot, 2024-01-23:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=25
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`trixie`

- Kernel version | `uname -a`
`Linux mars 6.1.68 #1 SMP Fri Dec 15 22:27:03 UTC 2023 riscv64 GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`StarFive VisionFive 2 (riscv64)`

- Power supply used | (EG: 5V 1A RAVpower)
USB 3.0 port of a USB-C PD powered RPi4 :)

- SD card used | (EG: SanDisk ultra)
Not a great one.

#### Steps to reproduce
1 - Run dietpi_update to go to upgrade to 9.0.2

#### Expected behaviour
Update completes

#### Actual behaviour
Updates stick on following
```
Setting up linux-image-6.6.13-riscv64 (6.6.13-1) ...
I: /vmlinuz.old is now a symlink to boot/vmlinux-6.6.11-riscv64
I: /initrd.img.old is now a symlink to boot/initrd.img-6.6.11-riscv64
I: /vmlinuz is now a symlink to boot/vmlinux-6.6.13-riscv64
I: /initrd.img is now a symlink to boot/initrd.img-6.6.13-riscv64
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.6.13-riscv64
cp: warning: behavior of -n is non-portable and may change in future; use --update=none instead
cp: warning: behavior of -n is non-portable and may change in future; use --update=none instead
...
```

#### Extra details
Searching [Reddit](https://www.reddit.com/r/debian/comments/199g2vm/warning_on_cp_when_upgrading/) [the bug could just be in cp](https://debbugs.gnu.org/cgi/bugreport.cgi?bug=62572), but reporting as I didn't run into this issue on my RPi4 running dietpi, which I also upgraded today.

MichaIng, 2024-01-23:
Jep, this is the same on all Trixie and Sid systems. `update-initramfs` seems to use the `cp -n` flag, which seems to have been deprecated between `coreutils` 9.1 and 9.4:
- 9.1: https://manpages.debian.org/bookworm/coreutils/cp.1.en.html
- 9.4: https://manpages.debian.org/unstable/coreutils/cp.1.en.html#n

There is actually an interesting story behind this change: https://bugs.debian.org/1058752
- GNU decided to change `-n` to fail loudly if the target file exists, instead of silently doing nothing.
- This of course is a breaking changes for all scripts which rely on it silently failing (succeeding) and not producing any output on STDERR.
- Hence Debian decided to revert the change and instead add this prompt to encourage users to switch to another more explicit option, so at some point in the future `-n` can be aligned with upstream.

Hence don't worry, all good. `initramfs-tools` will switch to another option as well soon:
- https://bugs.debian.org/1055694

rhinot, 2024-01-23:
Thank you for the thoughtful response!

MichaIng, 2024-01-23:
I'll mark the issue as closed, but feel free to post further questions, if you have some.

closed 2024-01-23 by MichaIng