MichaIng/DietPi #6885: Add support for Home Assistant Green

raoulbhatia, 2024-01-24:
### Creating an image request

#### Formal device information
- Device name | Home Assistant Green
- Official product URL | https://www.home-assistant.io/green/
- GitHub resource URL (if available) | https://github.com/home-assistant/operating-system
- Image download URL | https://github.com/home-assistant/operating-system/releases

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- No

#### If not, are there install instructions for Debian available?
- No

Joulinar, 2024-01-24:
This seems to be a hardware designed to run HA. What would be the benefit of running DietPi on it?

MichaIng, 2024-01-24:
Hmm, 99 $ for an RK3566 with 4 GB RAM and 2x USB2.0 only seems quite expensive. And as a rare/custom board, it will have bad kernel support, I suppose. There are much better SBCs with this chip for much less money. So the only real benefit, if one sees this as benefit, is that it has Home Assistant (OS) pre-installed. So I see no point to add DietPi support for it, which will be also hard to achieve and maintain, given that there is likely some custom/vendor kernel and bootloader used.

raoulbhatia, 2024-01-25:
I thought it to be quite reasonable as it comes with

- proper casing
- big heat sink
- 32 GB eMMC flash drive

These parts need to add on top when buying a regular board.  Also, the home assistant green will probably offer a higher resale value, if required.

The only thing missing seems to be an alternative OS that you can configure more freely.

On a side note: Which alternatives would you suggest?

MichaIng, 2024-01-25:
E.g. https://www.friendlyelec.com/index.php?route=product/product&product_id=287
- USB 3.0
- A really good case
- M.2 NVMe socket
- largest variant 85 $
- in the meantime mainline kernel support

But just an example with the same SoC. Of course there are other smaller or larger ones, based on what is needed.

olivievranska, 2024-09-08:
Agree with requester.

I started HA journey with Home Assistant Green and in few months I realised that my requirements are over HAG possibilities, especially eMMC and RAM limits.

After tests and trials, yesterday I moved my HA to Raspberry Pi 5 8 GB, 256 GB NVMe. And seems all previous problems were related to weak HAG hardware because are gone.

So now I'm thinking how to use HAG for other tasks as I don't like wasting. I guess there will be more cases like mine.

DietPi is very useful distro, I tried in bought RPI5 8 GB. I think it will be appreciated not only by me if available also for HAG, as another SBC.