MichaIng/DietPi #6886: Mosquitto installation: Password file setting fails since v9.0

StephanStS, 2024-01-26:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
  ```
  G_DIETPI_VERSION_CORE=9
  G_DIETPI_VERSION_SUB=0
  G_DIETPI_VERSION_RC=2
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  ```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
  `bookworm`
- Kernel version | `uname -a`
  `Linux work1 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
  `Virtual Machine (x86_64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | Mosquitto
- Was the software title installed freshly or updated/migrated?
  Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
  Yes

#### Steps to reproduce
1. Execute `dietpi-software install 123`
2. Execute `cat /etc/mosquitto/passwd`

#### Expected behaviour
- User `mosquitto` should be within the passwd file

#### Actual behaviour
- User `dietpi` occurs in the passwd file

#### Extra details
The change done in the [Buster code extraction](https://github.com/MichaIng/DietPi/commit/0aaba5a6133d8a7d8452f9ecf4a9245da6c8a094), file `dietpi-software`, line 6851ff was too much extraction.

The code changes done via https://github.com/MichaIng/DietPi/issues/4424 need to be kept.

MichaIng, 2024-01-26:
#4424 is obsolete since Bullseye, but we replace the username argument with the password, instead of the password. Fixed with: https://github.com/MichaIng/DietPi/commit/db3f6bf

StephanStS, 2024-01-27:
Worked also in my environment.

MichaIng, 2024-01-27:
Great, I'll mark this as closed then.

closed 2024-01-27 by MichaIng