MichaIng/DietPi #6887: Feature Request: Add option to integrate screen/byobu/etc. into console by default

erikpopp, 2024-01-26:
### Creating a feature request

This feature, if implemented, would significantly improve the user experience and make DietPi feel fancier.

I propose that you add the option to make it so when you connect to DietPi by console, it will automatically start a screen session for it. This would make it harder to brick your device if an SSH session drops when making changes to the system using tools like dietpi-config or dietpi-update.

I would like it properly integrated into DietPi's console startup/login script and into dietpi-config such that users would not need to enter any input except when DietPi cannot decide automatically which session to rejoin.

For the sake of ease of use, it should also:
X Check on connect/login if there is already an existing session
X If there is only one session and it was recently disconnected, automatically rejoin it
  + DietPi should notify the user that it is rejoining an existing session
X If someone is already connected to that session, ask the user if they would like to reconnect to that session
X If there are multiple sessions and DietPi does not see one that it knows was recently disconnected, ask the user which session to connect to
X DietPi should intelligently guess which session you are most likely to want based on:
  + Which sessions have been disconnected from
  + How recently someone interacted with the session
X DietPi should list sessions in order of likelihood that they are the one that the user wants and automatically select the most likely session
X Add a screen session manager section to dietpi-config to easily manage and join screen sessions
  + Would this be better split off into a separate utility? dietpi-session-manager or something like that?
X This session manager should give some kind of preview of the contents of each session so that users can easily tell which session is which
  + If it uses one line per session, it should probably list the last command run and/or the most recent line output
  + If it uses a tiling layout, the tile should have list the last command run and then the most recent lines output - however many can fit on screen with the number of tiles that are displayed

MichaIng, 2024-01-27:
I personally have no time any soon for a complex integration of multiple session solutions. But I have a simple GNU screen config, automatically creating a session on first SSH login, and automatically reconnecting to it: https://github.com/MichaIng/hacks/blob/main/screen_ssh_sessions.sh

Paired with some documentation for the shortcuts to create new screens, swap and detach from the session, we could add this moreless as is.

erikpopp, 2024-01-27:
> I personally have no time any soon for a complex integration of multiple session solutions. But I have a simple GNU screen config, automatically creating a session on first SSH login, and [...]

That sounds awesome! I agree that the integration I suggested would be complex. This would provide most of the benefits from my suggestion at a fraction of the cost in time and effort. Thank you for considering my idea.