MichaIng/DietPi #6889: dietpi_update problem

nyko100, 2024-01-27:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues -> https://github.com/MichaIng/DietPi/issues/5731

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=7
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

  bullseye 1

- Kernel version | `uname -a`

  Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

  RPi Zero (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)

  5 V 1,2 A power adapter

- SD card used | (EG: SanDisk ultra)

  xlync 16 GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

  driver for roller blind, written in c++ 17

- Was the software title installed freshly or updated/migrated?
 
  freshly from source
  
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

805cb436-6fdd-45eb-9220-2e791f8773a8

#### Steps to reproduce

1. sudo /boot/dietpi/func/dietpi-set_hardware enableipv6 enable
   (as explained in thread https://github.com/MichaIng/DietPi/issues/5731)

2. sudo dietpi-update

#### Expected behaviour

updating the system

#### Actual behaviour

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------

#### Extra details

#### Details:
- Date           | Sat Jan 27 10:23:58 CET 2024
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes
From fdfa:85b9:3061:0:209:4fff:fec2:93b2 icmp_seq=1 Destination unreachable: No route

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

Joulinar, 2024-01-27:
Seems like IPv6 is not correctly working. Did you already tried to disable IPv6?

nyko100, 2024-01-27:
I disabled ipv6 with
su root
echo 1 > /proc/sys/net/ipv6/conf/all/disable_ipv6

now dietpi-update runs. But this seems not to be the proper way to update. This problem came up approx. 2 month ago, before dietpi-update ran without any problems.

Joulinar, 2024-01-27:
> now dietpi-update runs. But this seems not to be the proper way to update

It is the correct way if your system doesn't have a valid IPv6 address. 

> This problem came up approx. 2 month ago, before dietpi-update ran without any problems

This statement somehow does not match the DietPi source version v8.7.1 as stated above. Version 8.7.1 is over 1.5 years old.

nyko100, 2024-01-27:
ok, maybe the last update was made more than 2 months ago, don't know. At least now the update script is running. Thanks for your help

nyko100, 2024-01-27:
-> It is the correct way if your system doesn't have a valid IPv6 address.

good to know, thanks

Joulinar, 2024-01-27:
> It is the correct way if your system doesn't have a valid IPv6 address.

Disable IPv6 if not present. This is what you did now.

MichaIng, 2025-06-09:
Btw: when IPv6 is enabled but the network check fails, the error handler menu shows you an option to disable IPv6. It basically does
```sh
sudo /boot/dietpi/func/dietpi-set_hardware enableipv6 disable
```
I'll close this issue.