MichaIng/DietPi #6890: Raspberry PI 4 + DietPi = problem with "kiauh"

Slawaspr0, 2024-01-27:
### Creating a bug report/issue
- [ X ] First I was trying to search for help in kiauh github here: https://github.com/dw-0/kiauh/issues/426

#### Required Information
- DietPi version / Distro version / Kernel version - lastest, I just downloaded system image a few hours earlier, and let it update itself
- Raspberry PI 4
- IDK - some step down from printer power supply
- I use SSD GoodRam 128 GB

#### Additional Information (if applicable)
- kiauh
- installed freshly
- Can this issue be replicated on a fresh installation of DietPi? - Yes

#### Steps to reproduce

Run DietPI
Login to user dietpi
Trying to run "cd ~ && git clone https://github.com/dw-0/kiauh.git" - cant, no access, need root
Login to root
Again trying to run "cd ~ && git clone https://github.com/dw-0/kiauh.git" - success
Trying to run "./kiauh/kiauh.sh" - can't run this as root
Login to dietpi
Again trying to run "./kiauh/kiauh.sh" - can't run, no directory access

#### Expected behaviour
I needed to make new/clean system install on my RaspberryPi. Fast configuration - WIFI connection - SSH and trying to install kiauh. Last time i did it more than year ago. Something changed since that, because I remember that last time I did it without problems.

#### Actual behaviour
Can't use kiauh. I can install it, but can't run/use.

#### Extra details
I'm using my RaspberryPi with DietPi system in 3D printer. Kiauh is most popular script to manage my printer software/firmware. More details in link I provided at begining.

MichaIng, 2024-01-27:
The `dietpi` user should of course have write access to its home directory. Can you verify that this is `/home/dietpi` as expected, and that it is owned by `dietpi`:
```sh
pwd
ls -l .
```

Slawaspr0, 2024-01-27:
But default install directory of kiauh is root/kiauh, i was trying to move it manually to /home/dietpi then I had to change owner from root to dietpi (because kiauh can only be installe by root), then I was trying to change chmod of all files, at the end I cant run kiauh

MichaIng, 2024-01-27:
Now I am confused. I thought the installer itself denies to be executed as root? Why does it explicitly try to install into the root's home directory then 🤔. And still, at first even `git clone` failed, as you wrote, so please run the above commands and paste the output here, so we can verify that `dietpi` has write access to its own home directory, as intended.

dw-0, 2024-01-27:
> But default install directory of kiauh is root/kiauh

That is incorrect. You shouldn't even download kiauh as root user nor in roots home directory. You should never use root for kiauh. Always a regular user. And then download kiauh into that users home directory.

> because kiauh can only be installed by root

That is also incorrect. There is no such thing as "installing" kiauh. It's a git clone. In other words a simple download.

MichaIng, 2024-01-27:
And why do you say "But default install directory of kiauh is root/kiauh" then? Or did I misunderstand it? Usually, such installers just stay in the PWD, which makes doubled sense here since it is a Git directory.

Slawaspr0, 2024-01-27:
Look at this
[image]

MichaIng, 2024-01-27:
Can you please run the above 3 commands and paste the output here? I can still not see whether the directory is the intended one, and whether the `dietpi` user has intended permissions for this directory.

Slawaspr0, 2024-01-27:
Sure
[image]

MichaIng, 2024-01-27:
Ah sorry, this one to see permissions of the directory:
```sh
ls -dl .
```

Slawaspr0, 2024-01-27:
Hmmm, now I see, shoul be there dietpi instead of root right?
[image]

MichaIng, 2024-01-27:
So indeed that directory is owned by `root`. That is pretty unintended, of course. Fix this, then you can clone and install the software there:
```sh
sudo chown -R dietpi:dietpi /home/dietpi
```

Slawaspr0, 2024-01-27:
Ok, gonna try.
[image]
Cloning complete, looks promising, now trying to run it
[image]
Ok, super! It's working, but now I'm wondering, is this possible, that /home/dietpi have root owner since clean install?

MichaIng, 2024-01-27:
> Ok, super! It's working, but now I'm wondering, is this possible, that /home/dietpi have root owner since clean install?

Okay perfect. And nope, `/home/dietpi` is definitely owned by `dietpi` on fresh images.

Slawaspr0, 2024-01-27:
Ok, so IDK how i did it, but I'm really happy that You helped me to fix this, thank You.

MichaIng, 2024-01-27:
Sure 🙂. I'll mark the issue as closed then.

closed 2024-01-27 by MichaIng