MichaIng/DietPi #6891: DietPi-Config | Options are saved in config.txt at the wrong location (9.x)

mann1x, 2024-01-29:
### Creating a bug report/issue
- [ X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

>G_DIETPI_VERSION_CORE=9
>G_DIETPI_VERSION_SUB=0
>G_DIETPI_VERSION_RC=2
>G_GITBRANCH='master'
>G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

> bookworm 0

- Kernel version | `uname -a`

> Linux DietPi5 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

> RPi 5 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

> Official Raspberry 5A 27W 

- SD card used | (EG: SanDisk ultra)

> Sandisk Extreme Pro 128GB

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?

> Freshly

- Can this issue be replicated on a fresh installation of DietPi?

> Didn't try

#### Steps to reproduce

- Open DietPi-Config
- Change any option Performance. eg. temperature limit to 60c

#### Expected behaviour
- Rebooting the pi should reflect the changes eg. stress testing the CPU should throttle after 60c

#### Actual behaviour
- Changes are not applied. eg. temp limit is at default and the CPU does not throttle at 60c

#### Extra details
- Changes are made in /boot/config.txt but since bookworm the file has been moved to /boot/firmware/config.txt

MichaIng, 2024-01-29:
Did you use our [firmware migration script](https://github.com/MichaIng/DietPi/issues/6676) when installing the new kernel/firmware packages? It creates symlinks from `/boot/config.txt` and `/boot/cmdline.txt` to `/boot/firmware`.

Joulinar, 2024-01-29:
our script did not support CPU temp changed done fot RPi5.

mann1x, 2024-01-29:
Sure I did use the script... mmm, maybe something messed it up?

> our script did not support CPU temp changed done fot RPi5.

Why is there something different for the Pi5?
I didn't check but it looked the same

MichaIng, 2024-01-29:
@Joulinar the `temp_limit` setting should still work. At least I did not see any hint that it would not work anymore.

@mann1x can you verify this:
```sh
ls -l /boot/config.txt
grep temp_limit /boot/config.txt /boot/firmware/config.txt
```

mann1x, 2024-01-29:
> @Joulinar the `temp_limit` setting should still work. At least I did not see any hint that it would not work anymore.
> 
> @mann1x can you verify this:
> 
> ```shell
> ls -l /boot/config.txt
> grep temp_limit /boot/config.txt /boot/firmware/config.txt
> ```

I have changed it manually of course:

>/boot/config.txt:temp_limit=75
>/boot/firmware/config.txt:temp_limit=85

I remember now I have seen someone talking about this symlink... now makes sense.
Wonder what could have wiped them out.
I have installed some stuff outside the software manager but not that much and nothing pi specific that could mess there.
Maybe something that accessed the /boot partition for something else but I can't imagine what.

I'm waiting for a 2nd pi5, this week should be able to double check on a fresh install before opening a bug.

Yes the temp_limit works on pi5 and also the throttling flag.
Thermals are quite poor and power management it's quite raw and unsophisticated as per Broadcom but at least runs pretty damn fast and after a very long time goes into proper idling.

mann1x, 2024-01-29:
@MichaIng 
Sorry I forgot:

>ls -l /boot/config.txt
>-rw-r--r-- 1 root root 118 Jan 28 18:33 /boot/config.txt

MichaIng, 2024-01-29:
Probably I missed one case of `sed` command when adding `--follow-symlinks`. Without it, it replaces symlinks with the edited file, instead of editing the target of the symlink. For now:
```sh
mv /boot/config.txt /boot/firmware/config.txt
ln -sf firmware/config.txt /boot/config.txt
```

mann1x, 2024-01-29:
> Probably I missed one case of sed command when adding --follow-symlinks. Without it, it replaces symlinks with the edited file, instead of editing the target of the symlink. For now:

Uhm, I'm not sure the pi5 would boot with the config.txt almost empty in /boot...
I kept the one in /boot/firmware/.

I guess I have to restore the symlink for cmdline as well.

I have another idea at that might have been, other than sed.
DietPi-Backup & DietPi-Sync.

At the beginning I just configured it on the fly without really checking what would do (so easy to do it!).
I wrongly assumed the backup would be an archive at the end.
Therefore I did setup DietPi-Sync to save on my OMV NAS share for backups via Samba, which doesn't support symlinks.
Once realized the mistake I configured it via NFS.
It did struggle for some unknown reason at the beginning, took 2 hours for a 20% sync.
I tinkered, updated, etc etc and at some point started working properly but I'm not really sure how.
Did some reboots of the pi5 and the NAS in between so thought something was just stuck.

Maybe all this could have messed up the backup; I will try to reproduce it with the new pi5 when it comes.

MichaIng, 2024-01-29:
> Uhm, I'm not sure the pi5 would boot with the config.txt almost empty in /boot...

`/boot/config.txt` is completely unused, not relevant for boot. Only `/boot/firmware/config.txt` is relevant. Hence, if you did changes to `/boot/config.txt`, for them to become effective, you must move it to `/boot/firmware/config.txt`. The symlink is currently necessary for our scripts to work, since they edit (and need to edit for backwards compatibility) `/boot/config.txt`. Same is true for `cmdline.txt`, so check it as well, though it is much less used.

If you ever restored a backup from files stored on a Samba server, then you can consider the whole system to be broken. I guess you have noticed this already, not sure if it would even boot successfully. Quick check:
```sh
ls -l /
```
`/bin` must be a symlink to `/usr/bin`, `lib` to `/usr/lib`.

MichaIng, 2024-01-29:
> you end up with an almost empty config.txt in /boot

Ah, you mean that copying a symlink to a Samba share creates an empty file there? Then you are right, though it would still boot (defaults are sane) and the problems when you restored a backup from a Samba share are much worse than an empty config file.

> I've never needed a restore thankfully.

Then we still need to watch our for the reason of the dereferences symlink. I checked all `sed`'s in `dietpi-config`, and they all use the `--follow-symlinks` flag. Also another check through all files in this repository did not provide any missed case:
```sh
grep -r 'sed.* -[^ ]*i.*config.txt' | grep -v '\--follow-symlinks'
```
Might it be possible that you have own scripts or used `sed` on our own on the config file?

mann1x, 2024-01-29:
No I never ran any custom script or used sed.
The only "stray" from anything a vanilla setup was installing some few additional apt packages, like keepalived.

MichaIng, 2024-01-29:
Hmm, unsure at this point. The `cmdline.txt` is never copied in the first place, but only the symlink generated, by our migration script. So it must have been some dereference like with `sed`.

I could not find a case in our code base, but probably I missed a script which is was installed earlier, and is not updated on `dietpi-update`. Or you ran the migration script when there was a bug inside which is not present anymore. Or there is another command which dereferences symlinks, that I do not know about yet. I'll keep this in mind. Check back by times whether the issue re-appears.

mann1x, 2024-01-29:
Yes indeed, I suspect maybe the update script cause I think it has been updated at least once since I used it.
Thanks a lot for the support, I will keep an eye on it and try to reproduce the steps I've done previously with the new pi5 just to be sure.

mann1x, 2024-01-29:
@Joulinar 
Forgot this bit; the initial_turbo on pi5 doesn't work, at least on mine. 
Set above zero will keep turbo enabled forever.

MichaIng, 2024-01-29:
Hmm, can you verify this again with some console commands, verifying that the CPU frequency is indeed pinned to max? Because `initial_turbo=20` is the default on DietPi, and we have a lot of related output in our RPi 5/migration issue, all showing successfully scaled CPU frequencies.

mann1x, 2024-01-29:
Sure let me know what.
The reference clock is pinned to the max as reported by vcgencmd measure_clock arm and cpuinfo_cur_freq.
This normally flips between the min and 1500-1800 MHz.
What scale normally is the dynamic clock as reported in scaling_cur_freq.
But the problem is that the voltage will be fixed, like with overclock voltage, to a static value (the max vcore allowed for the configuration).  
Always same voltage from vcgencmd measure_volts core.

MichaIng, 2024-01-29:
The voltage is not tied to the frequency. It is common that the voltage is max, as long as the frequency is not absolute min. It jumps between min and max, max whenever the frequency is only 1 MHz above min (100 MHz as of available p-states).

The CPU frequency should be between 1500 and 2400 MHz. Is it not going above 1800 MHz in your case?

What I meant with contradicting results is e.g.: https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1868327069
- Current frequency at 1600 MHz with `initial_turbo=20`. And there has not been another kernel upgrade since then.
- Though it reads from `/sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq`. Since it is what CPUFreq instructs, and can miss if the firmware overrides it, it may simply not show the actual frequency.

@Joulinar
I need some hours before I can test, so if you have time, could you check this in your case:
```sh
cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq
vcgencmd measure_clock arm
```

Joulinar, 2024-01-29:
@MichaIng 

```
root@DietPi5:~# cat /sys/devices/system/cpu/cpu*/cpufreq/cpuinfo_cur_freq
1500000
1600000
1600000
1600000
root@DietPi5:~# vcgencmd measure_clock arm
frequency(0)=2400023808
root@DietPi5:~#
```

mann1x, 2024-01-29:
@MichaIng 
Indeed it can be high also with a low clock; usually cause when it's being pulled didn't yet went down.

What I meant is that it works as expected normally and also what I would expect for the Turbo mode (it is described somewhere in the Raspberry documentation).
The problem is that it never gets out of it.

Let me check again with a vanilla config.txt because I've tested it almost always with some modifications and maybe always at least with a small bump on arm_freq_min.

MichaIng, 2024-01-29:
> Indeed it can be high also with a low clock; usually cause when it's being pulled didn't yet went down.

Let's keep voltage and frequency separate. As I said, there is no real dynamic voltage scaling, at least not up to RPi 4 based on what I tested, and based on what has been merged into the RPi documentation accordingly. So even with 1600 MHz, voltage is max, and 1600 MHz is nearly always shown, as of CPU utilisation caused by the commands to read the frequency.

So in Joulinar's case, at least `cpuinfo_cur_freq` shows expected scaled frequencies in `cpuinfo_cur_freq`. They also always show 2400 MHz in your case? `vcgencmd` is a relatively heavy command. Though on my RPi 2 it shows 700 MHz, hence close to min. @Joulinar does it never show anything else than 2400 MHz? And does this change when you remove/comment `initial_turbo` from `config.txt`?

mann1x, 2024-01-29:
I have exactly the same behavior, every pool is 4 seconds and the boost should last only 20 seconds:

`root@DietPi5:~# pistats -c
temp    clock   volt
52.1    2400    0.8703
52.1    2400    0.8703
52.1    2400    0.8703
52.1    2400    0.8703
52.1    2400    0.8703
52.7    2400    0.8703
52.1    2400    0.8703
53.2    2400    0.8703
52.7    2400    0.8703
52.7    2400    0.8703
52.7    2400    0.8703
52.7    2400    0.8703
52.1    2400    0.8703
53.2    2400    0.8703
52.1    2400    0.8703
52.7    2400    0.8703
52.1    2400    0.8703
53.2    2400    0.8703
52.7    2400    0.8703
51.6    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
52.7    2400    0.8703
53.8    2400    0.8703
53.2    2400    0.8703
53.8    2400    0.8703
53.2    2400    0.8703
53.8    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
53.8    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
53.2    2400    0.8703
53.8    2400    0.8703
53.8    2400    0.8703
54.3    2400    0.8703
53.8    2400    0.8703`

I wonder if I have the latest eeprom... I have seen somewhere there is one from 24th of Jan but mine is:

CURRENT: Fri Jan  5 15:57:40 UTC 2024 (1704470260)

I have no idea how it works... is there a different eeprom for different memory models? I have the 4GB version

mann1x, 2024-01-29:
@MichaIng 
In my case, voltage scaling seems to work as expected with the pi5 (at least without the boost).

Seems to be properly reported and there's an obvious and massive difference in temperature and power consumption when it's fixed.

MichaIng, 2024-01-29:
> In my case, voltage scaling seems to work as expected with the pi5 (at least without the boost).

You mean that it dynamically scales with multiple values between min and max? Probably it has changed for RPi 5 or for all models with a kernel update. Would be good of course, however, not relevant for this topic.

You can update to latest EEPROM with:
```sh
FIRMWARE_RELEASE_STATUS=latest rpi-eeprom-update -a
```
The binaries for all RPi 5 variants, so I doubt it makes a difference, but I cannot rule it out either. However, Joulinar has a 4 GB variant as well.

mann1x, 2024-01-29:
Okay, it's an apt package the eeprom so it wasn't updated yet.
The latest is indeed from the 24th of Jan:
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2712/release-notes.md

Yes the voltage scales beautifully now on the pi5.
You can also set an offset delta in config.txt and keep it dynamic so much better under and over volting.
Indeed not relevant :)

mann1x, 2024-01-29:
I was checking the initial_turbo description:

>Enables turbo mode from boot for the given value in seconds, or until cpufreq sets a frequency. The maximum value is 60.

So it can be controlled also by cpufreq.
Maybe my installation got some more issues than just the symlink of config.txt.

I'm wondering if anyone checked if initial_turbo actually works.

@Joulinar 
Can you check if enabled for 60 seconds the clock and voltage is effectively forced to max for that period?

mann1x, 2024-01-29:
No it doesn't report the max frequency, only vcgencmd:

`
root@DietPi5:~# cat /sys/devices/system/cpu/cpu[0-3]/cpufreq/cpuinfo_cur_freq
1500000
1600000
1600000
1600000

`

MichaIng, 2024-01-29:
Okay thanks, at least consistency across results now. But quite strange that `cpuinfo_cur_freq` does not report the correct frequency now. This is actually a 2nd bug, though probably one with the kernel, instead of with the firmware.

I will replicate this later, just to be sure, and then report at the RPi firmware repo.

mann1x, 2024-01-29:
I have the official case with the small heatsink and the fan on top (not the active cooler).
The clock is set at max because setting `arm_core_min` to 2400 will make report `cpuinfo_curr_freq` the max clock as well and the idle temperature is exactly the same.

The delta versus the default configuration is around +3° C for me.

Without `initial_turbo`:
```console
temp    clock   volt
53.2    1600    0.7200
53.2    1600    0.7200
53.8    1600    0.7200
53.2    1600    0.7640
53.2    1600    0.7200
53.2    1600    0.7200
53.2    1600    0.7640
51.6    1600    0.7640
53.8    1600    0.7200
53.2    1600    0.7200
53.2    1600    0.7200
52.7    1600    0.7640
53.2    1600    0.7200
53.2    1600    0.7200
52.7    1600    0.7640
52.1    1600    0.7640
```

With `initial_turbo=20`:
```console
temp    clock   volt
54.9    2400    0.8732
54.3    2400    0.8732
55.4    2400    0.8732
57.1    2400    0.8732
56.0    2400    0.8732
56.5    2400    0.8732
56.5    2400    0.8732
56.0    2400    0.8732
56.0    2400    0.8732
56.0    2400    0.8732
57.1    2400    0.8732
56.5    2400    0.8732
57.1    2400    0.8732
56.5    2400    0.8732
56.5    2400    0.8732
56.5    2400    0.8732
57.1    2400    0.8732
56.5    2400    0.8732
57.1    2400    0.8732
56.0    2400    0.8732
```

With `initial_turbo=20` and `arm_freq_min=2400`:
```console
temp    clock   volt
56.0    2400    0.8728
56.5    2400    0.8728
56.5    2400    0.8728
58.7    2400    0.8728
57.1    2400    0.8728
56.5    2400    0.8728
56.5    2400    0.8728
56.5    2400    0.8728
56.5    2400    0.8728
56.0    2400    0.8728
56.5    2400    0.8728
57.1    2400    0.8728
56.5    2400    0.8728
```

MichaIng, 2024-01-29:
Okay, not a dramatic difference, but not nothing either. Since the trip points are at 50 °C and 60 °C, the fan speed should be the same in both cases as well.

mann1x, 2024-01-29:
Well, it's not a big delta indeed.
If the fan spins up & down mainly depends on the ambient temperature where the pi5 is running and mine is not really low, probably around 22-24c.

Configured with the minimum frequencies can gain 1-2c and after a long while with low load can go down to 47-48c.
Without running monitoring probably even lower because I see it completely disappearing from the battery load of the small UPS where it's connected.
For reference, on mine undervolted at -5000 it's still stable at 2800 MHz and reduce the temperature by 1c.

```console
temp    clock   volt
49.4    1800    0.8014
48.8    1800    0.7885
48.3    1100    0.8014
48.8    1800    0.8014
48.8    1800    0.8014
48.3    1800    0.8014
48.8    1800    0.8014
48.3    1800    0.8014
49.4    1800    0.7885
48.8    1800    0.7885
48.8    1800    0.7885
49.4    1800    0.8014
48.8    1800    0.8014
48.8    1800    0.8014
48.3    1800    0.7885
48.8    1800    0.8014
49.4    1800    0.7200
49.4    1800    0.7885
48.8    1800    0.7949
49.4    1800    0.8014
48.3    1800    0.8014
48.8    1800    0.7885
48.3    1800    0.8014
49.4    1800    0.7885
48.8    1100    0.8014
```

terminet85, 2024-01-30:
My temp dropped about 8-10 °C when I removed initial turbo. I guess the difference with tou is that's because I've edited my fan profile, so fan start early to spin instead the stock values:

```
#Fan
dtparam=fan_temp0=30000,fan_temp0_speed=100 
dtparam=fan_temp1=40000,fan_temp1_speed=150
dtparam=fan_temp2=57500,fan_temp2_speed=200
dtparam=fan_temp3=65000,fan_temp3_speed=255 
```

MichaIng, 2024-02-01:
`initial_turbo` issue reported: https://github.com/raspberrypi/firmware/issues/1864

mann1x, 2024-02-02:
@MichaIng 

I have bug which is specific to dietpi, it can't be reproduced on raspbian with the same kernel.
Would be nice if someone could try to reproduce it.

The network I/O is slower than it should be; the answer to ping is about 2/3 ms, it should be below 1 ms same as raspbian.
I use a 2.5G USB NIC and, while testing the OC, I noticed the throughput with iperf was below the usual:

>[SUM] 0.0000-13.0212 sec  3.25 GBytes  2.14 Gbits/sec

Luckily, there's an easy and very weird solution
Just add to `/boot/firmware/config.txt`

> isp_freq=910

That's enough to fix it somehow....
Ping answers are back to normal below 1 ms and the USB NIC works properly:

>[SUM] 0.0000-12.6394 sec  3.31 GBytes  2.25 Gbits/sec

The power consumption of the pi5 is now 2.7W instead of 2.2W and, unlike before, there's no way to reduce it.

MichaIng, 2024-02-02:
`isp_freq=910` is/should be the default: https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking
You can check back with and without this setting applied via:
```sh
vcgencmd get_config isp_freq
```
Did you apply any other overlocking settings? AFAIK `gpu_freq` overrides all default GPU block frequencies, which were not set explicitly.

**However, please answer in the Raspberry Pi 5 issue (pinned) of our GitHub repo, to not mix topics here.**

mann1x, 2024-02-02:
> `isp_freq=910` is/should be the default: https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking You can check back with and without this setting applied via:
> 
> ```shell
> vcgencmd get_config isp_freq
> ```
> 
> Did you apply any other overlocking settings? AFAIK `gpu_freq` overrides all default GPU block frequencies, which were not set explicitly.

Yes it's the default, that's why the solution is weird...
It's the same without any other settings except the defaults.

If it's explicitly set to 910 works fine, as it should.
Not an issue on raspbian.

MichaIng, 2024-02-02:
Did you check back the value with this command with and without the setting applied? And when you say "Raspbian", do you mean Raspberry Pi OS 32-bit or 64-bit, and did you compare with the same DietPi?

mann1x, 2024-02-02:
The value reported is always 910:

```console
root@DietPi5:~# vcgencmd get_config isp_freq
isp_freq=910
```

2 different Pi5; 4GB with NOOBS and Pi OS 64-bit, 8GB with DietPi 5

MichaIng, 2024-02-03:
Do they have the same EEPROM/bootloader version as well? In case:
```sh
FIRMWARE_RELEASE_STATUS=latest rpi-eeprom-update -a
```

mann1x, 2024-02-03:
Yes both the same, the 5th of Jan:

```shell
root@DietPi5:~# rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Fri Jan  5 15:57:40 UTC 2024 (1704470260)
    LATEST: Fri Jan  5 15:57:40 UTC 2024 (1704470260)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)
            Use raspi-config to change the release.
```

I checked cause I was curious if running the Pi OS would updated it to the 24th Jan release.

MichaIng, 2024-02-20:
> I checked cause I was curious if running the Pi OS would update the eeprom to the 24th Jan release.

No need to install RPi OS to check this. We use the same repo and same package, so DietPi systems get exactly the same kernel and bootloader packages offered at exactly the same time 😉.

This is also why I am confused that both behave differently as you say regarding the USB Ethernet speed, on top of the wired solution of applying `isp_freq=910` which is the default and reported by `vcgencmd` as the currently applied value, regardless whether you set it explicitly or not. And then of course the fact that `isp_freq` is about the image sensor frequency, hence has (should have) absolutely nothing to do with USB. The problem is that I have no USB Ethernet adapter to test this.

Some anyone else here have a USB Ethernet adapter to test, at best a 2.5G one?

Onboard Ethernet performs normal? And you can perfectly toggle the USB NIC speed by adding/removing the setting? And how did you actually found out that this particular setting toggles it, given that it is (should be) completely unrelated? 😄

mann1x, 2024-02-20:
I just tried it again on the Pi 5, which got debian, eeprom and dietpi updates in the meantime and it does work normally.
There's no need to set it at 910 now.

Not sure how the video clock is connected but it's not surprising with this SoC; some unusual things, like access to the L2 cache goes thru the Videocore.

I found it out because I constantly monitor ping answers times to check for instability.
One Pi 5 was answering as expected at 1ms, the other instead at 3-4ms.
You can check this way with the onboard 1G NIC, you just need to be locally connected on the same network switch.

This is the normal answer:

```shell
root@solidpc:~# ping 192.168.178.11
PING 192.168.178.11 (192.168.178.11) 56(84) bytes of data.
64 bytes from 192.168.178.11: icmp_seq=1 ttl=64 time=0.409 ms
64 bytes from 192.168.178.11: icmp_seq=2 ttl=64 time=0.400 ms
64 bytes from 192.168.178.11: icmp_seq=3 ttl=64 time=0.367 ms
64 bytes from 192.168.178.11: icmp_seq=4 ttl=64 time=0.600 ms
64 bytes from 192.168.178.11: icmp_seq=5 ttl=64 time=1.04 ms
64 bytes from 192.168.178.11: icmp_seq=6 ttl=64 time=0.410 ms
64 bytes from 192.168.178.11: icmp_seq=7 ttl=64 time=0.368 ms
^C
--- 192.168.178.11 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6012ms
rtt min/avg/max/mdev = 0.367/0.512/1.035/0.225 ms
```

Specifying `isp_freq=910` could toggle this behavior, 100% reproducible.

MichaIng, 2024-02-20:
Ah good, probably the bootloader update fixed this as well. Just checked, and the one which fixes also `initial_turbo` is applied automatically on all systems on boot, as it is in "default":
```
/lib/firmware/raspberrypi/bootloader-2712/default/pieeprom-2024-02-16.bin
```

closed 2024-03-17 by MichaIng