MichaIng/DietPi #6893: Incorrectly named Amiberry package causes install failure

JerichoCross, 2024-02-01:
### Creating a bug report/issue
- [ Y] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 0~

- Kernel version | `uname -a`
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
Official RPi 4 power supply

- SD card used | (EG: SanDisk ultra)
None - booting from USB - SanDisk Ultra fit 32Gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Amiberry

- Was the software title installed freshly or updated/migrated?
Fresh install

- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`
13289830-5e5f-4dec-9694-5d135e56f998

#### Steps to reproduce

1. Fresh install dietpi
2. Attempt to add the Amiberry package in dietpi-launcher

#### Expected behaviour

Amiberry should be downloaded and installed

#### Actual behaviour

The system attempts to download https://dietpi.com/downloads/binaries/bookworm/amiberry_rpi4-sdl3-64-sdl2.deb which does not exist - the curl instruction returns a 404 error.

#### Extra details

Selecting the option to manually adjust the command and removing the reference to sdl3 works as expected.

MichaIng, 2024-02-01:
Many thanks, this should fix it: https://github.com/MichaIng/DietPi/commit/15ff5a3

JerichoCross, 2024-02-01:
Many thanks for the quick turn around.