MichaIng/DietPi #6894: [minor²] Rephrase dietpi-software whiptail entry

cprima, 2024-02-01:
This is a minor suggestion to rephrase a whiptail menu as following:

Maybe it's just me: When presented the whiptail screen of dietpi-software I tend to be momentarily confused which item to select to proceed. (I install DietPi maybe 2 or 3 times a year). As is:
`Install            : Go >> Start installation for selected software`

I suggest to change the "Go*" wording slightly.
`Install            : Mark this and go to <Ok> to start installation`
or
`Install            : Ok >> Start installation for selected software`

I considered to not report this because it feels too nit-picky. ;)

[image]

MichaIng, 2024-05-07:
You actually do not need to select "<Ok>". Just mark/select a menu entry and hit ENTER/RETURN. "<Ok>" is a very unnecessary button in such menus, but sadly there is no way to hide it in `whiptail`. Only the left side Cancel/Exit button can be hidden, but that one does have a reason, though hitting ESC does the same.

I wonder whether it helped more to change the text of the "Ok" button. Like "Select"? If we have "Go" on the button and in the "Install" text, one could also get the impression that the button implicitly means to start installs, while it actually means to do whatever menu entry is currently highlighted/selected 🤔.

MichaIng, 2026-11-16:
Since a while, the `<Ok>` button shows `<Select>` for our whiptail menus menus, which better reflects that the marked menu entry is selected with that button, resp. the selected entry used. So the description at each entry says what the entry does.

An optional little whiptail tutorial was something I thought about, too 😄.

closed 2026-11-16 by MichaIng