MichaIng/DietPi #6895: Unable to connect over WiFi

JerichoCross, 2024-02-01:
Creating a bug report/issue
[ Y] I have searched the existing open and closed issues

Required Information
DietPi version | cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | echo $G_DISTRO_NAME $G_RASPBIAN
bookworm 0~

Kernel version | uname -a
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

SBC model | echo $G_HW_MODEL_NAME or (EG: RPi3)
RPi 4 Model B (aarch64)

Power supply used | (EG: 5V 1A RAVpower)
Official RPi 4 power supply

SD card used | (EG: SanDisk ultra)
None - booting from USB - SanDisk Ultra fit 32Gb

Additional Information (if applicable)
I have tried setting up the Wifi in dietpi-config during the first boot, also by editing the txt files prior to the first boot and also booting with wired connection first and then setting up the Wifi. I have attempted this with two different access points but still couldn't get a DHCP lease. I have also tried setting a static IP address on the Wifi interface but still can't get connected.

Was the software title installed freshly or updated/migrated?
Fresh install

Can this issue be replicated on a fresh installation of DietPi?
Yes

JerichoCross, 2024-02-01:
Some further information, sometimes it connects, other times it doesn't. The pi is within 10 feet of the access point with no obstructions in the way. I am using the hdmi-0 port and it is booting from a USB 3 flash drive. There is also a wireless keyboard dongle plugged into one of the USB 2 ports. I have read that USB devices may cause issues with 2.4GHz wireless but this should be connecting at 5GHz.

JerichoCross, 2024-02-02:
I seem to have 'solved' this problem by changing the gpu memory split to 192. Could be coincidental but the wifi was flawless afterwards. I had rebooted several times previously with no luck so it wasn't the reboot that fixed it.

MichaIng, 2024-05-07:
Sorry for missing your issue.

> but this should be connecting at 5GHz.

This cannot be controlled, it connects to whichever has better signal strength, AFAIK. So unplugging the wireless keyboard would at least have been a reasonable test.

> I seem to have 'solved' this problem by changing the gpu memory split to 192.

That is basically impossible to have any effect. Any memory split above the default is just a waste of system memory and remains unused on RPi 4, unless you use very specific legacy camera/video applications.

Generally, in case you face this again, or really immediately when commenting/reverting the GPU memory, check `journalctl -u ifup@wlan0` and `dmesg -l 0,1,2,3` for possible hints.

I'll mark this as closed. But feel free to reopen if the issue re-appears or you want to run some test with reverted GPU memory.