MichaIng/DietPi #6896: MotionEye fails to restart after reboot.

ur7x, 2024-02-01:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bookworm 0

- Kernel version | `uname -a`
Linux MotioneyeBK 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- RPi 4 Model B (aarch64)
- Stock RPi 4 branded power supply

- SD card used | (EG: SanDisk ultra)
- USB -> NvMe Drive.  1TB Samsung SSD

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- MotionEye
- 
- Was the software title installed freshly or updated/migrated?
- Fresh Install using dietpi software install tool
- 
- Can this issue be replicated on a fresh installation of DietPi?
- Yes
- 

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Imaged SSD with Etcher
2. Logged in as root. Did all recommended updates etc
3. Installed  MotionEye
4. Install Cameras
5. Reboot system
6. Motion Eye is dead

#### Expected behaviour

- .MotionEye recovers from reboot or power cycle 

#### Actual behaviour

- If I reboot BEFORE Camera installs MotionEye restarts after reboot
- If I reboot AFTER Camera installs MotionEye not only fails to restart, but requires a complete uninstall and reinstall to get the system back... 

#### Extra details

- currently supporting 9 Cameras.  3 from Octoprint Servers, 6 others from rtsv security cameras around the house. 
- All camera's record (motion based) to the SSD
- 4 of the camera's report motion to HA via MQTT calls.

ur7x, 2024-02-02:
I think I have figured this  out... Part user issue part motioneye issue.

I started in Motion Eye in Home Assistant.  supporting 9+ cameras... This overloaded the Pi4 and caused stability issues  So I took a Backup of my config  Fired up a dedicated motioneye on a Dietpi PC...    Did a Restore and the camera's all came back.  
The system would work great until a reboot.  Then it was dead.. REALLY dead.

What I didn't know was that the Restored config file contain specific directories that were unique to HA.  And Motioneye rather than creating the missing folders... Just crashed...  After sniffing though the log files I figured it out.

Unfortunately even after changing all of the file paths and permissions the system was still unable to reboot....

The solution?  When porting to a new OS/system/environment.  Restore looks like it works, but results in an unstable solution...  

The only I could port to a new system was to manual re-add all of my cameras.

closed 2024-02-02 by ur7x

Joulinar, 2024-02-02:
how did you backup your configuration? Did you used `dietpi-backup`?

ur7x, 2024-02-02:
The issue was not the dietpi-backup.  I'm very impressed with the ease of use that DietPi offers.  Very slick system

The issue is with the backup within MotionEye.   That backup... stores the camera settings, but also stores a bunch of system variables and file locations.  These often change if you are running MotionEye within HA vs in MotionEyeOS vs in a Docker Container vs on DietPi...  

When the Restore happens MotionEye has a bunch of broken links,  that WILL fail and corrupt the MotionEye system upon any boot, or "restart" event.  Zero warning is offered to the user... it just looks like it works..  But it is running "wounded" 

I would be nice if MotionEye noted these problems in the Recovery and warned the user and asked the user for new locations/settings...  Even better if MotionEye noted these problems and mapped them to the new locations in the new system automatically.  Failing that, just backing up the basic Camera settings (without any of the system settings or file location stuff) would also work.. (maybe call that "system port") 

Regardless I spend the morning re-adding all nine of my Cameras, re-configuring, motion masks and streaming ports and retention  schedules and everything is fine now...

MichaIng, 2024-02-02:
Oh, this makes sense indeed. AFAIK, the locations should be the same with a native, DietPi and current Docker container (the ones from [here](https://github.com/motioneye-project/motioneye/pkgs/container/motioneye)). But the old Docker containers, motionEyeOS and of course the HA plugin will can use different file locations. Would make sense to store the config files without path, and copy them into the current system's config path on import, i.e. the one where the actual motionEye instance would write camera configs to.