MichaIng/DietPi #6898: PaperMC is not installing

boterocamilo, 2024-02-01:
#### Details:
- Date           | Thu Feb  1 22:07:55 GMT 2024
- Bug report     | 10727235-4751-4fff-ad46-bb23857ea31f
- DietPi version | v9.0.2 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `curl -sSfL https://ci.opencollab.dev/job/GeyserMC/job/Floodgate/job/master/lastStableBuild/artifact/spigot/build/libs/floodgate-spigot.jar -o floodgate-spigot.jar`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
curl: (22) The requested URL returned error: 404
```

Joulinar, 2024-02-02:
hmm looks like download is broken as the whole `master` directory has been removed by GeyserMC guys https://ci.opencollab.dev/job/GeyserMC/job/Floodgate/

MichaIng, 2024-02-02:
Same with Geyser plugin. The branch still exists, but is marked as "disabled": https://ci.opencollab.dev/job/GeyserMC/job/Geyser/

I have no real insights into the status Minecraft plugins, the Bedrock Edition, etc, so if someone can shad some light on it, that would be good.

The download links here btw fail as well: https://geysermc.org/download

As long as nobody has an alternative idea, I would just remove the Bedrock support dialogue for now. Not sure how widely used/required it is either?

Joulinar, 2024-02-02:
We could ask on their GitHub. At least there the download link is shown as well.

MichaIng, 2024-02-02:
https://github.com/GeyserMC/Geyser/issues

MichaIng, 2024-02-20:
Okay, the download URLs have changed on their download page:
- https://download.geysermc.org/v2/projects/geyser/versions/latest/builds/latest/downloads/spigot
- https://download.geysermc.org/v2/projects/floodgate/versions/latest/builds/latest/downloads/spigot

MichaIng, 2024-02-20:
Fixed with: https://github.com/MichaIng/DietPi/commit/facb80e
Ups, wrong commit name v9.1 => v9.2 😄

closed 2024-03-17 by MichaIng