MichaIng/DietPi #6900: fs_partition_resize failed on RPI4 (Bullseye image)

mschlenstedt, 2024-02-03:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bullseye 0

- Kernel version | `uname -a`

Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

- SD card used | (EG: SanDisk ultra)

16 GB Sandisk

#### Additional Information (if applicable)

-

#### Steps to reproduce

1. Install latest image (bullseye)

#### Expected behaviour

Rootfs should be expanded to maximum size

#### Actual behaviour

The RootFS isn't resiszed during first boot. After initial setup, this is the state:

Try to manually resoze RootFS:
```
root@DietPi:~# /bin/dash -c '/var/lib/dietpi/services/fs_partition_resize.sh 2>&1 | tee -a /var/tmp/dietpi/logs/fs_partition_resize.log'
[ INFO ] Detected root drive /dev/mmcblk0 with root partition 2
[ INFO ] No DietPi setup partition found, last partition is: "ext4 "
lsblk: unknown column: MOUNTPOINTS
```

Partition table on the SD:
```
root@DietPi:~# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 14.84 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xf8a5f4a0

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1 *      2048  264191  262144   128M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      264192 1665903 1401712 684.4M 83 Linux
```

Free space:
```
root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       655M  615M     0 100% /
devtmpfs        1.9G     0  1.9G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           769M  8.6M  761M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.9G     0  1.9G   0% /tmp
tmpfs            50M  8.0K   50M   1% /var/log
/dev/mmcblk0p1  127M   33M   95M  26% /boot
```

MichaIng, 2024-02-03:
Thanks. Was reported as well on the forum and fixed already: https://dietpi.com/forum/t/preventing-root-partition-resizing/19125

Fixed with: https://github.com/MichaIng/DietPi/commit/9e63b8a

I need to generate new images. Did so already, but forgot that, with `master` branch code, I need to hack the fix into the images with the installer, until v9.1 has been released.

mschlenstedt, 2024-02-03:
@MichaIng  Thanks! You are "the best" ;-)

MichaIng, 2024-02-03:
Builds running: https://github.com/MichaIng/DietPi/actions/runs/7760046902/job/21186064555

Joulinar, 2024-02-03:
@MichaIng If I understood correctly, you are building new Bullseye images for all platforms. Right?

MichaIng, 2024-02-04:
Thanks for testing. I moved them to stable downloads.

closed 2024-02-04 by mschlenstedt