MichaIng/DietPi #6901: RemoteIt CLI

oradke, 2024-02-04:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- DietPi allows convenient installation of remoteit, and connections TO the device can easily be managed through the desktop app on a  windows PC. However, the command line interface (CLI) is not installed by default, so OUTGOING connections from dietpi cannot be established.

#### Describe the solution you'd like:
- Install the CLI into /usr/bin when remoteit is installed via dietpi-software. 

#### Describe alternatives you've considered:
I have removed the dietpi version of remoteit and installed manually (following https://docs.remote.it/software/device-package/installation#3.-claim-and-register-the-device). 

#### Additional context
Seems to be working fine, but might break on next dietpi update

MichaIng, 2024-05-07:
Sorry for missing your request so long and thanks for the information. Your link seems to be invalid and pointing to the wrong docs page. Did you mean this one? https://docs.remote.it/software/cli/overview#id-4.-use-the-code-in-command-to-install-the-cli

Basically the agent and the CLI serve the opposite purposes, so I suggest to make the CLI an own install option. Downloading and installing the right binary is trivial. Additionally, we could add a `dietpi.txt` setting to add a registration code, and show a prompt, it it was not set, and do the registration automatically. The installer contains some logic to check whether a registration file exists already, and whether a claim code exists already. We could/should do the same, to in case skip the prompt and repeating the registration. But I do not fully understand it yet: https://downloads.remote.it/cli/install_cli.sh
- If `/etc/remoteit/registration` exists, it stores the contained key as `regfile_licensekey` and sets `regfile_exists=true`. This `regfile_licensekey` variable is however never used.
- If `/etc/remoteit/registration` does not exist, it is created and the passed license key stored in it. The registration itself is however never done based on the content of this file, but always based on the passed input variable. So I wonder whether this file has any effect, or whether it is just there for users to retrieve their license key.
- When the registration is done, `/etc/remoteit/config.json` is created, which contains the claim code, same as when the agent was installed. When this file exists, the registration is skipped entirely, also `/etc/remoteit/registration` would not be generated. Also the claim code is stored in variable, but never used (only to know whether to skip the `/etc/remoteit/registration` creation).
- Checking back the docs, the CLI contains/installs the agent? ... ah, indeed it does `remoteit agent install --yes`. Okay, so it makes sense then:
    - The license key is needed for clients/CLI/desktop to register with the Remote.It infrastructure. It is however not needed to setup an agent (?).
    - Installing the agent generates a claim code, which can be used to connect with clients/CLI/desktop to the agent.
    - So both is basically independent. I wonder whether/why the CLI installer implicitly installs the agent, and why the CLI registration is skipped when an agent config exists. E.g. when the agent is installed first separately, then the CLI, the registration is skipped. Is the agent required for connections in both directions?

In case really the agent is required to use the CLI for outgoing connections, we could still offer both as separate install options, but the CLI depending on the agent. While the CLI installer is incompatible with the agent DEB package, as it implicitly installs the agent again, most likely as raw executables, it overrides the DEB package files, and any DEB package upgrade reverts it. We however can just skip the `remoteit agent install` step and do the CLI registration regardless whether `/etc/remoteit/config.json` exists. I hope it works like that.

oradke, 2024-05-09:
Sounds good! Took me a while to figure out that the remote.it package with dietpi only allows incoming connections and that it breaks when the CLI is manually installed.