MichaIng/DietPi #6905: Using curl gives the error: "curl: error while loading shared libraries: libhogweed.so.6 ..."

John-OK, 2024-02-07:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
buster 1

- Kernel version | `uname -a`
Linux PiHole 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 2 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A Generic

- SD card used | (EG: SanDisk ultra)
Transcend 8GB Class 10 ( TS8GUSDHC10)

#### Additional Information (if applicable)

- Can this issue be replicated on a fresh installation of DietPi?
I'm not sure, but that is my next step if I can't get it resolved.
I restored from a month-old backup if that helps.

#### Steps to reproduce

1. run: `curl -I https://dietpi.com/`

#### Expected behaviour

- Should return headers for dietpi.com

#### Actual behaviour

- Get the following error: "curl: error while loading shared libraries: libhogweed.so.6: cannot open shared object file: No such file or directory"

#### Extra details

I was having issues connecting to the internet that I couldn't resolve, so I restored from a backup from 2023-12-12. 
My connection issues are resolved, but I noticed the curl error on my banner for "WAN IP" and "Weather."
This may have been an issue prior to restoring from backup that I didn't notice because I rarely SSH into this RPi.
I have tried the following:
`sudo apt purge curl `
`sudo apt install --reinstall libhogweed4`
`sudo reboot`
`sudo apt install curl -y`

MichaIng, 2024-02-07:
First of all: You are using a Debian version which is not fully supported anymore, and also not the newest DietPi version available on that Debian version. It makes sense to first update your system to the latest version, before doing bug reports:
```sh
dietpi-update
```

The `curl` binary shipped by Debian/Raspbian Buster depends on `libhogweed4`, which as installed accordingly already. `libhogweed6` is available only from Bullseye on. As your `curl` did depend on `libhogweed6`, it is a wrong binary, likely installed from Debian or Raspbian Bullseye. Probably you attempted but did not complete an upgrade to Bullseye earlier?
```sh
grep -r bullseye /etc/apt/
```

John-OK, 2024-02-16:
I've run `dietpi-update` each time I get the update notification, so I thought I was fully updated to the latest release, but I guess Buster isn't the latest. 
I ran `grep -r bullseye /etc/apt/` but it didn't return anything (because I'm still on Buster?)
I'll take a look at the link you provided to update to Bullseye or Bookworm and see if that's possible.

Joulinar, 2024-02-16:
Destro upgrades would need to be done manually according our blog post. It will never be done by `dietpi-update`

MichaIng, 2024-02-17:
I am still wondering where the `curl` binary with `libhogweed6` dependency was coming from. However, let's assume this is a single case, as long as we do not get a 2nd such report.

John-OK, 2024-02-20:
Waiting on a SD card to install the Bullseye on. Will let you know once I get it up.

John-OK, 2024-02-21:
Oops. That's what I meant: Bookworm - the latest.

John-OK, 2024-05-29:
Sorry for taking so long to update this issue.
I finally did a fresh install of Bookworm and installed all my apps, and everything is working perfectly.
Thanks for all the help and a HUGE thank you for all the work you do on DietPi!!!

John-OK, 2024-05-29:
I installed Bookworm and was able to load my apps. Everything is running perfectly as DietPi usually does!
Thanks so much for your help and a HUGE thank you for all the work you do keeping DietPi going!!!
Sorry for closing and reopening the issue. I wasn't sure what I was doing.
You can close this issue.

closed 2024-05-29 by John-OK