MichaIng/DietPi #6910: dietpi-update fails (as does apt-get update) on Pine64, with "Connection failed" unexpectedly

carstenraddatz, 2024-02-12:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Details:
- Date           | Sun Feb 11 20:18:16 CET 2024
- Bug report     | 0f492397-e7cb-4959-b998-013c99ff9b82
- DietPi version | v8.12.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux pineblock 5.15.80-sunxi64 #22.11.1 SMP Wed Nov 30 11:23:00 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y update`
- Exit code      | 255
- Software title | DietPi-Backup
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 http://ftp.de.debian.org/debian bullseye InRelease [116 kB]
Get:2 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:4 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [262 kB]
Get:5 https://download.docker.com/linux/debian bullseye InRelease [43.3 kB]
Get:6 https://download.docker.com/linux/raspbian bullseye InRelease [26.6 kB]
Get:7 https://download.docker.com/linux/debian bullseye/stable arm64 Packages [30.7 kB]
Err:1 http://ftp.de.debian.org/debian bullseye InRelease
  Connection timed out [IP: 141.76.2.4 80]
Err:3 http://mirrors.xtom.de/armbian bullseye InRelease
  Connection failed [IP: 62.133.35.14 80]
Err:8 http://ftp.de.debian.org/debian bullseye-updates InRelease
  Connection failed [IP: 141.76.2.4 80]
Err:9 http://ftp.de.debian.org/debian bullseye-backports InRelease
  Connection failed [IP: 141.76.2.4 80]
Fetched 411 kB in 2min 31s (2726 B/s)
Reading package lists...
W: Failed to fetch http://ftp.de.debian.org/debian/dists/bullseye/InRelease  Connection timed out [IP: 141.76.2.4 80]
W: Failed to fetch http://ftp.de.debian.org/debian/dists/bullseye-updates/InRelease  Connection failed [IP: 141.76.2.4 80]
W: Failed to fetch http://ftp.de.debian.org/debian/dists/bullseye-backports/InRelease  Connection failed [IP: 141.76.2.4 80]
W: Failed to fetch http://apt.armbian.com/dists/bullseye/InRelease  Connection failed [IP: 62.133.35.14 80]
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

#### Steps to reproduce
1. dietpi-update
2. select "update"

#### Expected behaviour
- upgrade should run and complete

#### Actual behaviour
- several "Connection failed", see above
- update does not pull all index files it should
- exit code 255 results

#### Extra details
- the IP addresses in question are reachable; the servers exist and are up
- setting different Armbian mirror addresses changes the target server, but still no success
- using the dietpi-update script as well as "raw" apt update shows the same error
- in an attempt to address more obscure issues I reconfigured the MTU value using the  it is set to 1360 now (DOCSIS connection)
- other than updates all services run normally on this host (pi-hole, docker and other)
- solving this is low priority maybe, I could just swap the install on this Pine64 board to a new version to workaround

Joulinar, 2024-02-12:
I see all of them are http connections. Maybe they got blocked somewhere. You could try to switch to https instead

carstenraddatz, 2024-02-12:
Good catch! I changed the connections to https, now I'm getting 11: Resource temporarily unavailable. Will change the mirrors for Armbian, it seems a problem there.

carstenraddatz, 2024-02-12:
That fixed the issue. Not a DietPi problem at all, sorry for the noise. (Took a few runs until an Armbian mirror was picked that did deliver the stuff needed.)

closed 2024-02-12 by carstenraddatz

Joulinar, 2024-02-12:
It's not the first time an Armbian apt mirror has been a little fussy. Good you solved it.

MichaIng, 2024-02-17:
Good to know that `ftp.de.debian.org` finally supports HTTPS as well. Not sure why it actually failed with plain HTTP, but HTTPS is better anyway.

... although, it does not work in my case 🤔:
```
Err:1 https://ftp.de.debian.org/debian trixie InRelease
  Certificate verification failed: The certificate is NOT trusted. The name in the certificate does not match the expected.  Could not handshake: Error in the certificate verification. [IP: 141.76.2.4 443]
Fetched 23.6 kB in 8s (2963 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: Failed to fetch https://ftp.de.debian.org/debian/dists/trixie/InRelease  Certificate verification failed: The certificate is NOT trusted. The name in the certificate does not match the expected.  Could not handshake: Error in the certificate verification. [IP: 141.76.2.4 443]
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
@carstenraddatz you really use `https://ftp.de.debian.org` now and it does not throw this TLS certificate warning?

With plain HTTP it works fine here:
```
Get:1 http://ftp.de.debian.org/debian trixie InRelease [157 kB]
Hit:2 https://archive.raspberrypi.com/debian bookworm InRelease
Get:3 http://ftp.de.debian.org/debian trixie/main armhf Packages [8762 kB]
Fetched 8919 kB in 9s (960 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
Confusing.

And regarding Armbian. Did you use `http://` or `https://` in your `/etc/apt/souces.list.d/*armbian.list`? We actually changed it to HTTPS during DietPi v8.11 update ... ah, I see a problem there: We did this for `armbian.list`, but later, we added a `pre-patch` to rename it to `dietpi-armbian.list`. So the patch for HTTPS does not work anymore. I'll fix this. Hence an issue our end, not one with Armbian.

carstenraddatz, 2024-02-18:
So help clear things up, as pasted, my log there showed no TLS certificate warning. But see the `Get:1/Err:1` pair, it shows a  timeout error for `ftp.de.debian.org`, nothing certificate related. (In many attempts I did get the unspecific "Connection failed" result, too, again nothing TLS related.) But you knew that.

TBH, I don't even recall where I changed http to https in my attempts. At the moment I have 

```
$ sudo cat /etc/apt/sources.list.d/*armbian.list
deb https://apt.armbian.com/ bullseye main
```
On update attempts I'm getting a curl: (56) OpenSSL SSL_read: Connection timed out, errno 110:

│ Downloading update archive                                                                                        │
│  - Command: curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz                                  │
│  - Exit code: 56

Might this be an MTU issue after all? The file downloads normally on any other machine I tried, on cable ISP or not.

FWIW The DietPi-Config dialogue itself offers to set `https://ftp.debian.org/debian` as the "Global" mirror. The upstream certificate policy apparently does not trickle down nicely (:

carstenraddatz, 2024-02-19:
The MTU of 1360 was found by the built-in tool earlier, about 15 months ago. Now I followed the MTU hunch doing a repetive `ping -4 -f github.com -M do -c4 -i1 -s <value>`, and any packetsize higher than `1290` brought errors for me, what gives. (Jumped down from 1360 in increments of 20, not 1 as `dietpi-optimal_mtu` tool does, this was quick.)

To make it survive a reboot I put it into `/etc/dhcp/dhclient.conf` near the top:

```
interface "eth0" {
default interface-mtu 1290;
supersede interface-mtu 1290;
}
```

This is getting ut of scope, my Pine64 has that wifi card installed. Downloading anything using the `wlan0` interface works without a fuss. Just change the metric in routing, dietpi-updates runs successfully.

Now I have _two_ ways of solving the original issue. Thank you for your time.

MichaIng, 2024-02-19:
Okay great. `/etc/dhcp/dhclient.conf` seems to be a common method to set this permanently when DHCP is used. Another way is using `mtu 1290` in `/etc/network/interfaces`. respectively `/etc/network/interfaces.d`.