MichaIng/DietPi #6914: Open Beta v9.1 | Please help testing and hardening our upcoming release

MichaIng, 2024-02-15:
RC version | v9.1.0
----------- | -------
v9.1.0 PR | #6913
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2024-01-17**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.1
____
### Beta v9.1.0
_(2024-02-15)_

#### New images
- Raspberry Pi 5 | Images for Raspberry Pi 5 and other Raspberry Pi models, based on the new Bookworm kernel and firmware package set from Raspberry Pi Ltd, are now available for testing. Note that some features, like changing screen resolution and camera module support, are not working yet: https://dietpi.com/downloads/binaries/testing/
- ROCK 4 SE | Since our Radxa ROCK 4 image has become incompatible with the ROCK 4 SE variant, we provide a new image for this particular variant. Many thanks to `@janno` for reporting failing boot on this SBC: https://dietpi.com/forum/t/radxa-rock-4-se-supported-not-booting/19263

#### Enhancements
- DietPi-Software | Domoticz: Enabled for x86_64 Bookworm and Trixie systems, since the latest x86_64 builds were compiled against libssl3.
- DietPi-Software | vaultwarden: Along with a regular update, RISC-V support has been added.
- DietPi-Software | Home Assistant: The Python version installed with Home Assistant has been raised to 3.12.1, to align with official appliances. Many thanks to @whyisthisbroken for making us aware if it: https://github.com/MichaIng/DietPi/issues/6906

#### Bug fixes
- DietPi-FS_partition_resize | Resolved an issue where on Bullseye images, the root partition and filesystem was not expanded on first boot. Many thanks to `@naddel91` and others for reporting this issue: https://dietpi.com/forum/t/increase-partition-size/19015
- DietPi-Software | Mosquitto: Resolved an issue where the global software password was accidentally applied as username instead of as password: https://github.com/MichaIng/DietPi/issues/6886
- DietPi-Software | Amiberry: Resolved an issue where the installation failed on 64-bit RPi systems. Many thanks to @JerichoCross for reporting this issue: https://github.com/MichaIng/DietPi/issues/6893
- DietPi-Software | GMediaRender: Resolved an issue where the default command arguments were not set correctly on a fresh install. Many thanks to `@ransur0t` for reporting this issue: https://dietpi.com/forum/t/gmediarender-default-args-on-allo-image-for-raspberry-pi/19205
- DietPi-Software | Samba Server: Resolved an issue where our configuration file was not installed if the Samba Client was installed as well, or an SMB/CIFS mount done. Many thanks to `@TheCrystalGipsy` for reporting this issue: https://dietpi.com/forum/t/samba-share-issues-need-help/19224
- DietPi-Software | Pydio: Resolved an issue where web access via Apache webserver did not work, due to an invalid config entry. However, Pydio 8 is not maintained since years, does not support PHP 8 and hence can only work on Bullseye systems. We will migrate to Pydio Cells, once someone finds time to work on this: https://github.com/MichaIng/DietPi/issues/3469

whyisthisbroken, 2024-02-15:
After updating and reinstall Home Assistant, i´ve got Errors.
Any Ideas ?
Evrytime i restart HA, webrtc-nois-gain runs again and fail...

Logger: homeassistant.setup
Source: setup.py:221
First occurred: 00:08:55 (4 occurrences)
Last logged: 00:09:01

Setup failed for 'cloud': Requirements for assist_pipeline not found: ['webrtc-noise-gain==1.2.3'].
Setup failed for 'mobile_app': Requirements for assist_pipeline not found: ['webrtc-noise-gain==1.2.3'].
Setup failed for 'assist_pipeline': Requirements for assist_pipeline not found: ['webrtc-noise-gain==1.2.3'].
Setup failed for 'default_config': (DependencyError(...), 'Could not setup dependencies: assist_pipeline, cloud, mobile_app')

Logger: homeassistant.setup
Source: setup.py:195
First occurred: 00:09:01 (1 occurrences)
Last logged: 00:09:01

Unable to set up dependencies of 'default_config'. Setup failed for dependencies: assist_pipeline, cloud, mobile_app

[image]

`Logger: homeassistant.util.package
Source: util/package.py:122
First occurred: 00:12:03 (3 occurrences)
Last logged: 00:12:48

Unable to install package webrtc-noise-gain==1.2.3: error: subprocess-exited-with-error × Building wheel for webrtc-noise-gain (pyproject.toml) did not run successfully. │ exit code: 1 ╰─> [77 lines of output] running bdist_wheel running build running build_py creating build creating build/lib.linux-aarch64-cpython-312 creating build/lib.linux-aarch64-cpython-312/webrtc_noise_gain copying webrtc_noise_gain/__init__.py -> build/lib.linux-aarch64-cpython-312/webrtc_noise_gain running build_ext gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -fPIC -I/home/homeassistant/.pyenv/versions/3.12.1/include/python3.12 -c -std=c++17 flagcheck.cpp -o flagcheck.o -std=c++17 gcc: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory compilation terminated. gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -fPIC -I/home/homeassistant/.pyenv/versions/3.12.1/include/python3.12 -c -std=c++17 flagcheck.cpp -o flagcheck.o -std=c++14 gcc: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory compilation terminated. gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O3 -Wall -fPIC -I/home/homeassistant/.pyenv/versions/3.12.1/include/python3.12 -c -std=c++17 flagcheck.cpp -o flagcheck.o -std=c++11 gcc: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory compilation terminated. Traceback (most recent call last): File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module> main() File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main json_out['return_val'] = hook(**hook_input['kwargs']) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/mnt/dietpi_userdata/homeassistant/deps/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel return _build_backend().build_wheel(wheel_directory, config_settings, ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 404, in build_wheel return self._build_with_temp_dir( ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 389, in _build_with_temp_dir self.run_setup() File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 480, in run_setup super().run_setup(setup_script=setup_script) File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 311, in run_setup exec(code, locals()) File "<string>", line 563, in <module> File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 103, in setup return distutils.core.setup(**attrs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 185, in setup return run_commands(dist) ^^^^^^^^^^^^^^^^^^ File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 201, in run_commands dist.run_commands() File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands self.run_command(cmd) File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 963, in run_command super().run_command(command) File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command cmd_obj.run() File "/tmp/pip-build-env-mx5gm85a/normal/lib/python3.12/site-packages/wheel/bdist_wheel.py", line 368, in run self.run_command("build") File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command self.distribution.run_command(command) File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 963, in run_command super().run_command(command) File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command cmd_obj.run() File "/tmp/pip-build-env-mx5gm85a/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/build.py", line 131, in run self.run_command(cmd_na [... cut ...]
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/requirements.py", line 249, in async_process_requirements
    self._raise_for_failed_requirements(name, missing)
  File "/home/homeassistant/.pyenv/versions/3.12.1/lib/python3.12/site-packages/homeassistant/requirements.py", line 278, in _raise_for_failed_requirements
    raise RequirementsNotFound(integration, [req])
homeassistant.requirements.RequirementsNotFound: Requirements for assist_pipeline not found: ['webrtc-noise-gain==1.2.3'].
`

Joulinar, 2024-02-15:
> `cannot execute ‘cc1plus’: execvp: No such file or directory compilation terminated`

probably you need to install `cc1plus` in addition

whyisthisbroken, 2024-02-15:
Mhm, i dont know this application "cc1plus" - what part of installation should it try ?

MichaIng, 2024-02-15:
```
apt install g++
```

whyisthisbroken, 2024-02-15:
hey, thanks - yeah i´ve found it trough google. i updatet my post above 5sec earlier.  now it seems to go forward...
installation progress is working. Thanks :)

Joulinar, 2024-02-16:
@MichaIng wiringpi is gone, should we remove it? http://wiringpi.com/index.html

MichaIng, 2024-02-17:
Also the WiringPi repository has been archived with a related note that it broke with Raspberry Pi OS Bookworm and Raspberry Pi 5: https://github.com/WiringPi/WiringPi

I'll test it, just in case. For Odroids, we use Hardkernel's fork: https://github.com/hardkernel/wiringPi
I'll also test whether Hardkernel's fork even works on Odroid N2 with mainline kernel. For Odroid XU4, the kernel is still heavily based on Hardkernel's one, but even if it works there, no point to keep it for a single old SBC, IMO. At some point, things should all move to `libgpiod`.

Alexander-Mundy, 2024-02-18:
Forgive me if this is not the correct place and please direct me if not.
I have followed the script to convert to DietPi on a Pi4B and moved it to a Pi5.
I found that the option to not wait on boot for network was not working so I changed to the beta to see if it worked.
Unfortunatly, on 2 different Pi5 with 2 different NVME SSD using 2 Pineberry Pi HatDrive! Bottoms (only mention in case it applies) they hold up boot until network connection unless I disable WiFi.
These are intended for use in a car so will not have network except when home.
Using the Raspi 64 bit image I can avoid the network wait on boot, but cannot figure out how to get it to work with DietPi.
I do have the no option selected in DietPi Config and also disabled network time sync.
Any help would be appreciated because I really like DietPi and is my choice on Pi4B.
Here is some information, let me know what else you would like to see.

root@DietPi:~# systemd-analyze critical-chain
The time when unit became active or started is printed after the "@" character.
The time the unit took to start is printed after the "+" character.

graphical.target @41.066s
└─multi-user.target @41.066s
  └─getty.target @41.066s
    └─getty@tty1.service @41.063s
      └─systemd-user-sessions.service @41.056s +5ms
        └─network.target @41.028s
          └─ifup@wlan0.service @1.451s +39.576s
            └─network-pre.target @1.448s
              └─dietpi-preboot.service @1.414s +33ms
                └─dietpi-ramlog.service @1.357s +32ms
                  └─basic.target @1.335s
                    └─sockets.target @1.334s
                      └─uuidd.socket @1.334s
                        └─sysinit.target @1.333s
                          └─systemd-timesyncd.service @1.291s +42ms
                            └─systemd-tmpfiles-setup.service @1.276s +12
ms
                              └─systemd-journal-flush.service @1.268s +7
ms
                                └─var-log.mount @1.261s +2ms
                                  └─local-fs-pre.target @354ms
                                    └─systemd-tmpfiles-setup-dev.service @344ms +10ms
                                      └─systemd-sysusers.service @328ms +15ms
                                        └─systemd-remount-fs.service @272ms +26ms
                                          └─systemd-fsck-root.service @223ms +48ms
                                            └─systemd-journald.socket @204ms
                                              └─-.mount @166ms
                                                └─-.slice @166ms

RPi 5 Model B (aarch64) | IP: 192.168.0.30

┌──────────────────────────────┤ DietPi-Config ├────────────
│                                                                              │
│                                ●─ Mirror Options ─────────────────●          │
│          APT Mirror            : [https://deb.debian.org/debian/]            │
│          NTP Mirror            : [debian.pool.ntp.org]                       │
│                                ●─ Connection Test Options ────────●          │
│          Boot wait for network : [Off]                                       │
│          Test timeout          : [10 seconds]                                │
│          Test attempts         : [2 tries]                                   │
│          Test IPv4 address     : [9.9.9.9]                                   │
│          Test IPv6 address     : [2620:fe::fe]                               │
│          Test domain name      : [dns9.quad9.net]                            │
│                                ●─ Additional Options ─────────────●          │
│          Network Drives        : Mount/control network storage               │
│          Dynamic DNS           : Manage domains for your dynamic IP          │
│                                                                              │
│                                                                              │
│                     <Ok>                         <Back>                      │
│                                                                              │
└────────────────────────────────────────────────────────

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[... 7 lines cut ...]
[ INFO ] DietPi-Update | Latest version  : v9.1.0
[ INFO ] DietPi-Update | Checking for new available live patches
 - Press any key to return to DietPi-Launcher...

RPi 5 Model B (aarch64) | IP: 192.168.0.30

┌──────────────────────────────┤ DietPi-Config ├───────────────
│ Please select an option:                                                     │
│                                                                              │
│    Swap file                  : [Off | /var/swap]                            │
│    APT                        : Manage APT cache and list storage            │
│    Time sync mode             : [Custom]                                     │
│    RTC mode                   : [Emulated]                                   │
│    Update RPi EEPROM firmware : [Bootloader: 1704470260 | VL805: N/A]        │
│    Serial/UART                : Manage available devices                     │
│    Bluetooth                  : [Off]                                        │
│    I2C state                  : [Off]                                        │
│    I2C frequency              : [100 kHz]                                    │
[... cut ...] ]                                        │
│    RPi kernel choice          : Select/deselect optional kernel packages     │
│                                                                              │
│                                                                              │
│                     <Ok>                         <Back>                      │
│                                                                              │
└───────────────────────────────────────────────────

MichaIng, 2024-02-18:
Hmm, it seems that systemd hardcoded this:
```console
root@VM-Bullseye:~# systemctl cat getty@.service | grep After
After=systemd-user-sessions.service plymouth-quit-wait.service getty-pre.target
After=rc-local.service
root@VM-Bullseye:~# systemctl cat systemd-user-sessions.service | grep After
After=remote-fs.target nss-user-lookup.target network.target home.mount
RemainAfterExit=yes
```
So the `systemd-user-sessions` service waits for network, and the local getty (login prompt) waits for `systemd-user-sessions`. Seems odd to me, as I see no reason why generally network, and also DNS resolution and even remote filesystem mounts would be required for a local user login.

This indeed breaks the point of the "Boot wait for network". This controls whether `/boot/dietpi/postboot` runs after `network-online.target` or not. That scripts triggers the DietPi update check or network time sync check, which require network anyway. But these checks fork into background, with internal loops. hence do not delay further execution of the script. It also triggers the pre-login banner (the one which shows the LAN IP and "Please hit <return> to login"), which does not make sense if the actual login prompt is not ready yet. The only thing it does where the toggle still has a reasonable effect is that it starts scripts in `/var/lib/dietpi/postboot.d/`. I think it makes sense to de-tangle/split these tasks and in case bind them to the actual service/target they require. But this is something for a next DietPi version.

You could remove `/etc/systemd/system/ifup@.service.d/dietpi.conf` so that `ifup@wlan0.service` "succeeds" immediately, but that again breaks the purpose and meaning of `network.target` and `network-online.target`, as those are then reached immediately, even if network is not yet up, and even if it fails completely.

Alexander-Mundy, 2024-02-18:
Thank you, (moved the file so I still have it) and that worked.
wlan0 came up and did not stop boot while it was coming up.
I have updates and time sync disabled and will use the RTC.
I am not a proficient coder, but will see if I can make a script that will toggle the updates on and off.
Since it will be used for a car dash display and control of the ECU it needs to boot as fast as possible whether connected or not.

root@DietPi:~# systemd-analyze critical-chain
The time when unit became active or started is printed after the "@" character.
The time the unit took to start is printed after the "+" character.

graphical.target @1.519s
└─multi-user.target @1.519s
  └─getty.target @1.519s
    └─getty@tty1.service @1.518s
      └─systemd-user-sessions.service @1.511s +5ms
        └─network.target @1.509s
          └─networking.service @1.432s +76ms
            └─network-pre.target @1.414s
              └─dietpi-preboot.service @1.379s +34ms
                └─dietpi-ramlog.service @1.350s +27ms
                  └─basic.target @1.347s
                    └─sockets.target @1.346s
                      └─uuidd.socket @1.346s
                        └─sysinit.target @1.346s
                          └─systemd-timesyncd.service @1.303s +42ms
                            └─systemd-tmpfiles-setup.service @1.290s +11ms
                              └─systemd-journal-flush.service @1.282s +8ms
                                └─var-log.mount @1.274s +2ms
                                  └─local-fs-pre.target @356ms
                                    └─systemd-tmpfiles-setup-dev.service @347ms +9ms
                                      └─systemd-sysusers.service @329ms +17ms
                                        └─systemd-remount-fs.service @273ms +38ms
                                          └─systemd-fsck-root.service @228ms +43ms
                                            └─systemd-journald.socket @209ms
                                              └─system.slice @169ms
                                                └─-.slice @169ms

MichaIng, 2024-02-19:
#### Beta v9.1.1 has been merged: #6920
I'll do a last review and likely release v9.1.1 afterwards.

MichaIng, 2024-02-19:
### DietPi v9.1 has been released: #6921
Many thanks to all testers ❤️ !

closed 2024-02-19 by MichaIng