MichaIng/DietPi #6915: Octoprint - sudoers.d changes required

samjw-nz, 2024-02-16:
### Creating a bug report/issue
- [O] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
`G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bookworm 1`
- Kernel version | `uname -a`
`Linux OctoPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W (armv6l)
Native PC UEFI
- Power supply used | (EG: 5V 1A RAVpower)
Any
- SD card used | (EG: SanDisk ultra)
Any

#### Additional Information (if applicable)
- Software title | Octoprint
- Was the software title installed freshly or updated/migrated? | Fresh install
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... Fresh or existing install of DietPi
2. ... Install Octoprint from dietpi-software
3. ... Go through initial Octoprint setup as standard
4. ... Attempt reboot, poweroff, or Octoprint restart from Octoprint page

#### Expected behaviour

- ... Reboot, Poweroff, or Octoprint restart

#### Actual behaviour

- ... Confirmation message in Octoprint but no action performed

#### Extra details

- ... Result of sudo command
`/etc/sudoers.d/octoprint:1:63: expected a fully-qualified path name
octoprint ALL=NOPASSWD: /usr/bin/systemctl restart octoprint, reboot, poweroff`

- ... modification of octoprint file in /etc/sudoers.d has resolved issue
ORIGINAL: `octoprint ALL=NOPASSWD: /usr/bin/systemctl restart octoprint, reboot, poweroff`
MODIFIED: `octoprint ALL=NOPASSWD: /usr/bin/systemctl restart octoprint, /usr/bin/systemctl reboot, /usr/bin/systemctl poweroff`

MichaIng, 2024-02-17:
How long ago did you install OctoPrint? Not sure whether/when we changed it, but what we do is:
```sh
echo "octoprint ALL=NOPASSWD: $(command -v systemctl) restart octoprint, $(command -v reboot), $(command -v poweroff)" > /etc/sudoers.d/octoprint
```
Hence the `reboot` and `poweroff` commands are already added with full path. `systemctl reboot` etc is not required and would not work as long as you do not also manually define this as `reboot`/`poweroff` commands. On our OctoPrint installation, we however use `sudo reboot`/`sudo poweroff`, matching the `sudoers` rule.

samjw-nz, 2024-02-17:
Both installs were this week, with the currently available images from the website. My RPI ZeroW is free so I'm happy to do a fresh install or troubleshoot from my side.

MichaIng, 2024-02-17:
Ah, dammit, now I know the problem: Since we added shell functions for `reboot`, `poweroff` and `halt`, to avoid logind calls as long as logind is not running, `command -v` gives the wrong output, i.e. not the path but the command only, as it sees the shell function.

Fixed with: https://github.com/MichaIng/DietPi/commit/4cb88f8

closed 2024-02-20 by MichaIng