MichaIng/DietPi #6916: [FAILED] DietPi-Update

Bluehmchen, 2024-02-16:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Feb 16 12:13:41 CET 2024
- Bug report     | 894b97e1-18b3-42b8-bcdf-258be6742e89
- DietPi version | v8.22.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux rp4b 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 https://deb.debian.org/debian bullseye InRelease [116 kB]
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Joulinar, 2024-02-16:
can you share full output of

```
apt update
```

Bluehmchen, 2024-02-16:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.0.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.22.3
[ INFO ] DietPi-Update | Latest version  : v9.0.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=24301 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : node-red
[  OK  ] DietPi-Services | stop : mosquitto
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Reverting now obsolete rpi-eeprom hold
[  OK  ] DietPi-Pre-patches | apt-mark unhold rpi-eeprom
[ INFO ] DietPi-Pre-patches | Migrating from raspberrypi.org to raspberrypi.com
[  OK  ] DietPi-Pre-patches | sed -i s/raspberrypi.org/raspberrypi.com/ /etc/apt/sources.list /boot/dietpi.txt
[  OK  ] DietPi-Pre-patches | sed -i s/raspberrypi.org/raspberrypi.com/ /etc/apt/sources.list.d/dietpi-mosquitto.list /etc/apt/sources.list.d/raspi.list /etc/apt/sources.list.d/tvheadend.list
[  OK  ] DietPi-Pre-patches | mkdir -p ../DietPi-Update/DietPi-master/dietpi
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Hit:4 https://archive.raspberrypi.com/debian bullseye InRelease
Hit:5 https://repo.mosquitto.org/debian bullseye InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Get:7 http://apt.tvheadend.org/stable raspbian-stretch InRelease
Err:7 http://apt.tvheadend.org/stable raspbian-stretch InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Reading package lists...
E: Failed to fetch http://apt.tvheadend.org/stable/dists/raspbian-stretch/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'http://apt.tvheadend.org/stable raspbian-stretch InRelease' is no longer signed.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y -eany update
root@rp4b:/tmp/DietPi-Update#

Bluehmchen, 2024-02-16:
ok, I have:
tvheadend removed
/etc/apt/sources.list.d/tvheadend.list deleted

Then it worked ;-)

Joulinar, 2024-02-16:
Yes there was quite an old repository added. I will close this issue for now.