MichaIng/DietPi #6919: [FAILED] DietPi-Software | APT install openjdk-8-jre-headless

maartenlangeveld, 2024-02-17:
```
#### Details:
- Date           | Sat Feb 17 21:32:58 CET 2024
- DietPi version | v9.0.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux M-MUSIC-PLAYER 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `apt-get -y --allow-change-held-packages install openjdk-8-jre-headless`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ca-certificates-java : Breaks: openjdk-8-jre-headless (< 8u382~b04-2~) but 8u312-b07-1+rpi1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2024-02-17:
Many thanks for reporting. Looks like an inconsistency in the Raspbian repository, where `ca-certificates-java` and `openjdk-8-jre-headless` versions do not match.

MichaIng, 2024-02-17:
Indeed it is: http://raspbian.raspberrypi.org/raspbian/dists/bookworm/main/binary-armhf/Packages.xz
```
Package: ca-certificates-java
Version: 20230710~deb12u1
Installed-Size: 43
Maintainer: Debian Java Maintainers <pkg-java-maintainers@lists.alioth.debian.org>
Architecture: all
Depends: ca-certificates (>= 20210120)
Breaks: ..., openjdk-8-jre-headless (<< 8u382~b04-2~)
...
Package: openjdk-8-jre-headless
Source: openjdk-8
Version: 8u312-b07-1+rpi1
```
This fits to the issue I reported regarding such inconsistencies in the Trixie/testing suite: https://bugs.launchpad.net/raspbian/+bug/2052306
I however do not remember that such ever happened on the stable suite as well 🤔.

MichaIng, 2024-03-19:
They do care in general, but the maintainers are also just volunteers, hence time is limited and tasks are a lot.

But I should probably make an own bug report from this, as I just mentioned it within a linked one.

maartenlangeveld, 2024-03-19:
Hi @MichaIng ,

Thanks, good you remind me that they are volunteers. You are absolutely right, good to realise (again) we are lucky with people like you and Raspbian volunteers that offer us so much !! Sometimes I'm a bit impatient. Forgive me.

Joulinar, 2024-04-05:
If I understood correctly, issue is with Raspbian repository and not with DietPi.

GeoffState, 2024-04-05:
Thanks, not a bother for me.
Best Regards
Geoff

MichaIng, 2024-11-08:
Own bug report done: https://bugs.launchpad.net/raspbian/+bug/2087632

maartenlangeveld, 2024-11-15:
Bump:  is there any progress?

Joulinar, 2024-11-15:
Simply check the bug report we open with RPI guys. As you can see, it is still not assigned.

MichaIng, 2024-11-21:
To be precise: "Raspbian" guys/bug tracker, which is a project independent of Raspberry Pi Ltd.

MichaIng, 2025-01-07:
We are disabling Java and all Java application install options with next release. It has become a maintenance burden since a long time, and neither Raspbian nor other RPi user seem to care about this issue, which exists since a year already. So I take from this, the interest to run Java apps on these old RPi models is minimal, not worth the efforts. The fact that Java 8 was the latest possible version, which is not supported anymore by a lot of apps (needing us to install older versions), is another limitations, satisfying this step, also for security reasons at some point: https://github.com/MichaIng/DietPi/commit/f1833d8

We also want to switch to Adoptium Java builds, allowing us to install modern Java versions on older Debian versions, which of course is all not supported on ARMv6 either: #4505

closed 2025-01-07 by MichaIng

maartenlangeveld, 2025-01-07:
What a pity. I need Java on my RPi0 (minimserver). Is there an option to block DietPi updates so I do not accidentalltly upgrade and am not bothered by upgrade warnings?By the way, Java 8 was not the latest possible version. I for example installed version 17 using an armhf build by another provider which name I forgot (not at home at the moment to check,).Thanks, Maarten

MichaIng, 2025-01-07:
You can set the related package on hold. I was already checking whether an older version is still in the repo, but sadly Raspbian gets cleaned up of old packages pretty fast:
```sh
apt-mark hold ca-certificates-java
```
Packages won't be uninstalled or any such on DietPi update, so a functional Java will remain functional.

The problem is not Java above 8, but JVM above 8. There is the alternative "Zero VM": https://packages.debian.org/openjdk-17-jre-zero
I tried it once, but it is crazy slow. As Java apps are already very resource intensive and slow (due to runtime), multiplied on slow SBCs like those old RPi models, this Zero VM rendered any of them completely unusable, taking like hours to start up.

At least newer versions of this package like `openjdk-17-jre-zero` are available on Raspbian now as well. The `openjdk-11-jre-zero` was missing: https://bugs.launchpad.net/raspbian/+bug/1998136

MichaIng, 2025-08-14:
FYI: We are re-enabling Java support for ARMv6. The Raspberry Pi devs started to provide Java 17 builds with the RPi repository to my surprise, and they really do work, and not too slow like with the Zero VM.

But no Java 21 (so far), hence at least openHAB and Minecraft versions are capped: #7667