MichaIng/DietPi #6922: Can't update Prowlarr using WebUI

gioxx, 2024-02-19:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye
- Kernel version | `uname -a`
Linux Jackett 5.10.0-28-amd64 #1 SMP Debian 5.10.209-2 (2024-01-31) x86_64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Virtual Machine (x86_64)

#### Steps to reproduce

Using Admin Web Interface of Prowlarr, I'm trying to update the software.

#### Expected behaviour

- Prowlarr updated to the latest release available.

#### Actual behaviour

- Can't finalize Prowlarr update. I have not already tried to reinstall the software because I wan't to contribute to debug the problem and solve it without re-installing the software, if it's possible.

#### Extra details

```
Feb 19 12:25:57 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Deleting old update files
Feb 19 12:25:58 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Downloading update 1.13.3.4273
Feb 19 12:26:01 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Verifying update package
Feb 19 12:26:01 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Update package verified successfully
Feb 19 12:26:01 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Extracting Update package
Feb 19 12:26:03 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Update package extracted successfully
Feb 19 12:26:03 Jackett Prowlarr[14913]: [Info] BackupService: Starting Backup
Feb 19 12:26:04 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Preparing client
Feb 19 12:26:17 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Starting update client /tmp/prowlarr_update/Prowlarr.Update
Feb 19 12:26:17 Jackett Prowlarr[14913]: [Info] InstallUpdateService: Prowlarr will restart shortly.
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Main process exited, code=killed, status=9/KILL
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Failed with result 'signal'.
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Unit process 15130 (Prowlarr.Update) remains running after unit stopped.
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Consumed 14.157s CPU time.
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 2.
Feb 19 12:26:59 Jackett systemd[1]: Stopped Prowlarr (DietPi).
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Consumed 14.168s CPU time.
Feb 19 12:26:59 Jackett systemd[1]: prowlarr.service: Found left-over process 15130 (Prowlarr.Update) in control group while starting unit. Ignoring.
Feb 19 12:26:59 Jackett systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Feb 19 12:26:59 Jackett systemd[1]: Started Prowlarr (DietPi).
Feb 19 12:26:59 Jackett Prowlarr[15152]: [Info] Bootstrap: Starting Prowlarr - /opt/prowlarr/Prowlarr - Version 1.12.2.4211
Feb 19 12:26:59 Jackett Prowlarr[15152]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 19 12:26:59 Jackett Prowlarr[15152]: [Debug] Bootstrap: Console selected
Feb 19 12:26:59 Jackett Prowlarr[15152]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/prowlarr/prowlarr.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3;busytimeout=100 ***
Feb 19 12:27:00 Jackett Prowlarr[15152]: SQLite notice (283): recovered 7 frames from WAL file /mnt/dietpi_userdata/prowlarr/prowlarr.db-wal
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Debug] MigrationController: Took: 00:00:00.1061747
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/prowlarr/logs.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3;busytimeout=100 ***
Feb 19 12:27:00 Jackett Prowlarr[15152]: SQLite notice (283): recovered 6 frames from WAL file /var/log/prowlarr/logs.db-wal
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Info] Microsoft.Hosting.Lifetime: Now listening on: http://[::]:9696
Feb 19 12:27:00 Jackett Prowlarr[15152]: [Info] CommandExecutor: Starting 3 threads for tasks.
Feb 19 12:27:01 Jackett Prowlarr[15152]: [Info] Microsoft.Hosting.Lifetime: Application started. Press Ctrl+C to shut down.
Feb 19 12:27:01 Jackett Prowlarr[15152]: [Info] Microsoft.Hosting.Lifetime: Hosting environment: Production
Feb 19 12:27:01 Jackett Prowlarr[15152]: [Info] Microsoft.Hosting.Lifetime: Content root path: /
```

MichaIng, 2024-02-19:
Hmm, the logs look okay: The updater kills the main process, systemd leaves the updater alive and just restarts the main process. Based on the 42 seconds between "Prowlarr will restart shortly." and main process kill I would say the updater finished, and itself killed Prowlarr once done. But indeed the same old v1.12 starts 🤔.

Can you check `journalctl` and `/var/log/prowlarr` whether there are logs from the updater process, at best with timestamps to compare/complement with Prowlarr service logs?

And lets also check whether the systemd service really is the latest one we ship:
```
systemctl cat prowlarr
```

gioxx, 2024-02-20:
Ciao @MichaIng and thanks.
I tested again today after upgrading to DietPi 9.1.1 and this time everything went smoothly. I cannot explain why but the procedure performed was the same as yesterday, this time, however, something has evidently changed:

```
Feb 20 11:48:59 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Downloading update 1.13.3.4273
Feb 20 11:49:02 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Verifying update package
Feb 20 11:49:03 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Update package verified successfully
Feb 20 11:49:03 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Extracting Update package
Feb 20 11:49:04 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Update package extracted successfully
Feb 20 11:49:04 Jackett Prowlarr[26050]: [Info] BackupService: Starting Backup
Feb 20 11:49:06 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Preparing client
Feb 20 11:49:18 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Starting update client /tmp/prowlarr_update/Prowlarr.Update
Feb 20 11:49:18 Jackett Prowlarr[26050]: [Info] InstallUpdateService: Prowlarr will restart shortly.
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Main process exited, code=killed, status=9/KILL
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Failed with result 'signal'.
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Unit process 28006 (Prowlarr.Update) remains running after unit stopped.
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Consumed 37.704s CPU time.
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 1.
Feb 20 11:50:26 Jackett systemd[1]: Stopped Prowlarr (DietPi).
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Consumed 37.721s CPU time.
Feb 20 11:50:26 Jackett systemd[1]: prowlarr.service: Found left-over process 28006 (Prowlarr.Update) in control group while starting unit. Ignoring.
Feb 20 11:50:26 Jackett systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Feb 20 11:50:26 Jackett systemd[1]: Started Prowlarr (DietPi).
Feb 20 11:50:26 Jackett Prowlarr[28048]: [Info] Bootstrap: Starting Prowlarr - /opt/prowlarr/Prowlarr - Version 1.13.3.4273
Feb 20 11:50:26 Jackett Prowlarr[28048]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 20 11:50:26 Jackett Prowlarr[28048]: [Debug] Bootstrap: Console selected
Feb 20 11:50:26 Jackett Prowlarr[28048]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/prowlarr/prowlarr.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3;busytimeout=100 ***
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: DatabaseEngineVersionCheck migrating
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: PerformDBOperation
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: Performing DB Operation
Feb 20 11:50:27 Jackett Prowlarr[28048]: SQLite notice (283): recovered 1 frames from WAL file /mnt/dietpi_userdata/prowlarr/prowlarr.db-wal
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] DatabaseEngineVersionCheck: SQLite 3.34.1
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.0321526s
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: DatabaseEngineVersionCheck migrated
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.0334971s
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: 39: email_encryption migrating
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: Beginning Transaction
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] email_encryption: Starting migration of Main DB to 39
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: PerformDBOperation
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: Performing DB Operation
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.0053188s
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: INSERT INTO "VersionInfo" ("Version", "AppliedOn", "Description") VALUES (39, '2024-02-20T10:50:27', 'email_encryption')
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: Committing Transaction
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: 39: email_encryption migrated
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.0287963s
Feb 20 11:50:27 Jackett Prowlarr[28048]: [Debug] MigrationController: Took: 00:00:00.1615278
[... cut ...]
-rw-r--r-- 1 prowlarr dietpi 700416 Feb 20 11:49 logs.db
-rw-r--r-- 1 prowlarr dietpi  32768 Feb 20 11:49 logs.db-shm
-rw-r--r-- 1 prowlarr dietpi  24752 Feb 20 11:49 logs.db-wal
-rw-rw-r-- 1 prowlarr dietpi   4237 Feb 20 11:49 prowlarr.txt
```

Yesterday I had also tried rebooting the VM to get a clean start but evidently that was not enough. No matter what, in the end the upgrade was successful and that is the important thing.
Thank you again for your support, I will close the issue directly.

closed 2024-02-20 by gioxx

MichaIng, 2024-02-20:
Actually the shutdown and updater part of the logs look exactly the same. Only on restart the correct new version is restarted. So why it was the old version, without any error message from the updater, will remain a mystery. But maybe there were errors done by the `Prowlarr.Update` process, just not visible in the journal. Not sure whether it logs to `prowlarr.txt`, but as this is a RAM log, those would be gone.

I remember we had a similar issue reported with another Servarr server. I'll keep this in mind, and when such happens again for someone, we need to check how to find logs from the dedicated updater process. Because it must have failed to replace the files, otherwise it would be impossible for the old version to start.