MichaIng/DietPi #6926: error while updating

dinucacid, 2024-02-20:
#### Details:
- Date           | Tue Feb 20 08:17:17 GMT 2024
- DietPi version | v8.25.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ASUS Tinker Board (armv7l) (ID=52)
- Kernel version | `Linux DietPi 6.1.63-current-rockchip #1 SMP PREEMPT Mon Nov 20 10:52:19 UTC 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ran dietpi-update from putty.
2. ...
#### Expected behaviour:

- update from 8 to 9
#### Actual behaviour:

- error
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Get:3 http://weewx.com/apt/python3 buster InRelease [4252 B]
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Get:7 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Err:3 http://weewx.com/apt/python3 buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY E788768B9CB16E10
Hit:6 https://au.sbcmirror.org/armbian/apt bullseye InRelease
Fetched 10.8 kB in 3s (3224 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://weewx.com/apt/python3 buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY E788768B9CB16E10
E: Failed to fetch http://weewx.com/apt/python3/dists/buster/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY E788768B9CB16E10
E: Some index files failed to download. They have been ignored, or old ones used instead.

### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ...
2. ...

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2024-02-20:
> Err:3 http://weewx.com/apt/python3 buster 

You seem to have added a 3rd party source manually to your system. The source is pointing to Debian Buster, while your systems is on Bullseye already. Something you would need to correct.

MichaIng, 2024-02-20:
WeeWX has no Bullseye suite, hence is probably not maintained anymore. But if you want to try using it on Bullseye, it seems the APT key is missing:
```sh
curl -sSf 'https://weewx.com/keys.html' | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/weewx.gpg --yes
```
For reference: https://weewx.com/apt/

closed 2025-02-26 by MichaIng