MichaIng/DietPi #6929: DietPi-Software | Netdata: Removed from Trixie

MichaIng, 2024-02-20:
- https://tracker.debian.org/pkg/netdata
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1042533
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1045145

Nothing seems to have happened in a year.

Worst case, if this is not fixed until Trixie release, we could switch to Netdata's own repository: https://repo.netdata.cloud/repos/stable/debian/, https://repo.netdata.cloud/netdatabot.gpg.key
But it is not ARMv6-compatible most likely.

MichaIng, 2025-05-15:
Raspbian Trixie has netdata compiled, v1.43.2-1 at time of writing:
- http://raspbian.raspberrypi.com/raspbian/dists/trixie/main/binary-armhf/
- http://raspbian.raspberrypi.com/raspbian/pool/main/n/netdata/

MichaIng, 2025-09-28:
Thanks for the voice on this. I am highly interested to get some opinion, as I am still unsure how we should react to it. I am Netdata is not the only closed source (full or partly) or otherwise commercial software we have install options for. And one can still hope Netdata remains valuable for free users as well. But I am also a FLOSS enthusiast and had no issue to support a clear reaction against the step from open to closed source they did.

Isildur981, 2025-10-02:
I've been using Netdata, since it was in beta, but I've been unhappy with the direction they've been going for a while.  One of the big reasons I started using Netdata is that it is very easy to setup.  Their remote monitoring makes it easy to provide support to remote computers.  I use Netdata to keep an eye on RPi's I helped my brother and sister setup(all running DietPi), who both live 2 hours away.  I've looked at other monitoring projects, but have never taken the time to install and play around with any. Two projects I've looked at recently are Beszel and Checkmate.  I'm pretty sure Beszel doesn't have all the features that Netdata does, but that's probably alright for monitoring a SBC installed at a home or small business.  I wouldn't be opposed at all if you decided to remove Netdata and replace it with some other monitoring software.  If I really need to keep using Netdata, I could always run their installer script.  

[Beszel](https://beszel.dev/)

[Checkmate](https://checkmate.so/)