MichaIng/DietPi #6935: Rock4 C+ Unable to get WiFi to work

arpegius5555, 2024-02-22:
### Creating a bug report/issue
- [ Unable to get WiFi to work on Rock4 C+] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
`
- Distro version | `bookworm`
- Kernel version | `6.1.63-current-rockchip64 #1 SMP PREEMPT Mon Nov 20 10:52:19 UTC 2023 aarch64 GNU/Linux`
- SBC model | `ROCK 4 (aarch64)` or (EG: RPi3)
- Power supply used | USB C
- SD card used |Sandisk Max Endurance

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

1. ...Install DietPi, SSH via LAN, try to enable WLAN, Unable to
2. ...

#### Expected behaviour

-Able to Configure WiFi

#### Actual behaviour

- ...WiFi is not found

#### Extra details

- ...

MichaIng, 2024-02-22:
Many thanks for your report.

You tried it via `dietpi-config`, right? Because some WiFi-related kernel modules are blacklisted by default, and need to be loaded first.

If so, the following command does not list the WiFi adapter, does it?
```sh
ip l
```
Are there some kernel errors shown? And can you also show the list of loaded kernel modules:
```sh
dmesg -l 0,1,2,3
lsmod
```
It can be tested with a newer kernel build we host on our server:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/{linux-{image,dtb}-current-rockchip64.deb,armbian-firmware.deb}
dpkg -i linux-{image,dtb}-current-rockchip64.deb armbian-firmware.deb
rm linux-{image,dtb}-current-rockchip64.deb armbian-firmware.deb
reboot
```

arpegius5555, 2024-02-22:
Correct, tried via dietpi-config

ip l:
1: lo: <LOOPBACK,UP,LOWER_UP>
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP>

dmesg -l 0,1,2,3
[    1.601255] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with 0-001b
[    1.764960] dw-apb-uart ff180000.serial: Failed to create device link (0x180) with vcc3v3-sys
[    1.930008] rk808 0-001b: failed to read the chip id at 0x17
[    7.122682] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    7.143217] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000000] -6
[    7.184481] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x0000000c] -6
[    7.186518] es8316 1-0011: ASoC: error at soc_component_write_no_lock on es8316.1-0011 for register: [0x00000003] -6
[    7.188760] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    7.190256] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    7.190612] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    7.192234] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    7.194809] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    7.198119] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    7.199856] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    7.200177] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    7.200875] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    7.201130] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000014] -6
[    7.203147] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000009] -6
[    7.203428] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000a] -6
[    7.203643] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x0000000b] -6
[    7.203846] es8316 1-0011: ASoC: error at snd_soc_component_update_bits on es8316.1-0011 for register: [0x00000001] -6
[    7.226889] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.232023] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.232756] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.233174] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    7.234128] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.234435] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.234962] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    7.235351] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    7.235794] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000022] -6
[    7.237306] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000025] -6
[    7.238718] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000a] -6
[    7.239092] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000030] -6
[    7.239428] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000000d] -6
[    7.239690] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000001] -6
[    7.239940] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[    7.240241] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x0000002f] -6
[    7.240717] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    7.241122] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000013] -6
[    7.241669] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000015] -6
[    7.242057] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000015] -6
[    7.242365] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000015] -6
[    7.242617] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000015] -6
[    7.242871] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000017] -6
[    7.243296] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011 for register: [0x00000017] -6
[... cut ...]
stmmac                237568  3 stmmac_platform,dwmac_rk
pcs_xpcs               20480  1 stmmac

_____________________________________________
I've tried the commands below, and after reboot I am unable to ssh anymore. Will plug in a mini HDMI and see what shows up  in the console

cd /tmp
wget https://dietpi.com/downloads/binaries/{linux-{image,dtb}-current-rockchip64.deb,armbian-firmware.deb}
dpkg -i linux-{image,dtb}-current-rockchip64.deb armbian-firmware.deb
rm linux-{image,dtb}-current-rockchip64.deb armbian-firmware.deb
reboot

MichaIng, 2024-02-22:
Can you check whether the correct device tree is loaded?
```sh
cat /proc/device-tree/model
```

arpegius5555, 2024-02-22:
cat /proc/device-tree/model
Radxa ROCK Pi 4Broot@

MichaIng, 2024-02-25:
Wrong device tree, that explains it. Try this:
```sh
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4c-plus.dtb' /boot/dietpiEnv.txt
```
A little sad that we then need to offer another dedicated image for this variant, like for the ROCK 4 SE before. Nasty that a single U-Boot build does not support all of them. It did in the past: https://dietpi.com/forum/t/rock-4c-how-do-i-get-the-temperature/17744/6
But we could at least check whether there is a variable in the U-Boot environment, like `$variant`, `$model` or `$board`, which indicates the variant, so we could select the correct device tree dynamically in our `boot.scr`.

Can you also do this:
```sh
sed -i '/^setenv bootargs/s/"$/ variant=${variant} board=${board} model=${model}"/' /boot/boot.cmd
mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
```
And after a reboot:
```sh
cat /proc/cmdline
```

arpegius5555, 2024-02-29:
Sorry about the delay but the old board did not boot anymore, so I had to place it. I just ran the following commands which I show below and rebooted. I haven't ssh'd yet but I just notice now the blue led on the board is blinking, it wasn't prior to tehse command and the reboot.

root@DietPi:~# G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4c-plus.dtb' /boot/dietpiEnv.txt
[  OK  ] G_CONFIG_INJECT | Added setting fdtfile=rockchip/rk3399-rock-4c-plus.dtb to end of file /boot/dietpiEnv.txt

root@DietPi:~# mkimage -C none -A arm64 -T script -d /boot/boot.cmd /boot/boot.scr
Image Name:
Created:      Thu Feb 29 00:51:58 2024
Image Type:   AArch64 Linux Script (uncompressed)
Data Size:    2749 Bytes = 2.68 KiB = 0.00 MiB
Load Address: 00000000
Entry Point:  00000000
Contents:
   Image 0: 2741 Bytes = 2.68 KiB = 0.00 MiB

root@DietPi:~# cat /proc/cmdline
root=UUID=699a00ce-f9e7-483c-888c-f71a2e182070 rootfstype=ext4 rootwait console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0 variant= board=evb_rk3399 model=

Now when I run  ip l I can see it there!!

3: wlan0: <BROADCAST,MULTICAST> 

I did proceed to configure via dietpi-config and it WORKS!!!!!!!!!!!!!!!!!!!

Thanks a million for always being so supportive!

MichaIng, 2024-04-21:
Could someone test the new image from here: https://dietpi.com/downloads/images/testing/
It uses a new ROCK 4C Plus U-Boot build, which selects the right device tree automatically.

arpegius5555, 2024-04-30:
Fresh Install, ran dietpi-config then > network options: adapters:

WiFi     : Not Found | [On] | Disconnected 

Enter
 
 No WiFi adapter was detected on your device.                                                                         │
                                         │                                                                                                                      │
                                         │ Would you like to disable WiFi?                                                                                      │
                                         │  - NB: Connections may drop!

MichaIng, 2024-04-30:
Has the correct device tree been selected?
```sh
cat /proc/device-tree/model
```

arpegius5555, 2024-04-30:
root@DietPi:~# cat /proc/device-tree/model
Radxa ROCK Pi 4Croot@DietPi:~#

MichaIng, 2024-04-30:
Hmm, the "Plus" is missing. So even that this is a U-Boot build for ROCK 4C Plus in particular, it seems to the the non-Plus device tree. So this does still fix it?
```sh
G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4c-plus.dtb' /boot/dietpiEnv.txt
```
Then I'll hardcode this device tree in our images.

arpegius5555, 2024-04-30:
root@DietPi:~# G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3399-rock-4c-plus.dtb' /boot/dietpiEnv.txt
[  OK  ] G_CONFIG_INJECT | Added setting fdtfile=rockchip/rk3399-rock-4c-plus.dtb to end of file /boot/dietpiEnv.txt
root@DietPi:~# reboot

So I immediately notice two things, blue led starts blinking and second my video finally comes up via HCMI cable (I was starting to hesitate about my cable since I did a headless install and fished out the IP by DHCP).

I am logging in now and after running dietpi-config:

 WiFi     : Available | [On] | Disconnected 
 
 Ran an ip add and see wlan0 has a valid IP address, so yeah it fixed a few things.
 Please le me know if you want me to test any further

MichaIng, 2024-04-30:
And it also shows the correct model name?
```sh
cat /proc/device-tree/model
```

MichaIng, 2024-05-02:
I moved the images to stable downloads and will add them to our download page now. Marking this as closed.

closed 2024-05-02 by MichaIng